Koenkk/zigbee2mqtt #889: Possible risks with "permit_join: true"

leobel96, 2019-01-19:
By default "permit_join: true". In my opinion this could led to conflicts (a neighbour could not succeed in connecting his device to his coordinator) or vulnerabilites (a malicious device can be connected by a stranger with zero effort) if a user isn't warned about the risks. It should be false by default and a timer should automatically set it to false after some time for me.

Koenkk, 2019-01-19:
The user is warned about this when you start zigbee2mqtt.

leobel96, 2019-01-19:
Yes but in my opinion it is not enough. A timer is necessary as in commercial gateways...

Koenkk, 2019-01-20:
@Landrash looks good, can you make a pr?

Landrash, 2019-01-20:
@Koenkk Doing it now and I'll update this with the PR #X

leobel96, 2019-01-20:
The solution proposed by @Landrash seems ok for me. I think the issue can be closed.

closed 2019-01-20 by leobel96