Koenkk/zigbee2mqtt #891: zigbee2mqtt > 0.16 stops responding

tb205gti, 2019-01-19:
I have some issues with a network of almost only Xiaomi devices.

(devices:
Currently 12 devices are joined:
(0x00158d0002370373): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
(0x90fd9ffffee0f460): LED1545G12 - IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (Router)
(0x00158d0002322769): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
(0x00158d0002320760): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
(0x00158d000232763c): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
(0x00158d00024edaa7): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
(0x00158d000273a494): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
(0x00158d0002751a27): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
(0x00158d000273d32e): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
(0x00158d0002b45124): WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice)
0x000b57fffecb7ccc (0x000b57fffecb7ccc): LED1624G9 - IKEA TRADFRI LED bulb E27 600 lumen, dimmable, color, opal white (Router)
(0x00158d000213295f): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
)

0.16 works perfectly, but I do not have support for vibration sensors. However if I use any of the version above 0.16 all devices stops responding after some time. I can manually send a MQTT command to my Ikea Tradfri bulb - and then all devices starts working again. But after some time it breaks again. No errors. Seems all updates from sensors doesn't get through. 0.16 works perfectly.

From the release logs I can see that some timeout have been changed, so I tried to add: soft_reset_timeout: 3600 to my config - but I still get the issue. Everything stops responding after some time, unless I push a MQTT message to the Ikea Tradfri bulb.

I'd _really_ like to run a newer version than 0.16 :/

Koenkk, 2019-01-19:
Can you provide the full log?

Koenkk, 2019-01-21:
When using 0.1.6, was the soft reset timeout required in order to keep the network running?

Koenkk, 2019-01-21:
Could you provide the complete startup log?

tb205gti, 2019-01-21:
Like this? https://pastebin.com/raw/u2m5nWcB

it is the immediate log from npm start on zigbee2mqtt 0.20

Koenkk, 2019-01-22:
As you are on an older coordinator firmware version, I would recommend updating to the latest one to see if that fixes your issues. Note that this requires repairing of your devices.

closed 2019-04-02 by stale[bot]