Koenkk/zigbee2mqtt #892: IKEA Fyrtur Kadrilj Blinds

Luftloch80, 2019-01-19:
Hey. In a few days These new devices will be Released. 
Will it be possible to Support Them. Because will be the First zigbee blinds? Or do you add code to Support blinds ?
Thx

Koenkk, 2019-01-19:
We can check when somebody gets their hands on one of them.

Luftloch80, 2019-01-19:
I will definitiv Buy as Soon as avaiable in Germany

osos, 2019-08-16:
They are now available:

And promising for DIY: https://www.youtube.com/watch?v=tfJn6oNGF74

osos, 2019-08-16:
Please reopen

Koenkk, 2019-08-18:
If somebody owns this device and wants to add support for it, we can reopen it.

osos, 2019-08-19:
Ok. A friend bought a couple of these.
We are going to add them to zigbee2mqtt asap.

However, looking through you website, I do not understand how to find out which commands are available to control and create the "toZigBee" converters.

Please provide some pointers.

Koenkk, 2019-08-21:
I think you can reuse the `cover_control` converter: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/toZigbee.js#L187

darkkatarsis, 2019-08-25:
There is some information about these blinds in this thread. It may be useful -> https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121

osos, 2019-08-31:
Two issues.

1. Getting Ikea E1766 remote (Open/Close Remote) working.

Pairing as usual. 4 short pushed on the reset button on the back. Paired by holding close to signal repeater.

It reveals itself as model = modelID = "TRADFRI open/close remote"

Added to devices.js:
`   {
        zigbeeModel: ['TRADFRI open/close remote'],
        model: 'TRADFRI open/close remote',
        vendor: 'IKEA',
        description: 'TRADFRI open/close remote',
        supports: 'up, down, release',
        fromZigbee: [ fz.genOnOff_cmdOn, fz.genOnOff_cmdOff, fz.TRADFRI_open_close_stop ],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {
                direction: 0, attrId: 33, dataType: 32, minRepIntval: 0, maxRepIntval: repInterval.MAX, repChange: 0,
            };

            const actions = [
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.foundation('genPowerCfg', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
    },
`

and into fromZigBee.js:
` 
   TRADFRI_open_close_stop: {
        cid: 'closuresWindowCovering',
        type: 'cmdStop',
        convert: (model, msg, publish, options) => {
            return {click: 'closure_stop'};
        },
    },
`

However, I do not get any events nor warnings for regular button push. The stop function is activated by long-press any button.

Seems supported by deconz: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/dc0d508d6a8c49d4cf59a3b72791f31f726d801b/sensor.cpp#L146

2. FYRTUR blinds.

Pairing easy.:
Hold both buttons on blind for 5 seconds, release and hold close to repeater.

Reveals itself as:
> Device with modelID 'FYRTUR block-out roller blind' is not supported.

Working on getting it working.

Any help on both appreciated.

alpenhase83, 2019-08-31:
I've installed two Fyrtur blinds this morning. After installing the blinds i've tried to add them to zigbee2mqtt but i experience the same issues as @osos. 

Pairing was very easy but afterwards i didn't receive any event in zigbee2mqtt.

tb-killa, 2019-08-31:
The Remote has a client Window Covering cluster, to send comands to the blind. This means that it can only control (directly) the blind, and that the blind cannot be controlled (directly) by the other control devices. Not too happy about the 0xFC7C cluster, which is probably used for some IKEA-specific configuration.
We need to sniff the specific configuration. 

The remote reporting its battery percentage (cluster 0x0001, attribute 0x0021).

The FYRTUR does support attribute reporting for Battery Percentage Remaining (0x0001/0x0021) and, for Current Position Lift Percentage (0x0102/0x0008).
For this we also need configure reporting to see the actually lift state. 

We could use also SW Build ID (0x0000/0x4000).

From what I see their should be a group ID where they act like we see from other ikea remotes.

tb-killa, 2019-08-31:
@osos please share your database file content.

osos, 2019-08-31:
From database.db:

Open/Close remote:
```
{"id":8,"type":"EndDevice","ieeeAddr":"0x000d6ffffee3c176","nwkAddr":56685,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI open/close remote","epList":[1],"zclVersion":3,"appVersion":34,"stackVersion":98,"hwVersion":1,"swBuildId":"2.2.008","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":515,"inClusterList":[0,1,3,9,32,4096,64636],"outClusterList":[3,4,6,8,25,258,4096],"clusters":{"64636":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"9":226,"10":{},"11":"","65533":1,"zclVersion":3,"appVersion":34,"stackVersion":98,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI open/close remote","dateCode":"20190311","powerSource":3,"appProfileVersion":1,"swBuildId":"2.2.008"}},"genPowerCfg":{"dir":{"value":1},"attrs":{"65533":1,"batteryVoltage":29,"batteryPercentageRemaining":74}},"genIdentify":{"dir":{"value":3},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{"65533":1}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{"65533":1,"checkinInterval":172800,"longPollInterval":1160,"shortPollInterval":4,"fastPollTimeout":40}},"closuresWindowCovering":{"dir":{"value":2},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"dateCode":"20190311","_id":"RcCUQl89Wv8qZqH2"}
```

Fyrtur:
```
{"id":9,"type":"EndDevice","ieeeAddr":"0x000d6ffffe2dea2f","nwkAddr":58822,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"FYRTUR block-out roller blind","epList":[1],"zclVersion":3,"appVersion":34,"stackVersion":99,"hwVersion":1,"swBuildId":"2.2.007","status":"online","joinTime":1567257368,"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,1,3,4,5,32,258,4096,64636],"outClusterList":[25,4096],"clusters":{"64636":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"9":255,"10":{},"11":"","65533":1,"zclVersion":3,"appVersion":34,"stackVersion":99,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"FYRTUR block-out roller blind","dateCode":"20190311","powerSource":3,"appProfileVersion":0,"swBuildId":"2.2.007"}},"genPowerCfg":{"dir":{"value":1},"attrs":{"65533":1,"batteryVoltage":0,"batteryPercentageRemaining":0}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":168}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{}},"closuresWindowCovering":{"dir":{"value":1},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}}},"dateCode":"20190311","_id":"jrVKz4IB9fT3m443"}
{"id":9,"type":"EndDevice","ieeeAddr":"0x000d6ffffe2dea2f","nwkAddr":58822,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"FYRTUR block-out roller blind","epList":[1],"zclVersion":3,"appVersion":34,"stackVersion":99,"hwVersion":1,"swBuildId":"2.2.007","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,1,3,4,5,32,258,4096,64636],"outClusterList":[25,4096],"clusters":{"64636":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"9":255,"10":{},"11":"","65533":1,"zclVersion":3,"appVersion":34,"stackVersion":99,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"FYRTUR block-out roller blind","dateCode":"20190311","powerSource":3,"appProfileVersion":0,"swBuildId":"2.2.007"}},"genPowerCfg":{"dir":{"value":1},"attrs":{"65533":1,"batteryVoltage":0,"batteryPercentageRemaining":0}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":168}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{}},"closuresWindowCovering":{"dir":{"value":1},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}}},"dateCode":"20190311","_id":"jrVKz4IB9fT3m443"}
```

osos, 2019-08-31:
@tb-killa, the blinds can be controlled via the Ikea App and google/youtube shows that the blinds can be connected to Philips Hue bridge, thus it should be possible to control without Ikea specifics. Right?

osos, 2019-09-01:
Inspiration may be found in deConz support for both the Open/Close remote and the blinds.

See pull request:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1774
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1783
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1785

I do not know where to go from here. I am able to sniff the traffic, but do not know how to use the information.
From traffic sniffing I see the open/close remote send 3 different commands, 0x00 (Up/Open), 0x01 (Down/close) and 0x02 (Stop). However, only the Stop triggered output in zigbee2mqtt.

serialtasted, 2019-09-03:
@osos How's it going? Any status update?

osos, 2019-09-03:
> @osos How's it going? Any status update?

Not good. Don't know how to continue with this.

serialtasted, 2019-09-03:
were you able to get the open close remote working?

Koenkk, 2019-09-23:
Can somebody try with the following configuration for the blinds:

```js
{
        zigbeeModel: ['FYRTUR block-out roller blind'],
        model: 'TODO',
        vendor: 'IKEA',
        description: 'Blinds',
        supports: 'open/close',
        fromZigbee: [fz.cover_position_report, fz.cover_position, fz.cover_state_change, fz.cover_state_report],
        toZigbee: [tz.cover_position, tz.cover_open_close],
    },
```

It should then work with `zigbee2mqtt/[DEVICE_ID]/set` `{"state": "OPEN"}` (and CLOSE)

Koenkk, 2019-09-24:
Can you try with

```js
{
        zigbeeModel: ['FYRTUR block-out roller blind'],
        model: 'TODO',
        vendor: 'IKEA',
        description: 'Blinds',
        supports: 'open/close',
        fromZigbee: [fz.cover_position_report, fz.cover_position, fz.cover_state_change, fz.cover_state_report],
        toZigbee: [tz.cover_control],
    },
```

And home assistant configuration `configurations.cover` (dont forgot to also restart home assistant after changing the configuration). Position wont work after this but atleast open close and stop should work.

syphernl, 2019-09-25:
> Can you try with
>
> ```js
> {
>         zigbeeModel: ['FYRTUR block-out roller blind'],
>         model: 'TODO',
>         vendor: 'IKEA',
>         description: 'Blinds',
>         supports: [...]

Awesome! It is opening, stopping and closing perfectly right now. The arrows work as they should, state reports "open" when it is open.

From what I understand it should be capable of sending its position and receiving a specific position, but for that some more work is required probably?

Koenkk, 2019-09-25:
Great, I've improved the naming over the cover converters a bit. Can you update to the latest dev brach and zigbee-herdsman-converters?

I expect it will fully work with the following configuration (including position):

devices.js
```js
{
    zigbeeModel: ['FYRTUR block-out roller blind'],
    model: 'E1757',
    vendor: 'IKEA',
    description: 'FYRTUR smart blinds',
    supports: 'open, close, stop, position, tilt',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
},
```

homeassistant.js

```js
'E1757': [cfg.cover_position_tilt],
```

syphernl, 2019-09-25:
> Great, I've improved the naming over the cover converters a bit. Can you update to the latest dev brach and zigbee-herdsman-converters?
>
> I expect it will fully work with the following configuration [...]

With the dev version/herdsman and the changes above the position does work however:
* 100% is 100% open, not 100% closed (what you'd expect). 
* When the blind is opened fully (position: 100%) the up arrow shows instead of the down arrow (which is greyed out). The only way to get the blind down is by changing the position slider from 100 to 95 for instance. However, the down arrow still remains greyed out.
* It shows tilting options but this has no effect on this type of blind.

Koenkk, 2019-09-25:
Can you:

- Add to https://github.com/Koenkk/zigbee-shepherd-converters/blob/zigbee-herdsman-converters/converters/fromZigbee.js#L2364 `console.log(msg.data)`.
- Indicate what is logged when it is fully open and closed.

syphernl, 2019-09-27:
> Can you:
> 
> * Add to https://github.com/Koenkk/zigbee-shepherd-converters/blob/zigbee-herdsman-converters/converters/fromZigbee.js#L2364 `console.log(msg.data)`.
> * Indicate what is logged when it is fully open and closed.

I moved the slider (since the buttons are not working) to 0% and 100% but it doesn't log anything (or it logs at a level that is below the one `npm start` uses by default?)

dsbaars, 2019-09-27:
> Great, I've improved the naming over the cover converters a bit. Can you update to the latest dev brach and zigbee-herdsman-converters?
>
> I expect it will fully work with the following configuration [...]

I made it also work with the Kadrilj blinds (E1926):

```js
    {
    zigbeeModel: ['KADRILJ roller blind'],
    model: 'E1926',
    vendor: 'IKEA',
    description: 'KADRILJ roller blind',
    supports: 'open, close, stop, position, tilt',
    fromZigbee: [fz.cover_position_tilt],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
},
```

```js
'E1926': [cfg.cover_position_tilt],
```

Koenkk, 2019-09-27:
@syphernl can you try changing https://github.com/Koenkk/zigbee-shepherd-converters/blob/zigbee-herdsman-converters/converters/fromZigbee.js#L2368 to ` result.position = liftPercentage <= 100 ? liftPercentage : null;`

alpenhase83, 2019-09-28:
> I will definitiv Buy as Soon as avaiable in Germany

I've bought mine in Netherlands. But some days later i've seen Fyrtur blinds at IKEA in Düsseldorf (GER) also. Even though the blinds are not offically listet online in the store.

syphernl, 2019-09-28:
> result.position = liftPercentage <= 100 ? liftPercentage : null;

Open / Close works fine (open = up, close = down as it should). 
However its state is unknown in HomeAssistant.

The tilt buttons do not work and neither does setting a specific position. 
Whether I set it to 0, 100, 50 or anywhere in between it won't move.

Koenkk, 2019-09-28:
@syphernl could you perhaps contact me on telegram (koenkk), in this way we can iterate faster.

syphernl, 2019-09-28:
> @syphernl could you perhaps contact me on telegram (koenkk), in this way we can iterate faster.

On it!

Koenkk, 2019-09-28:
Device is supported in the dev branch now, credit go to @syphernl :smile;

dsbaars, 2019-09-28:
> Device is supported in the dev branch now, credit go to @syphernl :smile;

Note that, next to the model ID, the Zigbee model naming between the two is actually different so I think you need to add `'KADRILJ roller blind'`  to the  `zigbeeModel` array as well :)

RaymondMouthaan, 2019-09-28:
Just when back from an Ikea tour and finally managed to get 2 Kadrilj blinds (140 cm). Really want these connected to zigbee2mqtt. Anyone willing to help a bit? :-)

Koenkk, 2019-09-28:
@dsbaars good catch, also added that one, both KADRILJI and FYRTUR are supported in the dev branch now.

alpenhase83, 2019-09-28:
@syphernl @Koenkk Thank you so much! My blinds are working like a charm with OpenHAB.

RaymondMouthaan, 2019-09-28:
Is this already available in a docker images? (koenkk/zigbee2mqtt:/latest-dev)

Koenkk, 2019-09-28:
Yes

RaymondMouthaan, 2019-09-28:
Hmmm couldn't get it to work. 😔

What are the steps to get the blinds, signal repeater and open-close button work?

syphernl, 2019-09-29:
> Hmmm couldn't get it to work. 😔
> 
> What are the steps to get the blinds, signal repeater and open-close button work?

**Signal repeater**: See https://www.zigbee2mqtt.io/devices/E1746.html
**Blinds**: Enable join mode in Z2M. Hold both buttons (~5 seconds) on the blind until the leds start flashing / turn on. After some time it should show up.
**Remote:** https://www.zigbee2mqtt.io/devices/E1743.html. I haven't used this one yet. IIUC it should be pairable directly to the blinds while they are part of the Zigbee network but haven't yet tried this. Otherwise you can bind the remote to your coordinator and use it from there.

The blinds work fine here without the repeater. I however do have some TRADFRI Bulbs and a power socket on that floor that they use as hop to my coordinator.

RaymondMouthaan, 2019-09-29:
@syphernl, thanks for your reply! Would you be available for a chat?

aimartin, 2019-09-29:
I've bought an Ikea blind today, and I'll try to connect to Z2M, but I've a question, is the Signal Repeater mandatory for the blinds to work?

Regards!

RaymondMouthaan, 2019-09-29:
@abmantis, no it's not.

aimartin, 2019-09-29:
> @abmantis, no it's not.

Thanks! With the remote (even quite near to the blind) I wasn't able to operate it without the SR connected... :S

RaymondMouthaan, 2019-09-29:
> Thanks! With the remote (even quite near to the blind) I wasn't able to operate it without the SR connected... :S

Well while pairing the blind, I had it as close as possible (could be close) to the coordinator. Since the SR came with the packages, I've added them as well to Z2M :)

Koenkk, 2019-09-29:
@RaymondMouthaan as you have paired the remote, could you start zigbee2mqtt with `DEBUG=zigbee-herdsman* npm start`) (I assume you use docker so set the `DEBUG=zigbee-herdsman*` environment variable inside the container). Can you provide the logging when pressing the buttons?

RaymondMouthaan, 2019-09-29:
I'm on it ...

RaymondMouthaan, 2019-09-29:
```
2019-09-29T19:58:24.363Z zigbee-herdsman:unpi:parser --- parseNext []

2019-09-29T19:58:24.364Z zigbee-herdsman:controller:zStack:startZnp Item 'networkKeyDistribute' is valid

2019-09-29T19:58:24.367Z zigbee-herdsman:znp:SREQ --> SAPI - readConfiguration - {"id":98,"configid":98,"len":16,"offset":0,"value":{"type":"Buffer","data":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]}}

2019-09-29T19:58:24.369Z zigbee-herdsman:unpi:writer --> frame [254,1,38,4,98,65]

2019-09-29T19:58:24.393Z zigbee-herdsman:unpi:parser <-- [254,19,102,4,0,98,16,1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13,0]

2019-09-29T19:58:24.394Z zigbee-herdsman:unpi:parser --- parseNext [254,19,102,4,0,98,16,1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13,0]

2019-09-29T19:58:24.395Z zigbee-herdsman:unpi:parser --> parsed 19 - 3 - 6 - 4 - [b	



] - 0

2019-09-29T19:58:24.396Z zigbee-herdsman:znp:SRSP <-- SAPI - readConfiguration - {"status":0,"configid":98,"len":16,"value":{"type":"Buffer","data":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]}}

2019-09-29T19:58:24.397Z zigbee-herdsman:unpi:parser --- parseNext []

2019-09-29T19:58:24.398Z zigbee-herdsman:controller:zStack:startZnp Item 'networkKey' is valid

2019-09-29T19:58:24.400Z zigbee-herdsman:znp:SREQ --> SYS - osalNvRead - {"id":131,"len":2,"offset":0,"value":{"type":"Buffer","data":[98,26]}}

2019-09-29T19:58:24.402Z zigbee-herdsman:unpi:writer --> frame [254,3,33,8,131,0,0,169]

2019-09-29T19:58:24.410Z zigbee-herdsman:unpi:parser <-- [254,4,97,8,0,2,98,26,23]

2019-09-29T19:58:24.411Z zigbee-herdsman:unpi:parser --- parseNext [254,4,97,8,0,2,98,26,23]

2019-09-29T19:58:24.412Z zigbee-herdsman:unpi:parser --> parsed 4 - 3 - 1 - 8 - [b] - 23

2019-09-29T19:58:24.412Z zigbee-herdsman:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":2,"value":{"type":"Buffer","data":[98,26]}}

2019-09-29T19:58:24.413Z zigbee-herdsman:unpi:parser --- parseNext []

2019-09-29T19:58:24.413Z zigbee-herdsman:controller:zStack:startZnp Item 'panID' is valid

2019-09-29T19:58:24.415Z zigbee-herdsman:znp:SREQ --> SYS - osalNvRead - {"id":45,"len":8,"offset":0,"value":{"type":"Buffer","data":[221,221,221,221,221,221,221,221]}}

2019-09-29T19:58:24.415Z zigbee-herdsman:unpi:writer --> frame [254,3,33,8,45,0,0,7]

2019-09-29T19:58:24.424Z zigbee-herdsman:unpi:parser <-- [254,10,97,8,0,8,221,221,221,221,221,221,221,221,107]

2019-09-29T19:58:24.425Z zigbee-herdsman:unpi:parser --- parseNext [254,10,97,8,0,8,221,221,221,221,221,221,221,221,107]

2019-09-29T19:58:24.426Z zigbee-herdsman:unpi:parser --> parsed 10 - 3 - 1 - 8 - [��������] - 107

2019-09-29T19:58:24.427Z zigbee-herdsman:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":8,"value":{"type":"Buffer","data":[221,221,221,221,221,221,221,221]}}

2019-09-29T19:58:24.428Z zigbee-herdsman:unpi:parser --- parseNext []

2019-09-29T19:58:24.429Z zigbee-herdsman:controller:zStack:startZnp Item 'extendedPanID' is valid

2019-09-29T19:58:24.432Z zigbee-herdsman:znp:SREQ --> UTIL - getDeviceInfo - {}

2019-09-29T19:58:24.433Z zigbee-herdsman:unpi:writer --> frame [254,0,39,0,39]

2019-09-29T19:58:24.443Z zigbee-herdsman:unpi:parser <-- [254,30,103,0,0,96,172,54,25,0,75,18,0,0,0,7,9,8,217,38,137,214,214,218,64,75,163,165,178,33,91,207,252,45,178]

2019-09-29T19:58:24.444Z zigbee-herdsman:unpi:parser --- parseNext [254,30,103,0,0,96,172,54,25,0,75,18,0,0,0,7,9,8,217,38,137,214,214,218,64,75,163,165,178,33,91,207,252,45,178]

2019-09-29T19:58:24.445Z zigbee-herdsman:unpi:parser --> parsed 30 - 3 - 7 - 0 - [`�6K	�&����@K���![��-] - 178

2019-09-29T19:58:24.449Z zigbee-herdsman:znp:SRSP <-- UTIL - getDeviceInfo - {"status":0,"ieeeaddr":"0x00124b001936ac60","shortaddr":0,"devicetype":7,"devicestate":9,"numassocdevices":8,"assocdeviceslist":[9945,54921,56022,19264,42403,8626,53083,11772]}

2019-09-29T19:58:24.450Z zigbee-herdsman:unpi:parser --- parseNext []
[... 227 lines cut ...]
2019-09-29T20:00:24.242Z zigbee-herdsman:unpi:writer --> frame [254,15,36,1,252,45,1,1,2,1,3,0,30,5,24,7,11,0,0,244]

2019-09-29T20:00:24.243Z zigbee-herdsman:unpi:parser --- parseNext []

2019-09-29T20:00:24.252Z zigbee-herdsman:unpi:parser <-- [254,1,100,1,0,100]

2019-09-29T20:00:24.253Z zigbee-herdsman:unpi:parser --- parseNext [254,1,100,1,0,100]

2019-09-29T20:00:24.254Z zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 4 - 1 - [] - 100

2019-09-29T20:00:24.254Z zigbee-herdsman:znp:SRSP <-- AF - dataRequest - {"status":0}

2019-09-29T20:00:24.255Z zigbee-herdsman:unpi:parser --- parseNext []

2019-09-29T20:00:25.272Z zigbee-herdsman:unpi:parser <-- [254,3,68,128,0,1,3,197]

2019-09-29T20:00:25.273Z zigbee-herdsman:unpi:parser --- parseNext [254,3,68,128,0,1,3,197]

2019-09-29T20:00:25.274Z zigbee-herdsman:unpi:parser --> parsed 3 - 2 - 4 - 128 - [] - 197

2019-09-29T20:00:25.275Z zigbee-herdsman:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":3}

2019-09-29T20:00:25.276Z zigbee-herdsman:unpi:parser --- parseNext []

```

Koenkk, 2019-09-30:
@RaymondMouthaan added the missing commands, please update to the latest dev branch and you should see the `No converter available..` messages when pressing buttons.

RaymondMouthaan, 2019-09-30:
@Koenkk, not sure if I captured all the logs you need, but this is what I got:

```
2019-09-30T21:56:10.628Z zigbee-herdsman:controller:device Interview - got simple descriptor for endpoint '1' device '0x086bd7fffe57f940'

2019-09-30T21:56:10.646Z zigbee-herdsman:znp:SREQ --> AF - dataRequest - {"dstaddr":58205,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":4,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,2,0,5,0,4,0,7,0]}}

2019-09-30T21:56:10.647Z zigbee-herdsman:unpi:writer --> frame [254,19,36,1,93,227,1,1,0,0,4,0,30,9,16,2,0,5,0,4,0,7,0,143]

2019-09-30T21:56:10.657Z zigbee-herdsman:unpi:parser <-- [254,1,100,1,0,100]

2019-09-30T21:56:10.657Z zigbee-herdsman:unpi:parser --- parseNext [254,1,100,1,0,100]

2019-09-30T21:56:10.658Z zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 4 - 1 - [] - 100

2019-09-30T21:56:10.658Z zigbee-herdsman:znp:SRSP <-- AF - dataRequest - {"status":0}

2019-09-30T21:56:10.659Z zigbee-herdsman:unpi:parser --- parseNext []

2019-09-30T21:56:11.645Z zigbee-herdsman:unpi:parser <-- [254,3,68,128,0,1,4,194]

2019-09-30T21:56:11.645Z zigbee-herdsman:unpi:parser --- parseNext [254,3,68,128,0,1,4,194]

2019-09-30T21:56:11.646Z zigbee-herdsman:unpi:parser --> parsed 3 - 2 - 4 - 128 - [] - 194

2019-09-30T21:56:11.646Z zigbee-herdsman:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":4}

2019-09-30T21:56:11.647Z zigbee-herdsman:unpi:parser --- parseNext []

2019-09-30T21:56:11.667Z zigbee-herdsman:unpi:parser <-- [254,77,68,129,0,0,0,0,93,227,1,1,0,178,0,94,50,216,0,0,57,24,2,1,5,0,0,66,25,84,82,65,68,70,82,73,32,111,112,101,110,47,99,108,111,115,101,32,114,101,109,111,116,101,4,0,0,66,14,73,75,69,65,32,111,102,32,83,119,101,100,101,110,7,0,0,48,3,93,227,29,165]

2019-09-30T21:56:11.668Z zigbee-herdsman:unpi:parser --- parseNext [254,77,68,129,0,0,0,0,93,227,1,1,0,178,0,94,50,216,0,0,57,24,2,1,5,0,0,66,25,84,82,65,68,70,82,73,32,111,112,101,110,47,99,108,111,115,101,32,114,101,109,111,116,101,4,0,0,66,14,73,75,69,65,32,111,102,32,83,119,101,100,101,110,7,0,0,48,3,93,227,29,165]

2019-09-30T21:56:11.669Z zigbee-herdsman:unpi:parser --> parsed 77 - 2 - 4 - 129 - []��^2�9BTRADFRI open/close remoteBIKEA of Sweden0]�] - 165

2019-09-30T21:56:11.669Z zigbee-herdsman:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":58205,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":178,"securityuse":0,"timestamp":14168670,"transseqnumber":0,"len":57,"data":{"type":"Buffer","data":[24,2,1,5,0,0,66,25,84,82,65,68,70,82,73,32,111,112,101,110,47,99,108,111,115,101,32,114,101,109,111,116,101,4,0,0,66,14,73,75,69,65,32,111,102,32,83,119,101,100,101,110,7,0,0,48,3]}}

2019-09-30T21:56:11.675Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":2,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":5,"status":0,"dataType":66,"attrData":"TRADFRI open/close remote"},{"attrId":4,"status":0,"dataType":66,"attrData":"IKEA of Sweden"},{"attrId":7,"status":0,"dataType":48,"attrData":3}],"Cluster":{"ID":0,"attributes":{"zclVersion":{"ID":0,"type":32,"name":"zclVersion"},"appVersion":{"ID":1,"type":32,"name":"appVersion"},"stackVersion":{"ID":2,"type":32,"name":"stackVersion"},"hwVersion":{"ID":3,"type":32,"name":"hwVersion"},"manufacturerName":{"ID":4,"type":66,"name":"manufacturerName"},"modelId":{"ID":5,"type":66,"name":"modelId"},"dateCode":{"ID":6,"type":66,"name":"dateCode"},"powerSource":{"ID":7,"type":48,"name":"powerSource"},"appProfileVersion":{"ID":8,"type":48,"name":"appProfileVersion"},"swBuildId":{"ID":16384,"type":66,"name":"swBuildId"},"locationDesc":{"ID":16,"type":66,"name":"locationDesc"},"physicalEnv":{"ID":17,"type":48,"name":"physicalEnv"},"deviceEnabled":{"ID":18,"type":16,"name":"deviceEnabled"},"alarmMask":{"ID":19,"type":24,"name":"alarmMask"},"disableLocalConfig":{"ID":20,"type":24,"name":"disableLocalConfig"}},"name":"genBasic","commands":{"resetFactDefault":{"ID":0,"parameters":[],"name":"resetFactDefault"}},"commandsResponse":{}}},"networkAddress":58205,"endpoint":1,"linkquality":178,"groupID":0}'

2019-09-30T21:56:11.684Z zigbee-herdsman:unpi:parser --- parseNext []


qtt:debug 2019-09-30T21:56:11: Received Zigbee message from 'tradfri-open-close-remote-1' of type 'readResponse' with data '{"modelId":"TRADFRI open/close remote","manufacturerName":"IKEA of Sweden","powerSource":3}' from endpoint 1 with groupID 0

zigbee2mqtt:warn  2019-09-30T21:56:11: Received message from unsupported device with Zigbee model 'TRADFRI open/close remote'

zigbee2mqtt:warn  2019-09-30T21:56:11: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

2019-09-30T21:56:11.747Z zigbee-herdsman:controller:device Interview - got 'modelId,manufacturerName,powerSource' for device '0x086bd7fffe57f940'

2019-09-30T21:56:11.757Z zigbee-herdsman:znp:SREQ --> AF - dataRequest - {"dstaddr":58205,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":5,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,3,0,0,0,1,0,2,0]}}

2019-09-30T21:56:11.758Z zigbee-herdsman:unpi:writer --> frame [254,19,36,1,93,227,1,1,0,0,5,0,30,9,16,3,0,0,0,1,0,2,0,138]

2019-09-30T21:56:11.767Z zigbee-herdsman:unpi:parser <-- [254,1,100,1,0,100]

2019-09-30T21:56:11.768Z zigbee-herdsman:unpi:parser --- parseNext [254,1,100,1,0,100]

[... cut ...]
2019-09-30T22:05:51.799Z zigbee-herdsman:unpi:parser --- parseNext []

2019-09-30T22:05:52.814Z zigbee-herdsman:unpi:parser <-- [254,3,68,128,0,1,28,218]

2019-09-30T22:05:52.815Z zigbee-herdsman:unpi:parser --- parseNext [254,3,68,128,0,1,28,218]

2019-09-30T22:05:52.815Z zigbee-herdsman:unpi:parser --> parsed 3 - 2 - 4 - 128 - [] - 218

2019-09-30T22:05:52.816Z zigbee-herdsman:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":28}

2019-09-30T22:05:52.817Z zigbee-herdsman:unpi:parser --- parseNext []
```

Koenkk, 2019-10-01:
@RaymondMouthaan thanks, exactly what I needed, please try with the following branch: https://github.com/Koenkk/zigbee-shepherd-converters/tree/E1766

RaymondMouthaan, 2019-10-02:
> @RaymondMouthaan thanks, exactly what I needed, please try with the following branch: https://github.com/Koenkk/zigbee-shepherd-converters/tree/E1766

@Koenkk. can you explain in a few steps how to do that in conjunction  with a running docker container? Should i install git, clone the repo and check out the branch? In which dir?

Koenkk, 2019-10-02:
@RaymondMouthaan https://www.zigbee2mqtt.io/information/docker.html#support-new-devices

For the git clone, check out the `E17661` branch and instead of `   -v $(pwd)/data/zigbee-shepherd-converters:/app/node_modules/zigbee-shepherd-converters \` do `   -v $(pwd)/data/zigbee-shepherd-converters:/app/node_modules/zigbee-herdsman-converters \`

RaymondMouthaan, 2019-10-02:
@Koenkk, can we have a chat on telegram or discord? Not sure if i do the right thing here

Koenkk, 2019-10-02:
Yes, (koenkk)

simonriskjaer, 2019-10-02:
Great work. When will the blinds be officially supported? Running Hassio with the official Z2M addon - can't wait to control my blinds in HA.

Koenkk, 2019-10-03:
@simonriskjaer they are already in the edge version of the addon.

RaymondMouthaan, 2019-10-03:
@koenkk I get this error: 

'''
Ik krijg deze vrij regelmatig, kan ik daar iets aan doen?
zigbee2mqtt:error 2019-10-02T18:13:13: Publish 'set' 'position' to 'tradfri-window-blind-2' failed: 'Error: Data request failed with error: 'MAC transaction expired' (240)'
'''

How can I fix this?

simonriskjaer, 2019-10-03:
@Koenkk i know. The edge version fails to launch on my system, so I reverted back to the main addon. Just wondering what the release-cycle is for Z2M.

RaymondMouthaan, 2019-10-22:
When you got the MAC error, try to remove the battery from the blind for about 30 seconds and add it back in.

I had this issue before and @Koenkk advised to place the blind close(r) to the coordinator (or visaversa). This helped to control the blinds again.

After that I've paired the repeater(s) with z2m and moved the coordinator to its original location and removed the battery again and re-added back in. Hoping that the blind connects via the repeater to the coordinator. Not sure if it does, but up to now I have no more MAC issues and the blinds are perfectly controlled. 

You might want to create a network map see for details [this page](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html)

FYI - all of my devices:
```

zigbee2mqtt:info  2019-10-22T18:14:52: hue-dimmer-1 (0x0017880106b736ad): 324131092621 - Philips Hue dimmer switch (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-remote-1 (0x90fd9ffffee6e4c0): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-remote-2 (0x90fd9ffffee8ea42): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-window-blind-2 (0x000d6ffffea940f2): E1926 - IKEA KADRILJ roller blind (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-window-blind-1 (0x000d6ffffe9d5e34): E1926 - IKEA KADRILJ roller blind (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-signal-repeater-1 (0x000d6ffffea9402f): E1746 - IKEA TRADFRI signal repeater (Router)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-signal-repeater-2 (0x000d6ffffea1f2f2): E1746 - IKEA TRADFRI signal repeater (Router)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-open-close-remote-1 (0x086bd7fffe57f940): E1766 - IKEA TRADFRI open/close remote (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-open-close-remote-2 (0x086bd7fffe21aca4): E1766 - IKEA TRADFRI open/close remote (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-window-blind-3 (0x086bd7fffe5be1c5): E1757 - IKEA FYRTUR roller blind (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-window-blind-4 (0x086bd7fffe56a5b1): E1757 - IKEA FYRTUR roller blind (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-window-blind-5 (0x086bd7fffe57e708): E1757 - IKEA FYRTUR roller blind (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-window-blind-6 (0x086bd7fffe56b0bc): E1757 - IKEA FYRTUR roller blind (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-open-close-remote-3 (0x086bd7fffe2c49de): E1766 - IKEA TRADFRI open/close remote (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-open-close-remote-4 (0x086bd7fffe57f7fa): E1766 - IKEA TRADFRI open/close remote (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-signal-repeater-3 (0x000d6ffffe8e8619): E1746 - IKEA TRADFRI signal repeater (Router)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-signal-repeater-4 (0x000d6ffffe9cbc04): E1746 - IKEA TRADFRI signal repeater (Router)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-open-close-remote-5 (0x086bd7fffe22a4ba): E1766 - IKEA TRADFRI open/close remote (EndDevice)

zigbee2mqtt:info  2019-10-22T18:14:52: tradfri-open-close-remote-6 (0x086bd7fffe599a4a): E1766 - IKEA TRADFRI open/close remote (EndDevice)
```

[81 more replies]

closed 2020-07-26 by stale[bot]