Koenkk/zigbee2mqtt #895: Gledopto bulb GL-B-007Z paired but doesn't appear in the integraton page

milakov, 2019-01-20:
Hi, I am using zigbee2mqtt hass.io addon version 1.0.1.
I've just got Gledopto GL-B-007Z bulb and cannot pair it fully. Here is my custom config:
```
    // Gledopto
    {
        zigbeeModel: ['GL-B-007Z'],
        model: 'GL-B-007Z',
        vendor: 'GLEDOPTO',
        description: 'Dual-color Light LED Bulb 6W',
        supports: generic.light_onoff_brightness().supports,
        fromZigbee: generic.light_onoff_brightness().fromZigbee,
        toZigbee: generic.light_onoff_brightness().toZigbee,
    },
```

zigbee2mqtt adds the device to zigbee network (I can see it as a router in the map of the network), but no new entry can be seen at the integration page, no new HA entities as well. Here is the log:
```
zigbee2mqtt:info 2019-1-19 21:16:36 Device incoming...
  zigbee2mqtt:info 2019-1-19 21:16:36 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2019-1-19 21:16:36 New device with address 0x00124b001b50416b connected!
  zigbee2mqtt:info 2019-1-19 21:16:36 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00124b001b50416b"}'
  zigbee2mqtt:debug 2019-1-19 21:16:36 Received zigbee message of type 'devStatus' with data '"online"' of device 'GL-B-007Z' (0x00124b001b50416b)
```

Nothing changes if I set ```fromZigbee``` and ```toZigbee``` to ```[]```.

Will be happy to get any ideas!
Thanks!

BudBundi, 2019-01-20:
The Gledopo GL-B-007Z is already compatible in the dev branch but you must edit your database.db see: [https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/getting_started/pairing_devices.md](url)

milakov, 2019-01-20:
Interestingly, my bulb reports its model fine, exactly ```GL-B-007Z```. I guess, I don't have to edit the database then.
I am checking supported devices at dev branch and I don't see GL-B-007Z in there. Bigger bulb (008) is there, but not 007.

khzd, 2019-01-20:
Hi see #751
There is a section more to configure it 
Cheerd

zipbee, 2019-01-21:
Hi,

I've recently purchased a GL-C-007 as well and am having issues getting it to work, when first paired I get the following log entries:
`2019-1-21 18:39:08 - info: Device incoming...`
`2019-1-21 18:39:08 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'`
`2019-1-21 18:39:08 - warn: Device with modelID 'undefined' is not supported.`
`2019-1-21 18:39:08 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html`
`2019-1-21 18:39:08 - warn: Device with modelID 'undefined' is not supported. `
`2019-1-21 18:39:08 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html`

And I end up with the following entries in my database.db:
`{"id":6,"type":"Router","ieeeAddr":"0x00124b001ada63c8","nwkAddr":17051,"manufId":0,"epList":[11,12,13,15],"status":"online","joinTime":1548095736,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{}},"15":{"profId":49246,"epId":15,"devId":256,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}}},"_id":"BfeqOslPJkzS78N3"}`
`{"id":6,"type":"Router","ieeeAddr":"0x00124b001ada63c8","nwkAddr":17051,"manufId":0,"epList":[11,12,13,15],"status":"online","joinTime":1548095736,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}},"15":{"profId":49246,"epId":15,"devId":256,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}}},"_id":"BfeqOslPJkzS78N3"}
`

Mine is supposed to be a RGBW controller, and I think it should show up as 2 separate lights, one to control the RGB set of lights and one to control the (warm) white lights, at least that is what it suggests would happen when linked to a Hue Bridge.

My database entries look very similar to the ones from original post, but my epList also has "15" in it.

Any suggestions for how to try and get this to be recognised as 2 lights?

Thanks in advance!

milakov, 2019-01-21:
Hi!
Thanks for the link, I got it working!
First, I switched to dev branch. Then I added GL-B-007Z entry to custom devices.js, basically just copied all the content from 008Z bulb:
```
    {
        zigbeeModel: ['GL-B-007Z'],
        model: 'GL-B-007Z',
        vendor: 'Gledopto',
        description: 'Smart 6W E27 RGB / CW LED bulb',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
        ep: (device) => {
            if (device.epList.toString() === '11,12,13') {
                return {'': 12};
            } else if (device.epList.toString() === '10,11,13' || device.epList.toString() === '11,13') {
                return {'': 11};
            } else {
                return {};
            }
        },
    },
```

I checked database.db, it already contained all the manufacturer related info in my case.

And then I manually added entity in the configuration.yaml:
```
light
- platform: "mqtt"
  name: "Gledopto GL-B-007Z"
  state_topic: "zigbee2mqtt/0x00124b001b50416b"
  availability_topic: "zigbee2mqtt/bridge/state"
  brightness: true
  color_temp: true
  xy: true
  schema: "json"
  command_topic: "zigbee2mqtt/0x00124b001b50416b/set"
```

And this worked! For some reason MQTT discovery don't work for this bulb for me.

It probably makes sense to add GL-B-007Z to the devices.js, dev branch.

Koenkk, 2019-01-21:
Can you make a pr? Once its integrated discovery can be supported.

milakov, 2019-01-22:
Will do!

milakov, 2019-01-23:
Sorry, it is taking a little more time than I expected to get formalities done at my work to make a contribution to the open source project (even if it is a tiny contribution!)

milakov, 2019-01-29:
Here is the PR!
https://github.com/Koenkk/zigbee-shepherd-converters/pull/251

closed 2019-01-29 by Koenkk

milakov, 2019-01-29:
> Can you make a pr? Once its integrated discovery can be supported.

Do you want me to create a PR for the automatic HA discovery as well?

Koenkk, 2019-01-29:
Already done that, thanks anyway! https://github.com/Koenkk/zigbee2mqtt/commit/7e74b8c5497c6f318745e7f91af8d7f48fc73f57

milakov, 2019-01-29:
Wonderful, thank you so much!