Koenkk/zigbee2mqtt #896: Cannot get the Node Descriptor of the Device: 0x8418260000da12f5 (Error: Timed out after 10000 ms)

omer1141, 2019-01-20:
Hello,

I'm on zigbee2mqtt ver 1.0.1 and coordinator firmware 20180815.

I have troubles getting my new OSRAM Flex RGBW Strip (Router) recognized by the network (cannot get Node Descriptor of the Device....). See error in the title.

At the beginning I moved the LED Flex closer to the coordinator and then I could get it recognized as a new device, however, there is no communication happening between the LED strip and the zigbee network as soon as I moveed it further away from the coordinator. It is about 4m away from one of the Xiaomi-Routers (I use Xiaomi single key wired wall switch as my routers).

I already have 11 other devices working without any problems in my network:

- QBKG11LM - Xiaomi Aqara single key wired wall switch (Router)
- QBKG11LM - Xiaomi Aqara single key wired wall switch (Router)
- QBKG11LM - Xiaomi Aqara single key wired wall switch (Router)
- QBKG11LM - Xiaomi Aqara single key wired wall switch (Router)
- RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
- RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
- RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
- RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
- SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
- RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)

Any suggestions about what I'm doing wrong here?

Koenkk, 2019-01-21:
I would advise adding a non xiaomi router (e.g. CC2531 flash with router firmware). Perhaps the OSRAM strip refuses to connect with the Xiaomi router. You can also try to repair the device when having it close to the router you want to connect it to.

WagnerBeccard, 2019-02-07:
This info just saved my sorry ***!
Broke my first human body sensor due to extensive pairing. 
Vote for including it to the [docs](https://koenkk.github.io/zigbee2mqtt/getting_started/pairing_devices.html).

closed 2019-04-15 by stale[bot]