Koenkk/zigbee2mqtt #897: Philips motion sensor LED off option

michel72, 2019-01-20:
Feature request:

The Philips motion sensor has this bright red LED that usually does not blink on motion.
When pairing it with zigbee2mqtt it does show the red led on motion.
 
Would it be possible to have this a as an option (default off)?

I'm running a CC2530 coordinator firmware 28-11-2018 and zigbee2mqtt 1.0.1 on hass.io

If so, consider this a feature request :)

Koenkk, 2019-01-20:
Is this a configurable option when using it together with the hue bridge? If so this can be supported when we know the commands being send (by sniffing zigbee traffic of the hue bridge)

michel72, 2019-01-20:
As far as I know the LED is not supposed to blink at all on motion, but only on the pairing process or if that fails. https://www2.meethue.com/nl-nl/ondersteuning/motion-sensor/top/where-is-the-led-indicator-of-the-hue-motion-sensor-located-and-what-does-it-indicate

michel72, 2019-01-20:
Maybe this helps: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/20#issuecomment-322221174

Koenkk, 2019-01-21:
Could you try adding

```js
                (cb) => device.foundation('genBasic', 'write', [{attrId: 51, dataType: 16, attrData: false}], cb),
```

to https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L766

michel72, 2019-01-21:
I would, but I am on hass.io and I don't know how to modify this file on that.

michel72, 2019-01-22:
Had to search a little and since I'm a little more the "nano" kind of guy, that was a little tricky ;)
It's line 716 in my version 1.01 and I added it.

[image]

Do I need to do anything else, because the RED light is still blinking on motion?

Koenkk, 2019-01-22:
Can you post the startup log (20 first seconds)?

michel72, 2019-01-22:
I'm not sure I follow you. When I restart zigbee2mqtt it restarts the docker container and that removes my change.

In the log are all my devices. Can I post that publicly?

michel72, 2019-01-23:
As I am just a average user and not a coder or expert, I don't know how much further I can help.

I managed to save the setting by restarting the container first and then zigbee2mqtt, but I'm not sure what log information you are looking for and I am not comfortable posting all my ID's publicly.

Koenkk, 2019-01-23:
@michel72 On the beginning of the log you should see a message: `Successfully configured....` or `Failed to configure`  for this device (you can recognize it by the device ID).

michel72, 2019-01-23:
zigbee2mqtt:info 2019-1-23 16:47:13 0x001788010213XXXX (0x001788010213XXXX): 9290012607 - Philips Hue motion sensor (EndDevice)

zigbee2mqtt:info 2019-1-23 16:47:25 Succesfully configured 0x001788010213XXXX 0x001788010213XXXX

michel72, 2019-01-23:
The sensor works, but the LED (still) lights up RED on motion.

Koenkk, 2019-01-23:
Can you try with 
```js
                (cb) => device.foundation('genBasic', 'write', [{attrId: 51, dataType: 16, attrData: true}], cb),
```

michel72, 2019-01-23:
I have changed it to:

`        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 2);

            const actions = [
                (cb) => device.foundation('genBasic', 'write', [{attrId: 51, dataType: 16, attrData: true}], cb),,
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.bind('msIlluminanceMeasurement', coordinator, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.bind('msOccupancySensing', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 1000, 0, cb),
                (cb) => device.report('msOccupancySensing', 'occupancy', 0, 600, null, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 1, cb),
                (cb) => device.report('msIlluminanceMeasurement', 'measuredValue', 0, 600, null, cb),
            ];
`

restarted the docker container, but the LED still lights up RED on motion.

Is the procedure I use correct?

Make the change, restart the container and test on motion?

I do NOT restart zigbee2mqtt in hass.io.

Koenkk, 2019-01-24:
Yes, you need to make the change and restart zigbee2mqtt. Can you confirm that the changes are still there after restarting zigbee2mqtt?

michel72, 2019-01-24:
No, they are not.

What I do is:
Enter the docker container with:
`docker exec -it [some-id] bash`
make the change and save it using VI
Restart the docker container with:
`docker restart [some-id]`
check if the change is still there [confirmed]
Restart zigbee2mqtt,
enter the container
check the change [failed]

Koenkk, 2019-01-26:
Please see [Persisting a custom devices.js](https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices_on_hassio.html)

michel72, 2019-01-28:
Sorry, I totally overlooked that one. 

I tried both lines, followed by a restart.

The first one does not work.

The second one, does not work either, but troughs this warning:

`  zigbee2mqtt:warn 2019-1-28 09:39:53 No converter available for '9290012607' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips","modelId":"SML001","dateCode":"20160630","powerSource":3,"swBuildId":"6.1.0.18912"}}'`

Koenkk, 2019-01-29:
Do you see successfully configured for both?

michel72, 2019-01-29:
`                (cb) => device.foundation('genBasic', 'write', [{attrId: 51, dataType: 16, attrData: false}], cb),`

```
  zigbee2mqtt:info 2019-1-29 15:36:16 0x001788010213f612 (0x001788010213f612): 9290012607 - Philips Hue motion sensor (EndDevice)
  zigbee2mqtt:info 2019-1-29 15:36:16 0x001788010213f612 (0x001788010213f612): 9290012607 - Philips Hue motion sensor (EndDevice)  zigbee2mqtt:info 2019-1-29 15:36:16 MQTT publish: topic 'homeassistant/binary_sensor/0x001788010213f612/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/0x001788010213f612","availability_topic":"zigbee2mqtt/bridge/state","name":"0x001788010213f612_occupancy","unique_id":"0x001788010213f612_occupancy_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010213f612","name":"0x001788010213f612","sw_version":"Zigbee2mqtt 1.0.1","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"}}'
  zigbee2mqtt:info 2019-1-29 15:36:16 MQTT publish: topic 'homeassistant/sensor/0x001788010213f612/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/0x001788010213f612","availability_topic":"zigbee2mqtt/bridge/state","name":"0x001788010213f612_temperature","unique_id":"0x001788010213f612_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010213f612","name":"0x001788010213f612","sw_version":"Zigbee2mqtt 1.0.1","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"}}'
  zigbee2mqtt:info 2019-1-29 15:36:16 MQTT publish: topic 'homeassistant/sensor/0x001788010213f612/illuminance/config', payload '{"unit_of_measurement":"lx","device_class":"illuminance","value_template":"{{ value_json.illuminance }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/0x001788010213f612","availability_topic":"zigbee2mqtt/bridge/state","name":"0x001788010213f612_illuminance","unique_id":"0x001788010213f612_illuminance_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010213f612","name":"0x001788010213f612","sw_version":"Zigbee2mqtt 1.0.1","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"}}'
  zigbee2mqtt:info 2019-1-29 15:36:16 MQTT publish: topic 'homeassistant/sensor/0x001788010213f612/battery/config', payload '{"device_class":"battery","value_template":"{{ value_json.battery }}","json_attributes":["linkquality","voltage","action","sensitivity"],"state_topic":"zigbee2mqtt/0x001788010213f612","availability_topic":"zigbee2mqtt/bridge/state","name":"0x001788010213f612_battery","unique_id":"0x001788010213f612_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010213f612","name":"0x001788010213f612","sw_version":"Zigbee2mqtt 1.0.1","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"}}'
  zigbee2mqtt:error 2019-1-29 15:36:46 Failed to configure 0x001788010213f612 0x001788010213f612
```

--------------

`                (cb) => device.foundation('genBasic', 'write', [{attrId: 51, dataType: 16, attrData: true}], cb),
````
  zigbee2mqtt:info 2019-1-29 14:50:52 0x001788010213XXXX (0x001788010213XXXX): 9290012607 - Philips Hue motion sensor (EndDevice)
  zigbee2mqtt:info 2019-1-29 14:50:52 MQTT publish: topic 'zigbee2mqtt/0x001788010213XXXX', payload '{"battery":97,"linkquality":0,"temperature":13.49,"illuminance":12598,"occupancy":false}'
  zigbee2mqtt:info 2019-1-29 14:50:52 MQTT publish: topic 'homeassistant/binary_sensor/0x001788010213XXXX/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/0x001788010213XXXX","availability_topic":"zigbee2mqtt/bridge/state","name":"0x001788010213XXXX_occupancy","unique_id":"0x001788010213XXXX_occupancy_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010213XXXX","name":"0x001788010213XXXX","sw_version":"Zigbee2mqtt 1.0.1","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"}}'
  zigbee2mqtt:info 2019-1-29 14:50:52 MQTT publish: topic 'homeassistant/binary_sensor/0x001788010213XXXX/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/0x001788010213XXXX","availability_topic":"zigbee2mqtt/bridge/state","name":"0x001788010213XXXX_occupancy","unique_id":"0x001788010213XXXX_occupancy_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010213XXXX","name":"0x001788010213XXXX","sw_version":"Zigbee2mqtt 1.0.1","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"}}'
  zigbee2mqtt:info 2019-1-29 14:50:52 MQTT publish: topic 'homeassistant/sensor/0x001788010213XXXX/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/0x001788010213XXXX","availability_topic":"zigbee2mqtt/bridge/state","name":"0x001788010213XXXX_temperature","unique_id":"0x001788010213XXXX_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010213XXXX","name":"0x001788010213XXXX","sw_version":"Zigbee2mqtt 1.0.1","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"}}'
[... cut ...] age","action","sensitivity"],"state_topic":"zigbee2mqtt/0x001788010213XXXX","availability_topic":"zigbee2mqtt/bridge/state","name":"0x001788010213XXXX_battery","unique_id":"0x001788010213XXXX_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010213XXXX","name":"0x001788010213XXXX","sw_version":"Zigbee2mqtt 1.0.1","model":"Hue motion sensor (9290012607)","manufacturer":"Philips"}}'
  zigbee2mqtt:error 2019-1-29 14:51:22 Failed to configure 0x001788010213XXXX 0x001788010213XXXX
```

Koenkk, 2019-01-29:
Could you update to the latest dev branch and try again (-edge for hassio). This includes logging why configure failed.

michel72, 2019-01-29:
Could you give me a little hint about how risky that would be? This is an "production environment" for me as I actively use all that stuff in my house ;)

If the risk is "low" I'm more then willing to test.

Koenkk, 2019-01-29:
@michel72 you can use zigbee2mqtt 1.1 (just released)

michel72, 2019-01-30:
I'm afraid 1.1 is not yet available for hass.io.

I'll test right after it's available and post the logs.

michel72, 2019-01-31:
I have tried both changes with 1.1.0, but no more errors, In both cases:
zigbee2mqtt:info 2019-1-31 20:12:32 Succesfully configured 0x001788010213XXXX (0x001788010213XXXX)

But the LED still blinks on motion ;)

Koenkk, 2019-01-31:
Can you post the log when running with 
```js
                (cb) => device.foundation('genBasic', 'write', [{attrId: 51, dataType: 16, attrData: true}], cb),
                (cb) => device.foundation('genBasic', 'read', [{attrId: 51}], cb),
```

michel72, 2019-01-31:
```
  zigbee2mqtt:warn 2019-1-31 21:00:43 No converter available for '9290012607' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"51":null}}'
  zigbee2mqtt:warn 2019-1-31 21:00:43 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 2019-1-31 21:00:49 Succesfully configured 0x001788010213XXXX(0x001788010213XXXX)
```

michel72, 2019-02-02:
Hi Koen,

Anything else I could try?

michel72, 2019-02-02:
I have a hue bridge with several motion sensors on it. I should be able to sniff, probably won't be able to analyse the data ;)

Koenkk, 2019-02-02:
If you can post a screenshot of wireshark I can help with this. Please capture the command when the behaviour of the LED is changed.

michel72, 2019-02-02:
The problem is that the hue app does not allow the led to be controlled on or off. The default behavior is ALWAYS OFF, even on motion.

I have 2 full captures: One of the pairing process and a capture of detecting motion.

I just don't know what to look for and since you need my keys to decrypt I can't post them publicly. I am willing to send them to you privately though.

Koenkk, 2019-02-02:
@michel72 if you filter on `write` commands to the basic cluster, how much messages are left?

michel72, 2019-02-02:
I'm quit a noob on the wireshark front. Can you give me the filter rule?

michel72, 2019-02-02:
If you are looking for the string "write" in the info column, there is none.

[image]

Koenkk, 2019-02-03:
Could you filter based on the cluster? `genBasic`, cluster ID: `0x00`.

michel72, 2019-02-03:
I know little about the zigbee protocol and little about zigbee sniffing. If you could only give me the filter rule like "zbee_zcl.cmd.id == 0x00" I can send you the results.

When I apply the above filter I find al sorts of read attributes, but I really don't know what you are looking for.

michel72, 2019-02-03:
[image]

michel72, 2019-02-03:
[image]

This?

Koenkk, 2019-02-04:
@michel72 yes that's the correct cluster, beside the bind requests, do you see any other messages of this cluster?

michel72, 2019-02-04:
[image]

michel72, 2019-02-04:
[image]

michel72, 2019-02-04:
Besides current level and Lux nothing else...

Koenkk, 2019-02-04:
So it seems that `device.foundation('genBasic', 'write', [{attrId: 51, dataType: 16, attrData: true}], cb),` is not done by the Hue bridge. Could you perhaps share the wireshark file with me so I can take a look?

michel72, 2019-02-04:
Sure, where can I send it?

y0ast, 2019-02-13:
I'm also interested in fixing this. Let me know if I can be of help for testing. Running zigbee2mqtt straight from source (on hassbian).

michel72, 2019-02-14:
Can you test this:

add this attribute:

`"ledindication":{"id":51,"type":"boolean"}},`

to:
https://github.com/Koenkk/zcl-id/blob/master/definitions/cluster_defs.json#L18

and then test these:

https://github.com/Koenkk/zigbee2mqtt/issues/897#issuecomment-459477849

Koen asked me to test this on hass.io, but that is not possible since the changes are gone after a restart of zigbee2mqtt.

Please post your results here.

y0ast, 2019-02-18:
Updated to latest master with commit `92d88b6`

step 1: add `"ledindication":{"id":51,"type":"boolean"}},` after L18 of `node_modules/zcl-id/definitions/cluster_defs.json` (and remove one of two `}` of line before)

step 2: add:
```
                (cb) => device.foundation('genBasic', 'write', [{attrId: 51, dataType: 16, attrData: true}], cb),
                (cb) => device.foundation('genBasic', 'read', [{attrId: 51}], cb),
```
after [L774](https://github.com/Koenkk/zigbee-shepherd-converters/blob/2606e4710e1ec4b152c60852c7e0de767abd283d/devices.js#L774): ` node_modules/zigbee-shepherd-converters/devices.js` (node pulled in version 7.0.27)

Step 3: start `zigbee2mqtt` using `npm start` and see following in log:

```
  zigbee2mqtt:warn 2/18/2019, 9:05:16 PM No converter available for '9290012607' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"51":null}}'
  zigbee2mqtt:warn 2/18/2019, 9:05:16 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

Also the red light still shows.

@michel72 @Koenkk Let me know if I can try anything else or if you spot any issue with my setup.

y0ast, 2019-02-25:
I did some more poking around (and reading about the Zigbee protocol).

I've tried binding first to cluster 0x0000:
```
                (cb) => device.bind('genBasic', coordinator, cb),
```

I've tried specifically setting the `usertest` attribute:

```
                (cb) => device.foundation('genBasic', 'write', [{attrId: 50, dataType: 16, attrData: false}], cb),
                (cb) => device.foundation('genBasic', 'write', [{attrId: 51, dataType: 16, attrData: false}], cb),
                (cb) => device.foundation('genBasic', 'read', [{attrId: 51}], cb),
```
I've played with just setting `false` the attribute false and not reading it out again, perhaps it's a write only. But no, red light remains.

I dug into how deCONZ does this and found this reference: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/d976ed33dfe4f27de33869b248f826de8aff2c10/bindings.cpp#L1267

However it doesn't seem that much special is going on. Just an attribute write.

I read this analysis of a data dump: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Philips-Hue-sniffing#hue-motion-sensor but also nothing special.

The red light simply disappears the moment you connect the sensor to the hue hub (it comes back on briefly if you configure the sensor in the app, to aid with setting the sensitivity correctly).

I'm a bit lost for options here. Curious to hear your opinion @michel72 and @Koenkk. If we figure this out then there are a number of other attributes on the sensor that are worth exposing (duration and sensitivity).

Koenkk, 2019-02-26:
Can you try with:

```js
(cb) => device.foundation(
  'genBasic', 
  'write', 
  [{attrId: 51, dataType: 16, attrData: false}], 
  {manufSpec: 1, manufCode: 0x100B}, 
  cb,
),
```

y0ast, 2019-02-27:
Thanks @Koenkk! The attribute can now be set/read, but weirdly the led continues to be on with movement, regardless of the setting 0 or 1.

Output of first setting then reading (both using manufCode):

```
  zigbee2mqtt:warn 2/27/2019, 7:46:55 PM No converter available for '9290012607' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"51":0}}'
  zigbee2mqtt:warn 2/27/2019, 7:46:55 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 2/27/2019, 7:46:55 PM Succesfully configured 0x0017880102112db2 (0x0017880102112db2)
```

With `attrData: True`:

```
  zigbee2mqtt:warn 2/27/2019, 7:51:59 PM No converter available for '9290012607' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"51":1}}'
  zigbee2mqtt:warn 2/27/2019, 7:51:59 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 2/27/2019, 7:51:59 PM Succesfully configured 0x0017880102112db2 (0x0017880102112db2)
```

I can also read `usertest`, but not `sensitivity`:

```
(cb) => device.foundation(
  'msOccupancySensing', 
  'read', 
  [{attrId: 48}], 
  {manufSpec: 1, manufCode: 0x100B}, 
  cb,
),
```

Gets ignored, I think I'm not setting the cluster id quite right.

Koenkk, 2019-02-28:
The read looks good, can you try to write it? (should set sensitivity to high).
```js
(cb) => device.foundation(
  'msOccupancySensing', 
  'write', 
  [{attrId: 48, dataType: 32, attrData: 0x02}], 
  {manufSpec: 1, manufCode: 0x100B}, 
  cb,
),
```

MrLight, 2019-03-04:
just a short note from my tests:
By setting the usertest bit to true, I can switch the led from red to green.

`(cb) => device.foundation(
  'genBasic',
  'write',
  [{attrId: 50, dataType: 16, attrData: false}],
  {manufSpec: 1, manufCode: 0x100B},
  cb,
),`

By setting both to true the led is green, by setting both to false the led is red.
Out of my tests the value of the ledindication has no effect to the led at all...
It would be great if someone can try to sniff the sensor in a hue enviroment. You can switch on the ledindication when you start the setup process of the sensitifity in the hue app...

y0ast, 2019-03-04:
I can verify the green led light too with usertest set to true.

It seems like the first occupancy does not blink the led, but every subsequent one does.

There is a peculiar section in the deconz code: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/d976ed33dfe4f27de33869b248f826de8aff2c10/bindings.cpp#L1744

which seems to do some minimal delay, but I don't fully understand it.

@Koenkk even after your suggested write, the read of sensitivity just gets ignored. Configuring is successful though.

Koenkk, 2019-03-05:
@y0ast is it possible to set the sensitivity from the Hue hub?

y0ast, 2019-03-05:
Yep, when my motion sensor was still paired to the hub I was able to adjust
the sensitivity from the hue app. I was also able to empirically verify
that the sensitivity was different.

Koenkk, 2019-03-05:
Could you sniff that? We need to find the difference with what zigbee2mqtt sends. Please post 2 captures, one with the hue hub, one with zigbee2mqtt.

y0ast, 2019-03-05:
Unfortunately I only have one CC2531 stick (so can't sniff zigbee2mqtt itself) and I only have access to an OSX machine.

The how-to seem to indicate that it's only possible with ubuntu/windows:
https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

@michel72 could you perhaps help out with the sniffing?

Koenkk, 2019-03-05:
@y0ast it works well with vmware fusion (doing that on my macbook).

y0ast, 2019-03-06:
I've ordered a second CC2531, so will report back when it arrives (might be a few weeks).

MrLight, 2019-03-07:
HI Guys,
I spend this afternoon with some sniffing of my hue network and I have seen that the Hue Bridge is `Configure Reporting` all three values (ledindication, usertest and sensitivity).
After playing around I found a partly solution for our Problem:
I added the following to the device.js:
```
(cb) => device.foundation('genBasic', 'configReport', [{direction: 0, attrId: 51, dataType: 16, minRepIntval: 5,
maxRepIntval: 7200, repChange: 0}], {manufSpec: 1, manufCode: 0x100B}, cb),

(cb) => device.foundation('genBasic', 'configReport', [{direction: 0, attrId: 50, dataType: 16, minRepIntval: 5,
maxRepIntval: 7200, repChange: 0}], {manufSpec: 1, manufCode: 0x100B}, cb),
```
to get it running I added `fz.generic_state,` to `fromZigbee:`

For the sensitifity I haven't found a fromZigbee converter so I was not able to setup this too. 

Would be great if someone can verify my results.

[22 more replies]

closed 2019-10-03 by Koenkk