Koenkk/zigbee2mqtt #898: Device leaving not handled

neerdoc, 2019-01-20:
While debugging another issue, I noticed that when I reset (i.e., unpairing) my device, it sends a message to indicate this. 

`zigbee2mqtt:debug 1/20/2019, 1:00:02 PM Received zigbee message of type 'devLeaving' with data '"0x005043c91f292a8f"'`

This message is ignored, and the device is still in my database. I believe that it would be desirable that a device that sends a message that it is leaving the network should actually be removed permanently.

MarsWarrior, 2019-01-20:
I like this 'feature' 😃 

Once a year I connect my Ikea bulbs to the Tradfri gateway for firmware updates. As zigbee2mqtt ignores the reset/unpair, all my lights are still reported to Home Assistant, including the friendly name etc.

So if zigbee2mqtt would implement the devLeaving, could it made configurable as in remove or keep the device?

Koenkk, 2019-01-21:
If I understood correctly the device is still in de `database.db` but contains a key which indicates that the device has been deleted. Can you post your `database.db` after the `devLeaving`?

neerdoc, 2019-01-21:
database before leaving: [https://pastebin.com/cTXwhKNB](url)
database after leaving: [https://pastebin.com/wuCJFPzB](url)

sehraf, 2019-01-21:
out of curiosity: would compact (see #908) maybe remove obsolete db entries?

Koenkk, 2019-01-23:
@neerdoc you see that the device is marked as deleted:

```
{"$$deleted":true,"_id":"EaN4nFZLM41XrrrC"}
```

neerdoc, 2019-01-23:
Aha, I understand. Guess it is a matter of expectation then!

I would have expected a deleted device to be removed completely, i.e., not a trace left. But I can see that one could want it to be kept as well when temporarily disconnecting.

My suggestion is a configuration flag or a button to "cleanup" the database (which seems possible with the compacting suggestion).

neerdoc, 2019-01-29:
After testing some more, it seems that at least for my HS1SA smoke detectors, this "deleted" functionality does not work. If I reset a device it shows up in the database as "deleted", but it fails to rejoin unless I clear it from the database. Reconnecting it looks like it is working, no error message or anything, but the device does not work anymore.

Koenkk, 2019-01-30:
Can you check if this works in the latest dev branch? Device should be removed from `database.db` on `devLeaving`.

neerdoc, 2019-01-30:
Works!

It gets removed from the database during startup it seems. Which means that we also get the full functionality:
1. If you want to keep the data in the database: don't restart before rejoining the device.
2. If you want to remove the device completely: unlink, restart and it's gone.

Awesome!

Koenkk, 2019-01-31:
Great, thanks for the help again!

closed 2019-01-31 by Koenkk