Koenkk/zigbee2mqtt #900: Can't add more devices even after successful removal to free space

jarrah31, 2019-01-20:
I have 25 devices and one router on my network, with three being unknown, and I'm unable to add any more. I'm hoping someone is able to suggest some ideas to avoid me having to re-flash my coordinator and pair all the devices again please. My aim is to add another router to the network to help with expansion.

zigbee2mqtt version 1.0.1
Coordinator firmware version: '20181024'
2 x CC2531 as coordinator and router (using CC2531ZNP-Prod.hex and router-cc2531-std.hex)

Debug startup logs to show details: https://hastebin.com/voraribaco.coffeescript
configuration.yaml: https://hastebin.com/uzaragacus.js
Database.db: https://hastebin.com/folagatoxu.json

I guess my first question is, how do I remove those unknown devices as they may be preventing me adding new ones? I see that all three unknowns appear twice in the database.db file - is it safe/ok to just remove those lines? Note that they don't appear in the configuration.yaml.

I originally had 27 devices when I tried to add an extra router, but as I couldn't pair it I removed a couple of existing ones using the following which seemed to work ok:

```
mosquitto_pub -V mqttv311 -u $mqtt_broker_user -P $mqtt_broker_pass -h $mqtt_broker_ip -m "0x00158d0001ab406f" -t zigbee2mqtt/bridge/config/remove

Jan 20 10:49:25 zigbeepi npm[31738]:   zigbee2mqtt:info 2019-1-20 10:49:25 Successfully removed 0x00158d00025298d3
Jan 20 10:49:25 zigbeepi npm[31738]:   zigbee2mqtt:info 2019-1-20 10:49:25 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0x00158d00025298d3"}'
Jan 20 10:50:14 zigbeepi npm[31738]:   zigbee2mqtt:info 2019-1-20 10:50:14 Successfully removed 0x00158d0001ab406f
Jan 20 10:50:14 zigbeepi npm[31738]:   zigbee2mqtt:info 2019-1-20 10:50:14 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0x00158d0001ab406f"}'
Jan 20 11:12:03 zigbeepi npm[32414]:   zigbee2mqtt:info 2019-1-20 11:12:03 Successfully removed 0x00158d0002c227ef
Jan 20 11:12:03 zigbeepi npm[32414]:   zigbee2mqtt:info 2019-1-20 11:12:03 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"xiaomi_wallswitch_study"}'
```

However this hasn't helped and I still can't pair new devices. I appreciate any assistance!
Thanks!

jarrah31, 2019-01-20:
I found another post from @Koenkk saying that the unknown entries from database.db can be manually removed, so I'm now down to 22 device. 

Unfortunately though I'm still unable to add any more devices. Is there anything that can be done with this?

I've seen reports that some people have been able to update their firmware without having to pair devices again. I may flash my coordinator with the same firmware to see if that helps clear out the memory.

Koenkk, 2019-01-22:
When removing them from `database.db` they are not removed from the CC2531, so this will not give space for a new device. Devices should be removed by using `zigbee2mqtt/bridge/config/remove` (https://koenkk.github.io/zigbee2mqtt/information/mqtt_topics_and_message_structure.html). When removing end devices make sure that they are awake, otherwise they can not be removed successfully.

jarrah31, 2019-01-24:
Thanks for the tip, I'll try that next time (ended up re-pairing my devices, but at least I'm on the latest CC2531 firmware now).

closed 2019-01-24 by jarrah31