Koenkk/zigbee2mqtt #901: Can't pair IKEA LED1546G12

sturlig, 2019-01-20:
I'm unable to fully pair a IKEA LED1546G12. It is on the list of supported devices. Zigbee2MQTT does connect somehow, but the bulb is not available in MQTT.

Please see the logfile:
https://pastebin.com/PfNz1U32

Prior to the session in the logfile the bulb is paired (see line 1081, 1082).

What is going wrong?

sehraf, 2019-01-20:
i have two IKEA bulbs (different model though) and while one paired almost instantly the second one took me a half an hour of constantly resetting and checking the log.
As @mason-xx said, ensure that you see the short pulse.

sturlig, 2019-01-20:
I did reset the bulb multiple times (and saw the short pulse). Another LED1546G12 had the same problem, while a LED1649C5 directly worked after the reset.

sturlig, 2019-01-21:
Somehow the bulb paired. I can now control it. No idea why the bulb wasn't published to the MQTT in the first place.

closed 2019-01-21 by sturlig