Koenkk/zigbee2mqtt #902: using "brightness_percent" returns wrong state

andreasbrett, 2019-01-20:
Prerequisite:
 - zigbee2mqtt on branch DEV with commit a2ea9c548fb882b45030c1fbe496f5033d31ef43 (Jan 7, 2019 18:18:14)
 - paired IKEA TRADFRI bulb is OFF

Scenario #1 (working):
 - sending { "brightness": 204 } will...
 - set the bulb to brightness level 204 (of 255)
 - turn ON the light
 - return {"state":"ON","brightness":204,"last_seen":"2019-01-20T15:55:06.921Z"}

Scenario #2 (NOT working):
 - sending { "brightness_percent": 80 ] will...
 - set the bulb to brightness level 204 (of 255)
 - turn ON the light
 - return {"state":"OFF","brightness":204,"last_seen":"2019-01-20T15:55:45.957Z"}

Could you please check why this is happening? The bulb turns on in both cases but through MQTT a "state  = OFF" is reported when using the new "brightness_percent" attribute.

andreasbrett, 2019-01-20:
Just tested with latest DEV commit e21bfc156790515d5244f95f3aec75e4d019bcaf but still the same issue.

Koenkk, 2019-01-21:
Can you check if this has been fixed in the dev branch?

andreasbrett, 2019-01-21:
Yes, this is fixed now.

closed 2019-01-21 by andreasbrett

Koenkk, 2019-01-21:
Thanks for reporting!