Koenkk/zigbee2mqtt #903: Discussion -> implement basic rules without mqtt

svrooij, 2019-01-20:
How do you guys think about having some support for basic rules? So your switches still work even it mqtt, node red or some other app is down.

E.g.:
- if button x pressed toggle light y
- if switch v toggled reflect state on light w

Would that speed up responsiveness of switches? And how would you guys notate it in the configuration?

And second question, what rule engine do you guys use for the complicated stuff?

Vout, 2019-01-20:
See https://github.com/Koenkk/zigbee2mqtt/issues/782

People are already working on it ;)

svrooij, 2019-01-20:
The issue you are referring to is a feature in the signed protocol to connect a remote to a specific bulb.

What I'm referring is to have local rules for devices that don't support zigbee binding, like Xiaomi buttons, being able to control other devices. Without going through Mqtt -> node red (or other rule engine) -> Mqtt -> zigbee2mqtt.

And just have a simple rule in zigbee2mqtt.

MarsWarrior, 2019-01-21:
Are you referring to rules like [Tasmota](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules) has on ESPxxx devices?

It could be nice. On the other hand, zigbee has its own protocols for that (groups, bindings), and I would prefer to define those extra rules in some domotica system (Home Assistant or others) as a central place to define & manage those rules.

Given the fact that groups and bindings can be defined through mqtt topics, the same could be applied to rules: defined from another system, but executed locally by zigbee2mqtt.

As a Home Assistant user however, why would I not use Home Assisant for this? I understand the point of round trip delays,  but those are (on my simple NUC) very small.

closed 2019-03-29 by stale[bot]

gelsas, 2021-05-28:
I would love a feature like this!

Anonym-tsk, 2022-02-01:
Look at this
https://github.com/Anonym-tsk/zigbee2mqtt-extensions