Koenkk/zigbee2mqtt #904: send command to different endpoints in single device

msadeghz, 2019-01-21:
Hi friends,
I can't find the guide of how can I send "set" or "get" commands to different endpoints of a device. for example 2-gangs Aqara switch. can anyone help me? 
thank you.

Koenkk, 2019-01-21:
`zigbee2mqtt/[DEVICE_ID]/left/set` and `zigbee2mqtt/[DEVICE_ID]/right/set`

msadeghz, 2019-01-21:
thank you,
where can I find the complete guides?

Koenkk, 2019-01-22:
https://koenkk.github.io/zigbee2mqtt/information/mqtt_topics_and_message_structure.html

msadeghz, 2019-03-10:
I'm confused
I want to send command but {'state':'on'} not works, but just 'on' works
moreover I can't send command to the endpoints based on the way you said

device.js
[image]

toZigbee.js
[image]

when I publish "mosquitto_pub -t zigbee2mqtt/0x00124b0014dc270b/set -m {'state':'on'}"
[image]

when I publish "mosquitto_pub -t zigbee2mqtt/0x00124b0014dc270b/a/set -m {'state':'on'}"
[image]

Koenkk, 2019-03-10:
The postfix has to be in this list (otherwise it will not be parsed): https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/devicePublish.js#L7

Your ep function should e.g. look like:

```js
return {'top': 16, 'center': 17, 'bottom': 18};
```

msadeghz, 2019-03-10:
thank you @Koenkk , got it.
and how about `'state'` key. when I set the mqtt paylaod `{"state":"on"}` or` {"state":"on",}`, it doesn't work. when I set it `"on"`, it works.

Koenkk, 2019-03-10:
Can you post a more complete log of this?

closed 2019-05-16 by stale[bot]