Koenkk/zigbee2mqtt #905: the cluster list of connected device is not complete

msadeghz, 2019-01-21:
Hi guys,
I have a device and want to check its compatibility with zigbee2mqtt,
It can connect and pair to the network but the cluster list is not complete. you can check the file attached. 
the device send the alarm by " IASzoneChangeNotifications" which is not in the cluster list.
how can I solve the problem?

Koenkk, 2019-01-22:
`IASzoneChangeNotifications` is not a cluster, it should be part of the `ssIasZone` cluster.

msadeghz, 2019-01-22:
excuse me, cluster attributes, I've checked and is not in the attributes of the ssIasZone cluster. a more general question: If I have a device which is HA v1.2 profile compatible which is modified (for example extra attributes with different id and ...) is it possible to modify the firmware to support that device?

Koenkk, 2019-01-22:
I don't have any experience with this but I guess https://github.com/Koenkk/zcl-id/blob/master/definitions/cluster_defs.json has to be updated.

closed 2019-03-30 by stale[bot]