Koenkk/zigbee2mqtt #906: Questions on how and when to firmware update coordinators and routers

h4nc, 2019-01-21:
This isn't really an issue, but I don't know where to put this and I think I reach the right people here.

1) Do you regularly update your firmware of coordinators and routers? In which rhythm?

2) Are the linked devices linked to the coordinator it self (hardware) or to the zigbee addon (software)?
So after an update, do I have to relink all those devices again? Would it be possible to switch to another updated coordinator, so just change it to try if the new firmware works as expected.

3) I think some of us do have that questions. If yes, it would be nice to add this to the docs. "How to update Coordinator and Routers". It should contain information on how often this should be done and maybe an information box when a mejor update is released. This could be linked to an rss feas, to remind people to update.

What do you think @Koenkk ?

Koenkk, 2019-01-22:
1. Depends, I would only recommend doing a firmware upgrade when you experience stability issues or if you want a new feature (as of now, the only feature that requires a firmware update is group support).
2. Devices are linked to the coordinator, however it's possible to rebuild your network when the panID doesn't change. This can be done by starting the coordinator one time when having it out of the network range.

h4nc, 2019-01-22:
Could you add a link or explain "group support".

> This can be done by starting the coordinator one time when having it out of the network range.
I don't understand. So technically I take the coordinator with my raspi away from my house and than I plug it in?

Landrash, 2019-01-22:
@h4nc https://github.com/Koenkk/zigbee2mqtt/issues/764

Bruceforce, 2019-02-07:
> 2. Devices are linked to the coordinator, however it's possible to rebuild your network when the panID doesn't change. This can be done by starting the coordinator one time when having it out of the network range.

@Koenkk I'm currently writing a PR to improve the docs with this. Can you elaborate a bit more on this? Is it sufficient if you flash the coordinator out of range (on a different system without zigbee2mqtt) or do you have to start zigbee2mqtt with the flashed coordinator out of range once?
It would also be nice if someone could link me to a place (or describe briefly) why devices need to be re-paired when flashing "normally".

h4nc, 2019-02-07:
I once flashes my coordinator with a new firmware. The connected devices still worked, so no re-pairing was necessary.
However I changed the network key afterwards, than re-pairing was necessary.

Koenkk, 2019-02-07:
> > 1. Devices are linked to the coordinator, however it's possible to rebuild your network when the panID doesn't change. This can be done by starting the coordinator one time when having it out of the network range.
> 
> @Koenkk I'm currently writing a PR to improve the docs with this. Can you elaborate a bit more on this? Is it sufficient if you flash the coordinator out of range (on a different system without zigbee2mqtt) or do you have to start zigbee2mqtt with the flashed coordinator out of range once?
> It would also be nice if someone could link me to a place (or describe briefly) why devices need to be re-paired when flashing "normally".

- Coordinator needs to be started with zigbee2mqtt once, outside the network range
- When this is not done ('normally'), the coordinator will use  a different `pan_id` because the old `pan_id` is already in use by the old network (e.g. bulbs are still powered = zigbee network). Different `pan_id` = repair.

h4nc, 2019-02-07:
To the second point: if I have the old pan_id than I can simple add it to the advances section right?
So in this case it’s not needed to power it up somewhere?

Koenkk, 2019-02-07:
> To the second point: if I have the old pan_id than I can simple add it to the advances section right?
> So in this case it’s not needed to power it up somewhere?

No, Zigbee will **never** use a duplicate `pan_id`. When you specify a `pan_id` which is already used, a new `pan_id` will be picked. 

In most cases people use the default `pan_id` `0x1a62`. When they start a new flashed coordinator within the network range, it will result in `pan_id` `0x1a63`.

h4nc, 2019-02-07:
> When they start a new flashed coordinator within the network range, it will result in `pan_id` `0x1a63` (= configured `pan_id` + 1)

That happened with mine. It's still hard for me to understand the whole thing arround the pan_id.
Would this mean if I turn of alle my zigbee devices (even putting out the battery) and than start a newly flashed coord it will start again with 0x1a62? It only uses pan_id +1 if it sees a device with the 0x1a62. Is this right?

Bruceforce, 2019-02-07:
Would a re-pair require a reset of all devices (eg. pressing the button / toggle light 5 times...) Or simply allow new devices to join in Zigbee2mqtt?

Koenkk, 2019-02-07:
> Would this mean if I turn of alle my zigbee devices (even putting out the battery) and than start a newly flashed coord it will start again with 0x1a62? It only uses pan_id +1 if it sees a device with the 0x1a62. Is this right?

Yes, I don't think you need to remove batteries from device, you only need to turn of the routers.

Koenkk, 2019-02-07:
> Would a re-pair require a reset of all devices (eg. pressing the button / toggle light 5 times...) Or simply allow new devices to join in Zigbee2mqtt?

I don't understand your question.

h4nc, 2019-02-07:
So in other words you could add to the docs that you can do two things

- start the reflashed coord wide away from your network
- unplug (no power) your routers (hue bulbs, etc) enddevices are not needed.

If you do this, than you don't have to repair.

> I don't understand your question.
I think he means a "reflash" instead of re-pair.
From my experience repairing means reseting the devices and than add them again to your coord. So its not enough to simple turn on permit_join. You will also have to add every device manuelly.
However, this will not mess up the entity list in home assistant. 
Also you can keep the configuration.ymal file of zigbee2mqtt.
After repairing simple restart and everything will be as before.

Note that I got some Warning while I repaired some devices "message without device".
Waiting some seconds and restarting the addon fixed it.

Bruceforce, 2019-02-07:
Okay I'll try to make it more clear. 

Would the devices automatically re-pair if I set `permit_join: true` in `configuration.yaml` or do I have to reset all my devices if I flash "normally"

I'm asking because I think that I have once flashed the coordinator and then saw a lot of entries in the log with `new device` (or so).

Koenkk, 2019-02-07:
When you successfully kept your `pan_id` you need to set `permit_join: true`, after a few minutes most devices should be reconnected.

h4nc, 2019-02-09:
@Koenkk I have another question about pan_id so I think I have to botther you again on that one

Because I reflashed my coordinator to get a new firmware I got pan_id: 0x1a63 (so +1).
I asking myself what will happen when I update it the next time and follow the instructions (
https://koenkk.github.io/zigbee2mqtt/information/flashing_without_re-pairing.html)

So if I shield it or cut the power for the routers, will it get 0x1a63 again. Or will it use the default value 0x1a62?

Koenkk, 2019-02-09:
It will always try to use the configuted value first (so default 0x1a62). The next time you should set the pan_id to 0x1a63 in your configuration.yaml.

h4nc, 2019-02-09:
Ok thanks, i thought it might do this, so I asked you to be sure.

I already filled in the pan_id in advanced to ...63.
This way I don’t have to remember this when I update it again.

apeeters, 2019-02-09:
Some helpful addition might be the procedure to determine the current pan_id.

h4nc, 2019-02-10:
1) Set log_level to debug in your configuration. 
https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html

2) Restart zigbee2mqtt 
3) search for pan_id in your log

jarrah31, 2019-02-18:
I am curious to know how many people have had success using these instructions please?

I actually tried the method where you wrap the coordinator in kitchen roll and then tin foil around a month ago. I was able to successfully start Z2M and verify it had the same pan_id as before, but when I took the foil off and started it up again, my existing devices didn't reconnect. However I didn't give it chance to reconnect which is where it may have gone wrong...

Can I use the foil method to test a different coordinator with updated firmware, but then go back to the original in case it doesn't work? e.g. 

1. Shutdown Z2M
2. Wrap a second CC2531 (with the latest firmware) in kitchen roll / tin foil
3. Plug this into the same Pi and boot it up with Z2M
4. Verify it is still using the same pan_id via the debug log
5. Shutdown Z2M
6. Remove the foil 
7. Boot Z2M and wait to see if existing routers and devices re-connect
8. If after say 10 mins nothing had happened, shut down Z2M and remove new CC2531
9. Plug old CC2531 back into the Pi and boot
10. Will everything re-connect as it was again?

Will using different physical CC2531's cause issues with Z2M?

h4nc, 2019-02-18:
@jarrah31 this is not the answer to your question but maybe also interesting for you:
https://github.com/Koenkk/zigbee2mqtt/issues/1088#issuecomment-464471396

jarrah31, 2019-02-18:
Ignore what I wrote above, this looks very promising!
https://github.com/Koenkk/zigbee2mqtt/issues/1088

jarrah31, 2019-02-18:
Aha, sorry h4nc, my phone didn’t show your reply when I wrote mine. :)

h4nc, 2019-02-18:
I still like to know the answer your question above. 

@koenkk do you know that? This would be nice to know if your device gets faulty or you did a mod and destroyed something.

Koenkk, 2019-02-19:
@h4nc I've just checked:

In case you want to switch from coordinator hardware, nothing special has to be done, just flash it with the coordinator firmware and it works.

h4nc, 2019-02-19:
So no re-pairing if I flash a new coordinator?

If yes, no more questions. Can be closed.

jarrah31, 2019-02-19:
I would say yes given that the new firmware can be flashed without having to repair anyway

h4nc, 2019-02-19:
So all of enddevice data is stored in zigbee2mqtt? 

I thought maybe the enddevices look for a certain I’d of the coordinator, something like the mac adress

Koenkk, 2019-02-19:
@h4nc yes

To document this properly, I've written the following article: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/information/what_does_and_doesnt_require_repairing.md

closed 2019-02-19 by Koenkk

h4nc, 2019-02-19:
Great job @Koenkk

Thank you very much!

h4nc, 2019-02-20:
@Koenkk

However there is still the issue that the links in the zigbee map disappeared after a firmware update.

Only think I found to show the links again was to re-pair.

I don’t know if this is re-pairing, as the devices already worked.

permit join true
place enddevice near coord
press the button like repairing
repeat until “Message without device”
—> works again.
Other users also reported this issue (in ha community).

Should I open a new issue on that?

Koenkk, 2019-02-20:
@h4nc it's the same as #1001 right?

h4nc, 2019-02-20:
You are right, I forgot about that. Sry ;)