Koenkk/zigbee2mqtt #907: ARM Docker images are tagged as platform amd64

mkroman, 2019-01-21:
Hello,

Since the Docker images are built on an x86_64 platform, the images have an architecture field that says `amd64`, which prevents the image from being used in a Docker Swarm cluster with ARM nodes, as the manager sees them as incompatible.

```
% docker image inspect -f '{{ .RepoTags }} {{ .Architecture }}' koenkk/zigbee2mqtt:1.0.1-{arm64v8,arm32v6}
[koenkk/zigbee2mqtt:1.0.1-arm64v8 koenkk/zigbee2mqtt:arm64v8] amd64
[koenkk/zigbee2mqtt:1.0.1-arm32v6] amd64
```

Docker has an experimental option during build (`--platform`), but requires experimental features to be enabled.

After building the image myself on an ARM node I was able to create services in the swarm, but since docker swarm services don't support linking devices like docker run/docker-compose does, giving the container access to the serial device of the coordinator does not seem possible.

So until docker/swarmkit#1244 is fixed, this is very low priority.

Koenkk, 2019-01-21:
Thanks for reporting, once they above bug is fixed, https://github.com/Koenkk/zigbee2mqtt/blob/master/.travis/docker.sh should be updated.

amondit, 2019-01-25:
Tried the same thing, running Zigbee2MQTT in a docker swarm on a arm box (amlogic s812 tv box running armbian with mainline kernel). Got as well the incompatible platform error, tried as well to build locally and could run the image, but since passing through devices is not supported, I had to revert to the hacky way of passing devices, here is the stack I use in Portainer:

```
version: '3.2'
services:
  app:
    image: docker
    command: "docker run -v /docker/zigbee2mqtt/app/data:/app/data --name=zigbee2mqtt --device=/dev/ttyACM0 koenkk/zigbee2mqtt:arm32v6"
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock
```

mkroman, 2019-01-27:
Ah, cool, thanks for the hack, @amondit! I had no idea it could be that simple.

I've submitted a PR that should fix the tagging.

Koenkk, 2019-01-27:
Thanks both!

closed 2019-01-27 by Koenkk