Koenkk/zigbee2mqtt #911: Sensors do not respond to server

FRFrancisqueti, 2019-01-21:
Hello friends.
I'm working with cc2531 / zigbee2mqtt / mosquitto / domoticz at localhost: 1883. I can make the server work, I can pair my device (DoorSensor HEIMAN HS1DS) but when I operate the sensor I get nothing on the server, no information. Could you please help me ?. Thank you.

Koenkk, 2019-01-22:
Could you share your log with `log_level: debug`? (https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html)

FRFrancisqueti, 2019-01-22:
Hi Koen, thank you very much for your response. This is my log_level debug. Note that it finds my device, in this case a HEIMAN DoorSensor, but when I turn on the sensor switch it does not return anything to me. It is a sensor equal to what exists in the library but the name is different (HS1DS-E). I followed your tutorial and created this new sensor. With Node-red I also find the device but I can not see the sensor's performance. What do you think might be happening?

2019-1-22 15:05:40 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-22.15-05-40'
2019-1-22 15:05:40 - debug: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-01-22.13-11-36'
2019-1-22 15:05:41 - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
2019-1-22 15:05:41 - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
2019-1-22 15:05:41 - info: Starting zigbee2mqtt version 1.0.1 (commit #9380bbc)
2019-1-22 15:05:41 - info: Starting zigbee-shepherd
2019-1-22 15:05:45 - info: zigbee-shepherd started
2019-1-22 15:05:45 - info: Coordinator firmware version: 'undefined'
2019-1-22 15:05:45 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b001201387d","nwkAddr":0},"firmware":{"error":"Unable to get firmware version"},"startTime":1548169545,"joinTimeLeft":0}
2019-1-22 15:05:45 - info: Currently 1 devices are joined:
2019-1-22 15:05:45 - info: 0x000d6f000e1e407c (0x000d6f000e1e407c): HS1DS-E - HEIMAN Door sensor (EndDevice)
2019-1-22 15:05:45 - warn: `permit_join` set to  `true` in configuration.yaml.
2019-1-22 15:05:45 - warn: Allowing new devices to join.
2019-1-22 15:05:45 - warn: Set `permit_join` to `false` once you joined all devices.
2019-1-22 15:05:45 - info: Zigbee: allowing new devices to join.
2019-1-22 15:05:45 - info: Connecting to MQTT server at mqtt://localhost:1883
2019-1-22 15:05:46 - info: zigbee-shepherd ready
2019-1-22 15:05:46 - info: Connected to MQTT server
2019-1-22 15:05:46 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2019-1-22 15:05:46 - debug: Mounted the cieApp (epId {epId})

Thank you

Koenkk, 2019-01-23:
Could you provide the `database.db`?

FRFrancisqueti, 2019-01-23:
Follow the db...

https://pastebin.com/0fLS6VMc

Koenkk, 2019-01-23:
It's not compatible with the HS1DS, meaning this is a new/unsupported device. Can you add to `devices.js`:

```js
{
        zigbeeModel: ['DoorSensor-EM'],
        model: 'HS1DS-E',
        vendor: 'HEIMAN',
        description: 'Door sensor',
        supports: 'contact',
        fromZigbee: [fz.heiman_contact],
        toZigbee: [],
    },
```

FRFrancisqueti, 2019-01-24:
Koen, man, thank you so much for the help. I can now receive the sensor data.
Could you help me with one more question? I tried using the tutorial https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin along with zigbee2mqtt, but when installing the plugin with the same localhost: 1883 domoticz stops working. Do you know if there is another tutorial to integrate zigbee2mqtt with domoticz?

Thank you.

Koenkk, 2019-01-26:
Thanks, this device will be supported in the next version of zigbee2mqtt.

I don't know about any other tutorial, I would recommend asking for help by creating an issue here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues

closed 2019-01-26 by Koenkk