Koenkk/zigbee2mqtt #914: New device - 30s timeout for every state change

fredrikgk, 2019-01-21:
I am running:
zigbee2mqtt version 1.0.1 (commit #9380bbc)
CC2531 FW version 20180815

I am trying to add support for http://proff.elko.no/elko-smart-home-316gled/esh-plus-316gled-dimmer-rf-ph-article2658-1781.html (this is in Norwegian since the brand is only sold in the Scandinavian markets) - Basically a dimmer switch with a Silabs Zigbee module integrated. 

I have successfully set it up so that HomeAssistant receives status from the dimmer switch, and can also send commands to it. 

When I change state of the dimmer from HomeAssistant, the dimmer updates immediately. However, then something happens (or does not happen) which effectively blocks any new command from being sent over the air. I can see in the log that the first message sent is reported as failed with a 30 s timeout, although it seems like the dimmer has actually acknowledged the first message. 

Any message sent after the initial command will be transmitted over the air right after the 30 s timeout, but will trigger a new similar timeout. 

Here is a log of an OFF command until the timeout happens:
https://pastebin.com/0KuAmHtY

Here is a log of an OFF command sent with an ON command following immediately after:
https://pastebin.com/RyfwYXaW 

If required I can provide a sniffer capture of the actual Zigbee communication.

I am in over my head here with limited knowledge about Zigbee and even less about JS. Any help is appreciated.

Thanks,
Fredrik

Koenkk, 2019-01-22:
Can you try setting https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/devicePublish.js#L24 

to 

`this.queue.concurrency = 10;`

fredrikgk, 2019-01-22:
That does the trick, sort of. I am now able to send commands back to back, but I do still get the "failed with error Error" timeouts after 30 s. Sending more than 10 commands also halts the system. 

Any insight into what causes the error?

fredrikgk, 2019-01-23:
I confirm that clogging issues are resolved on the dev branch.

Koenkk, 2019-01-26:
Great, despite the errors in the log, does everything work? If so, can you make a PR?

About the errors, it's hard to debug but my guess is that the device is not fully ZCL compliant.

closed 2019-03-28 by Koenkk

fredrikgk, 2019-06-05:
@Koenkk 

I just updated to the latest version on the dev branch, and this problem has reappeared. There seems to have been significant changes in devicePublish.js since the proposed workaround above, as well as the official fix you added at the end of January.

Any suggestion on how to fix this (again)?

Koenkk, 2019-06-06:
Did you change the firmware? Can you post the `DEBUG=* npm start` log?

fredrikgk, 2019-06-08:
I have not updated the FW. I will try that first.

fredrikgk, 2019-06-11:
I updated to the 20190425 FW and the problem is still there. It is only affecting the dimmer switches, I can toggle my Trådfri outlet while waiting for the 30s timeout of the dimmer. 

Here is the DEBUG=* nom start output: https://pastebin.com/2ZPDTG3x

Koenkk, 2019-06-12:
The reasons of these errors seems that zigbee-shepherd is waiting for an APS ACK of the device while the device never does this. Can you confirm this by sniffing the zigbee traffic?

fredrikgk, 2019-07-07:
That is the case, indeed! Is there anyway to work around this issue?

[image]

Koenkk, 2019-07-08:
Probably yes, can you add a section like this https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L4424 to the device in `devices.js`

fredrikgk, 2019-07-10:
Adding the disFeedbackRsp option does not solve the issue. It stops commands to other similar devices from being blocked, but commands to the same device is still blocked for 30 s.

Koenkk, 2019-07-10:
Can you try changing https://github.com/Koenkk/zigbee-herdsman/blob/master/src/lib/components/af.ts#L797 to 

```js
    var afOptions = ZSC.AF.options.DISCV_ROUTE;
```

fredrikgk, 2019-07-15:
Nah, that did not do it either. 

I have sent an email to the manufacturer also. This is probably not an issue for them as they only support their own gateway, so I am not sure they will actually fix it.

Koenkk, 2019-07-15:
Did you change the `.ts` or `.js` file? The `.ts` file requires recompilation (forgot to mention that, sorry). Therefore I would advise changing the `.js` file

fredrikgk, 2019-07-18:
I did actually change the .js file.

Koenkk, 2019-07-20:
Ok, so https://github.com/Koenkk/zigbee2mqtt/issues/914#issuecomment-509913662 does make a difference right? Can you post the `DEBUG* npm start` logging when running with this?

fredrikgk, 2019-07-22:
I took a step back and updated to the latest version on the dev branch. This also meant that I switched from Shepherd to Herdsman.

Out of the box, this gave a 30 s delay, but only for the same device. Meaning I could control multiple devices sequentially without delays, but any consecutive command would be halted for 30 s. 

`var afOptions = ZSC.AF.options.DISCV_ROUTE;` does not change anything

`options: {
            disFeedbackRsp: true,` however, now resolves the problem. 

Thanks for all your help! I will let you know if I here anything from the manufacturer.

Do you want me to do a PR to [zigbee-shepherd-converters](https://github.com/Koenkk/zigbee-shepherd-converters) or can you add the line directly?

Koenkk, 2019-07-22:
Great! Can you make a PR?

fredrikgk, 2019-07-22:
Done!

fredrikgk, 2019-07-22:
Updating to the latest zigbee-shepherd-converters also seems to have resolved issues I was having with Xiaomi Aqara sensors not reconnecting to network by the way.

napsio, 2019-08-19:
I have the same problem with the Gledopto devices and Home-Assistant.

When I try to turn the device on I got:

`zigbee2mqtt:error 2019-8-19 7:06:24 PM Zigbee publish to device 'Gledopto_GL_B_007Z', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - 12 failed with error Error: Timed out after 30000 ms`

When I try to change the brightness I got:

`zigbee2mqtt:error 2019-8-19 7:10:51 PM Zigbee publish to device 'Gledopto_GL_B_007Z', genLevelCtrl - moveToLevelWithOnOff - {"level":129,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 12 failed with error Error: Timed out after 30000 ms`

fredrikgk, 2019-08-20:
Did you try the workarounds discussed above?

napsio, 2019-08-20:
Yes I did, but nothing helped to fix the problem.

Koenkk, 2019-09-09:
@fredrikgk I'm not sure if I correctly moved the fix to zigbee-herdsman (#1888), can you check if the device still works OK in the zigbee2mqtt dev branch?

fredrikgk, 2019-09-14:
@Koenkk , I checked out the latest dev branch, the 30s timeout is back. 

I see things have changed a bit in devices.js, how do I tell it to not wait for the APS ACK now?

Koenkk, 2019-09-14:
@fredrikgk can you try adding this for the device? https://github.com/Koenkk/zigbee-shepherd-converters/blob/0cf4bb0cd99d81ea2daa021633328e62182989eb/devices.js#L2517

fredrikgk, 2019-09-14:
yep, that fixed it!