Koenkk/zigbee2mqtt #915: Disable Xiaomi Aqara pairing

untitledlt, 2019-01-21:
I am using CC2531 and Xiaomi Aqara WXKG02LM wall switch.
I have successfully paired device and i see messages on mqtt broker.
I have set `permit_join: false` in `configuration.yaml` file.
The problem is if someone clicks and holds Aqara button for 10 seconds, it enters pairing mode and zigbee2mqtt forgets this device. To make it work again I need to `permit_join` and restart pairing process.

Is it possible to force `zigbee2mqtt` to remember this device even if someone has started pairing mode on Aqara wall switch?

Koenkk, 2019-01-22:
No, the permit joining logic is completely handled by the Zigbee stack  and therefore not forwarded to zigbee2mqtt.

untitledlt, 2019-01-22:
But if `permit_join` is set to `false` and `zigbee2mqtt` does not allow joining, why it forgets this device? How does Zigbee device clears pairing with `zigbee2mqqt`?

BudBundi, 2019-01-22:
Not zigbee2mqtt is forgetting the device, the device is factory reset so the device forget the coordinator

Koenkk, 2019-01-22:
When a device is factory reset it forgets about the network it was paired to. This is not something that is handled inside the coordinator/zigbee2mqtt. When the device want's to rejoin the network it needs to talk to the coordinator in order to share e.g. the encryption keys. Because permit join is disabled, the coordinator will refuse to do this.

untitledlt, 2019-01-22:
Yeah, that makes sense. I would say it's a bad design from Xiaomi side as wall switch could easily be reset by anyone just by holding it for 10 secs.
Thank you for taking your time to explain!