Koenkk/zigbee2mqtt #916: change network key -> repair devices

chiefe86, 2019-01-22:
Dear all,

I want to change the network key now.

What does it mean to repair the devices afterwards?

What network key is allowed? only numbers? is there a restriction?

Many thanks

Koenkk, 2019-01-22:
Because the network key changes, the devices cannot communicate with the coordinator, therefore you have to repair all devices.

Only an array of 16 numbers is allowed.

chiefe86, 2019-01-22:
Thank you for your reply.
How can I repair them? 
Do I have to pair them again, or repair them? Sorry for the confusion...

ryanbeaton, 2019-01-22:
Yes that term gets quite confusing here. You will need to re-pair (pair again) not repair (as in some fixing).

daveyrb, 2019-01-27:
`advanced:
  network_key: [7, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 11, 12, 13],`
Should the above have the comma on the end?
If I put that in my config it fails to load if I remove the comma it loads OK. I have yet to add any devices as I have just started setting it up.

Koenkk, 2019-01-27:
There shouldnt be a comma (otherwise its not valid yaml)

h4nc, 2019-01-30:
@koenk I also switched changed my network key. 

I also had to delete that comma at the end to be able to set it up. Otherwise I got an error in the zigbee logs. This error pointet to this comma

So instead of 

`advanced: network_key: [7, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 11, 12, 13],`

this: 

`advanced: network_key: [7, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 11, 12, 13],`

Of course with other numbers. Also after saving without comma and rebooting the addon, my yaml now looks like this (not my real code).

[image]

Could you also add the restriction to that network_key to the docs?
Like 

- 16 numbers
- only one or two characters (so 1, 11)? Is 99 allowed? Is 999 allowed?

Koenkk, 2019-01-30:
@h4nc this is already being worked on: #963 #969

Koenkk, 2019-01-31:
@h4nc fixed in #963. I assume this can be closed now.

closed 2019-01-31 by Koenkk