Koenkk/zigbee2mqtt #917: Salus FC600 - new device

frozenwave11, 2019-01-22:
Hello everyone,

Im trying to pair fan coil thermostat ( https://salus-controls.com/uk/product/fc600/#description ) with Hass.io. 

 Thermostat measures room temperature, on thermostat can be set cooling/heating mode, three fan speeds, schedule (not currently interested in), working and standby function and etc.

So fare I add device description in configuration.yaml and basic description in device.js file. 
It would be great if anyone is interested in helping me configure this product. 

Im ready to provide any logs, descriptions and etc as well as test any configuration posted.

Thanks for participating.

khzd, 2019-01-22:
Hi
Please post your modification of device.js

frozenwave11, 2019-01-22:
Hi, I delete everything from device.js apart from the name of device. and try to pair it. this is what recorded inside dB:

{"id":2,"type":"Router","ieeeAddr":"0x001e5e090223703d","nwkAddr":26048,"manufId":4216,"manufName":"SALUS","powerSource":"Mains (single phase)","modelId":"FC600","epList":[9],"status":"online","joinTime":1548176339,"endpoints":{"9":{"profId":260,"epId":9,"devId":769,"inClusterList":[0,3,4,5,513,514,516,1026,64516,64518,64521],"outClusterList":[3,10,25,64516],"clusters":{"64516":{"dir":{"value":3},"attrs":{}},"64518":{"dir":{"value":1},"attrs":{}},"64521":{"dir":{"value":1},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":87,"hwVersion":2,"manufacturerName":"SALUS","modelId":"FC600","dateCode":"20170503","powerSource":1}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genTime":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"hvacThermostat":{"dir":{"value":1},"attrs":{"localTemp":2000,"ocupancy":0,"localTemperatureCalibration":0,"occupiedCoolingSetpoint":2400,"occupiedHeatingSetpoint":2100,"unoccupiedCoolingSetpoint":1500,"unoccupiedHeatingSetpoint":3000,"minHeatSetpointLimit":500,"maxHeatSetpointLimit":4000,"minCoolSetpointLimit":500,"maxCoolSetpointLimit":4000,"ctrlSeqeOfOper":4,"systemMode":4,"runningMode":0,"programingOperMode":0,"runningState":0,"acErrorCode":0}},"hvacFanCtrl":{"dir":{"value":1},"attrs":{"fanMode":5,"fanModeSequence":2}},"hvacUserInterfaceCfg":{"dir":{"value":1},"attrs":{"tempDisplayMode":0,"keypadLockout":0}},"msTemperatureMeasurement":{"dir":{"value":1},"attrs":{"measuredValue":0,"minMeasuredValue":0,"maxMeasuredValue":5000}}}}},"_id":"oOkTYOIHhQ9clKyj"}

Chrischi-, 2019-01-22:
Check out the ecozy or bitron thermostat implementation. 
You should be up and running in no time. 
I think this is just a copy & paste job.

frozenwave11, 2019-01-22:
I've tried both but no luck (Unsupported cluster), also I tried my custom setting which are written below still same. I'm not sure if I wrote everything correctly. 

zigbeeModel: ['FC600'],
        model: 'FC600',
        vendor: 'SALUS',
        description: 'Smart heating thermostat',
        supports: 'temperature, occupancy, un-/occupied heating, schedule',
        fromZigbee: [
            fz.ignore_basic_change, fz.thermostat_att_report, fz.thermostat_dev_change,
        ],
        toZigbee: [
            tz.factory_reset, tz.thermostat_local_temperature, tz.thermostat_local_temperature_calibration,
            tz.thermostat_occupancy, tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_unoccupied_heating_setpoint, tz.thermostat_setpoint_raise_lower,
            tz.thermostat_remote_sensing, tz.thermostat_control_sequence_of_operation, tz.thermostat_system_mode,
            tz.thermostat_weekly_schedule, tz.thermostat_clear_weekly_schedule, tz.thermostat_weekly_schedule_rsp,
            tz.thermostat_relay_status_log, tz.thermostat_relay_status_log_rsp,
        ],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 9);
            const actions = [
                // from https://github.com/ckpt-martin/Hubitat/blob/master/eCozy/eCozy-ZigBee-Thermostat-Driver.groovy
                (cb) => device.bind('genBasic', coordinator, cb),
                (cb) => device.bind('genIdentify', coordinator, cb),
                (cb) => device.bind('genGroups', coordinator, cb),
                (cb) => device.bind('genScenes', coordinator, cb),
                (cb) => device.bind('genTime', coordinator, cb),
                (cb) => device.bind('genOta', coordinator, cb),
                (cb) => device.report('hvacThermostat', 'localTemp', 300, 5500, 0, cb),
                (cb) => device.report('hvacThermostat', 'localTemperatureCalibration', 1, 0, 0, cb),
                (cb) => device.report('hvacThermostat', 'occupiedCoolingSetpoint', 1, 0, 1, cb),
                (cb) => device.report('hvacThermostat', 'occupiedHeatingSetpoint', 1, 0, 1, cb),
                (cb) => device.report('hvacThermostat', 'unoccupiedCoolingSetpoint', 1, 0, 1, cb),
                (cb) => device.report('hvacThermostat', 'unoccupiedHeatingSetpoint', 1, 0, 1, cb),
                (cb) => device.report('hvacThermostat', 'minHeatSetpointLimit', 500, 4000, 1, cb),
                (cb) => device.report('hvacThermostat', 'maxHeatSetpointLimit', 4000, 500, 1, cb),
                (cb) => device.report('hvacThermostat', 'minCoolSetpointLimit', 500, 4000, 1, cb),
                (cb) => device.report('hvacThermostat', 'ctrlSeqeOfOper', 1, 0, 1, cb),
                (cb) => device.report('hvacThermostat', 'systemMode', 1, 0, 1, cb),
                (cb) => device.report('hvacThermostat', 'runningMode', 1, 0, 0, cb),
                (cb) => device.report('hvacThermostat', 'programingOperMode', 1, 0, 0, cb),
                (cb) => device.report('hvacThermostat', 'runningState', 1, 0, 0, cb),
                (cb) => device.report('hvacThermostat', 'runningState', 1, 0, 0, cb),
                (cb) => device.report('hvacThermostat', 'acErrorCode', 1, 0, 0, cb),
                (cb) => device.report('hvacFanCtrl', 'fanMode', 1, 0, 3, cb),
                (cb) => device.report('hvacFanCtrl', 'fanModeSequence', 1, 0, 5, cb),
                (cb) => device.bind('hvacUserInterfaceCfg', coordinator, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 1, 0, 0, cb),
                (cb) => device.report('msTemperatureMeasurement', 'minMeasuredValue',1, 0, 0, cb),
                (cb) => device.report('msTemperatureMeasurement', 'maxMeasuredValue',5000, 0, 0, cb),

frozenwave11, 2019-01-22:
this is log: 

 zigbee2mqtt:warn 1/22/2019, 11:05:33 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 1/22/2019, 11:05:33 PM Allowing new devices to join.
  zigbee2mqtt:warn 1/22/2019, 11:05:33 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 1/22/2019, 11:05:33 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 1/22/2019, 11:05:33 PM Connecting to MQTT server at mqtt://192.168.0.43
  zigbee-shepherd:request REQ --> ZDO:mgmtPermitJoinReq +53ms
  cc-znp:SREQ --> ZDO:mgmtPermitJoinReq, { addrmode: 15, dstaddr: 65532, duration: 255, tcsignificance: 0 } +0ms
  serialport:main _write 10 bytes of data +73ms
  serialport:bindings write 10 bytes +73ms
  serialport:unixWrite Starting write 10 bytes offset 0 bytesToWrite 10 +89ms
  serialport:unixWrite write returned null 10 +1ms
  serialport:unixWrite wrote 10 bytes +0ms
  serialport:unixWrite Finished writing 10 bytes +0ms
  serialport:main binding.write write finished +1ms
  zigbee2mqtt:info 1/22/2019, 11:05:33 PM zigbee-shepherd ready
  zigbee-shepherd:request REQ --> AF:register +4ms
  zigbee2mqtt:info 1/22/2019, 11:05:33 PM Connected to MQTT server
  zigbee2mqtt:info 1/22/2019, 11:05:33 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  serialport:poller received "readable" +94ms
  serialport:bindings read +23ms
  serialport:unixRead Starting read +94ms
  serialport:unixRead Finished read 6 bytes +1ms
  serialport:main binding.read finished +23ms
  cc-znp { sof: 254,
  cc-znp   len: 1,
  cc-znp   type: 'SRSP',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'mgmtPermitJoinReq',
  cc-znp   payload: { status: 0 },
  cc-znp   fcs: 82,
  cc-znp   csum: 82 } +22ms
  serialport:main _read reading +1ms
  serialport:bindings read +2ms
  serialport:unixRead Starting read +1ms
  cc-znp:SRSP <-- ZDO:mgmtPermitJoinReq, { status: 0 } +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +0ms
  serialport:poller Polling for "readable" +2ms
  cc-znp:SREQ --> AF:register, { endpoint: 11, appprofid: 260, appdeviceid: 1024, appdevver: 0, latencyreq: 0, appnuminclusters: 3, appinclusterlist: [ 0, 1281, 3 ], appnumoutclusters: 3, appoutclusterlist: [ 1280, 1282, 3 ] } +1ms
  serialport:main _write 26 bytes of data +1ms
  serialport:bindings write 26 bytes +1ms
  serialport:unixWrite Starting write 26 bytes offset 0 bytesToWrite 26 +25ms
  serialport:unixWrite write returned null 26 +1ms
  serialport:unixWrite wrote 26 bytes +0ms
  serialport:unixWrite Finished writing 26 bytes +0ms
  serialport:main binding.write write finished +1ms
  serialport:poller received "readable" +5ms
  serialport:bindings read +4ms
  serialport:unixRead Starting read +5ms
  serialport:unixRead Finished read 6 bytes +0ms
  serialport:main binding.read finished +3ms
  cc-znp { sof: 254,
  cc-znp   len: 1,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'permitJoinInd',
  cc-znp   payload: { duration: 255 },
  cc-znp   fcs: 112,
  cc-znp   csum: 112 } +5ms
  serialport:main _read reading +1ms
  serialport:bindings read +1ms
  serialport:unixRead Starting read +1ms
  cc-znp:AREQ <-- ZDO:permitJoinInd, { duration: 255 } +0ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:permitJoinInd +1ms
  serialport:unixRead waiting for readable because of code: EAGAIN +1ms
  serialport:poller Polling for "readable" +2ms
  serialport:poller received "readable" +3ms
  serialport:bindings read +4ms
  serialport:unixRead Starting read +3ms
[... 195 lines cut ...]
  serialport:unixRead Starting read +2ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 64516, srcaddr: 4491, srcendpoint: 9, dstendpoint: 1, wasbroadcast: 0, linkquality: 110, securityuse: 0, timestamp: 8080470, transseqnumber: 0, len: 75, data: <Buffer 1d 78 10 16 00 28 2b 32 30 30 30 2b 32 30 30 30 32 30 32 31 30 32 34 30 32 31 30 32 34 30 32 31 30 32 34 30 30 30 30 31 44 30 30 30 30 30 30 30 30 30 ... 25 more bytes> } +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +1ms
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:27
        throw new Error('Unrecognized cluster');
        ^

Error: Unrecognized cluster
    at new FuncPayload (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:27:15)
    at /opt/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:33:22
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:121:9)
    at Object.onceWrapper (events.js:276:13)
    at Dissolve.emit (events.js:188:13)
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/dissolve-chunks/index.js:73:29)
    at Dissolve.emit (events.js:188:13)
    at emitReadable_ (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:456:10)
    at emitReadable (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:450:7)
    at readableAddChunk (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:206:11)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.0.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

Chrischi-, 2019-01-22:
I would try to build ignore converter for
"64516":{"dir":{"value":3},"attrs":{}},
"64518":{"dir":{"value":1},"attrs":{}},
"64521":{"dir":{"value":1},"attrs":{}}

frozenwave11, 2019-01-23:
[FC600 - log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2785236/FC600.-.log.txt)
Hi again, 
I did as you said, removed all and only add configure: and everything worked perfectly, you can see in text file. 
zigbee2mqtt:info 1/23/2019, 1:47:58 AM Succesfully configured FC600 0x001e5e090223703d.
 can you please tell me what to do next as im stuck.

Chrischi-, 2019-01-23:
let it run and see which messages you receive... After that include the converters.

Why do you want to report all that stuff? I think you didn't understand the syntax.
And this `(cb) => device.report('hvacThermostat', 'runningState', 1, 0, 0, cb),` is set twice.

Chrischi-, 2019-01-23:
please show your devices.js and build ignore converter for these unrecognized clusters (i hope that will work).

you could easily paste logs etc. at [pastebin.com](pastebin.com)

Koenkk, 2019-01-23:
The cluster should be added to zcl-id first.

Can you add to https://github.com/Koenkk/zcl-packet/blob/master/lib/functional.js#L25 `console.log("This is the cluster", clusterId, direction, cmd)`

Chrischi-, 2019-01-23:
Thanks for the clarification. So zigbee-shepherd needs to know the `zcl-id` to ignore it, right?

Koenkk, 2019-01-23:
@Chrischi- yes, otherwise it will crash

frozenwave11, 2019-01-23:
zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +0ms
This is the cluster 64516 1 0
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:27
        throw new Error('Unrecognized cluster');

frozenwave11, 2019-01-23:
@Koenkk  Now i get
 This is the cluster 64516 1 0
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:33
        throw new Error('Unrecognized command');

https://pastebin.com/6vcfk0Su
@Chrischi- What do you think, how should I build ignore converter?

Koenkk, 2019-01-23:
@frozenwave11

add to https://github.com/Koenkk/zcl-id/blob/master/definitions/common.json#L197
`"manuSpecificCluster1": 64516`

 add to https://github.com/Koenkk/zcl-id/blob/master/definitions/cluster_defs.json#L1

```
"manuSpecificCluster1": {
        "attrId":{},
        "cmd":{"cmd1":0},
        "cmdRsp":null},
```

frozenwave11, 2019-01-23:
@Koenkk  same error
https://pastebin.com/1PfsV26H error
https://pastebin.com/LHKjsu58 common.json
https://pastebin.com/9SFjC4J4 cluster_defs.json

Koenkk, 2019-01-23:
add to https://github.com/Koenkk/zcl-id/blob/master/definitions/cluster_defs.json#L1

```
"manuSpecificCluster1": {
        "attrId":{},
        "cmd": null,
        "cmdRsp":{"cmd1":0}},
```

frozenwave11, 2019-01-23:
@Koenkk 
Error: Unrecognized direction
full log - https://pastebin.com/BufCkLPB

Koenkk, 2019-01-23:
add to https://github.com/Koenkk/zcl-id/blob/master/definitions/cluster_defs.json#L1

```
"manuSpecificCluster1": {
        "attrId":{},
        "cmd": null,
        "cmdRsp":{"cmdRsp1":0}},
```

add to https://github.com/Koenkk/zcl-packet/blob/master/lib/defs/zcl_meta.json#L131
```
        "genBasic": {
            "cmdRsp1":{
                "params":[],
                "dir":1}
        },
```

frozenwave11, 2019-01-23:
@Koenkk  tried, same error

zigbee2mqtt:info 1/23/2019, 3:41:23 PM Succesfully configured FC600 0x001e5e090223703d
  serialport:unixRead waiting for readable because of code: EAGAIN +1ms
  serialport:poller Polling for "readable" +2ms
  serialport:poller received "readable" +13s
  serialport:bindings read +13s
  serialport:unixRead Starting read +13s
  serialport:unixRead Finished read 8 bytes +0ms
  serialport:main binding.read finished +13s
  cc-znp { sof: 254,
  cc-znp   len: 3,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'srcRtgInd',
  cc-znp   payload: { dstaddr: 48945, relaycount: 0, relaylist: <Buffer > },
  cc-znp   fcs: 12,
  cc-znp   csum: 12 } +13s
  serialport:main _read reading +2ms
  serialport:bindings read +3ms
  serialport:unixRead Starting read +2ms
  cc-znp:AREQ <-- ZDO:srcRtgInd, { dstaddr: 48945, relaycount: 0, relaylist: <Buffer > } +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +0ms
  serialport:poller Polling for "readable" +3ms
  serialport:poller received "readable" +60ms
  serialport:bindings read +60ms
  serialport:unixRead Starting read +60ms
  serialport:unixRead Finished read 100 bytes +0ms
  serialport:main binding.read finished +60ms
  cc-znp { sof: 254,
  cc-znp   len: 95,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 64516,
  cc-znp      srcaddr: 48945,
  cc-znp      srcendpoint: 9,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 120,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 5431960,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 75,
  cc-znp      data:
  cc-znp       <Buffer 1d 78 10 18 00 28 2b 32 30 30 30 2b 32 30 30 30 32 30 31 39 35 32 34 30 32 31 30 32 34 30 31 39 35 32 34 30 30 30 30 32 43 30 30 30 30 30 30 30 30 30 ... 25 more bytes> },
  cc-znp   fcs: 111,
  cc-znp   csum: 111 } +61ms
  serialport:main _read reading +2ms
  serialport:bindings read +2ms
  serialport:unixRead Starting read +2ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 64516, srcaddr: 48945, srcendpoint: 9, dstendpoint: 1, wasbroadcast: 0, linkquality: 120, securityuse: 0, timestamp: 5431960, transseqnumber: 0, len: 75, data: <Buffer 1d 78 10 18 00 28 2b 32 30 30 30 2b 32 30 30 30 32 30 31 39 35 32 34 30 32 31 30 32 34 30 31 39 35 32 34 30 30 30 30 32 43 30 30 30 30 30 30 30 30 30 ... 25 more bytes> } +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +0ms
This is the cluster 64516 1 0
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:41
        throw new Error('Unrecognized direction');

Koenkk, 2019-01-23:
@frozenwave11 are the changes of https://github.com/Koenkk/zigbee2mqtt/issues/917#issuecomment-456809777 still there?

frozenwave11, 2019-01-23:
@Koenkk  yes they are

Koenkk, 2019-01-23:
The cluster is different this time, add to https://github.com/Koenkk/zcl-id/blob/master/definitions/common.json#L197
`"manuSpecificCluster2": 64518`

Chrischi-, 2019-01-23:
see my comment ;)

https://github.com/Koenkk/zigbee2mqtt/issues/917#issuecomment-456801432

Koenkk, 2019-01-23:
@Chrischi- good point, @frozenwave11  can you also add the third cluster?

frozenwave11, 2019-01-23:
@Koenkk @Chrischi- 
I add all three, but getting same old error - https://pastebin.com/7SqSet4q
https://pastebin.com/cdeeZ6a7 - common.json

Koenkk, 2019-01-23:
Please try with https://github.com/Koenkk/zcl-id/commits/salus and https://github.com/Koenkk/zcl-packet/tree/salus

Chrischi-, 2019-01-23:
@Koenkk 
why not add the range from 0xfc00 to 0xffff? 
These are according to ZCL the Manufacturer specific cluster id's

Koenkk, 2019-01-23:
@Chrischi- AFAIK we also need to add commands, otherwise it will still crash. (which we cannot do without knowing what is send?)

frozenwave11, 2019-01-23:
@Koenkk @Chrischi- 
here is the latest log.
https://pastebin.com/KUjmSZGW

Chrischi-, 2019-01-23:
@Koenkk 
why not output a `console.log` as default so you can add the command with your PR.

frozenwave11, 2019-01-23:
when i do nothing at this point it stops,

zigbee-shepherd-converters:devices Configured '(cb) => device.bind('hvacUserInterfaceCfg', coordinator, cb)' with result 'OK' +468ms
  zigbee2mqtt:info 1/23/2019, 6:28:57 PM Succesfully configured FC600 0x001e5e090223703d
  serialport:unixRead waiting for readable because of code: EAGAIN +1ms
  serialport:poller Polling for "readable" +3ms

after min or i get this and still continue working

serialport:poller received "readable" +1m
  serialport:bindings read +1m
  serialport:unixRead Starting read +1m
  serialport:unixRead Finished read 8 bytes +1ms
  serialport:main binding.read finished +1m
  cc-znp { sof: 254,
  cc-znp   len: 3,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'srcRtgInd',
  cc-znp   payload: { dstaddr: 48945, relaycount: 0, relaylist: <Buffer > },
  cc-znp   fcs: 12,
  cc-znp   csum: 12 } +1m
  serialport:main _read reading +1ms
  serialport:bindings read +2ms
  serialport:unixRead Starting read +1ms
  cc-znp:AREQ <-- ZDO:srcRtgInd, { dstaddr: 48945, relaycount: 0, relaylist: <Buffer > } +0ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +1ms
  serialport:poller Polling for "readable" +4ms
  serialport:poller received "readable" +55ms
  serialport:bindings read +56ms
  serialport:unixRead Starting read +55ms
  serialport:unixRead Finished read 32 bytes +0ms
  serialport:main binding.read finished +56ms
  cc-znp { sof: 254,
  cc-znp   len: 27,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 64516,
  cc-znp      srcaddr: 48945,
  cc-znp      srcendpoint: 9,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 63,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 5994177,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 7,
  cc-znp      data: <Buffer 04 78 10 7e 00 0f 00> },
  cc-znp   fcs: 250,
  cc-znp   csum: 250 } +57ms
  serialport:main _read reading +2ms
  serialport:bindings read +2ms
  serialport:unixRead Starting read +2ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 64516, srcaddr: 48945, srcendpoint: 9, dstendpoint: 1, wasbroadcast: 0, linkquality: 63, securityuse: 0, timestamp: 5994177, transseqnumber: 0, len: 7, data: <Buffer 04 78 10 7e 00 0f 00> } +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +0ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +1ms
  serialport:unixRead waiting for readable because of code: EAGAIN +2ms
  serialport:poller Polling for "readable" +4ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms

Chrischi-, 2019-01-23:
please show your `devices.js`

frozenwave11, 2019-01-23:
https://pastebin.com/jcEcvVjs 
Starts at Line 1842

frozenwave11, 2019-01-23:
i tried  at the bottom there are two messages, i dont beleve it works correctly 
 https://pastebin.com/KnKE89Ey

Chrischi-, 2019-01-23:
wrong clusterid _64516_

wait and let it run until you get other messages.

frozenwave11, 2019-01-24:
@Chrischi- 
Thermostat can be set in online and offline mode. 

When from factory reset I set it in pair (online mode) I got this error
 - https://pastebin.com/M2GNnVY8

If I first set it in offline mode then from menu pair it it doesn't show any error
 - https://pastebin.com/CvuN8e51

This is switching on 
- https://pastebin.com/jYJKAqXJ

Temperature rising and lowering 
- https://pastebin.com/8N0idF7L

Changing from heating to cooling and cooling to heating 
- https://pastebin.com/0f93wT21

frozenwave11, 2019-01-24:
@Chrischi- 

This is error when I switch fan from  max  to off

serialport:poller received "readable" +2m
  serialport:bindings read +2m
  serialport:unixRead Starting read +2m
  serialport:unixRead Finished read 8 bytes +1ms
  serialport:main binding.read finished +2m
  cc-znp { sof: 254,
  cc-znp   len: 3,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'srcRtgInd',
  cc-znp   payload: { dstaddr: 60254, relaycount: 0, relaylist: <Buffer > },
  cc-znp   fcs: 55,
  cc-znp   csum: 55 } +2m
  serialport:main _read reading +8ms
  serialport:bindings read +9ms
  serialport:unixRead Starting read +9ms
  cc-znp:AREQ <-- ZDO:srcRtgInd, { dstaddr: 60254, relaycount: 0, relaylist: <Buffer > } +4ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +1ms
  serialport:unixRead waiting for readable because of code: EAGAIN +2ms
  serialport:poller Polling for "readable" +13ms
  serialport:poller received "readable" +50ms
  serialport:bindings read +53ms
  serialport:unixRead Starting read +50ms
  serialport:unixRead Finished read 100 bytes +1ms
  serialport:main binding.read finished +54ms
  cc-znp { sof: 254,
  cc-znp   len: 95,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 64516,
  cc-znp      srcaddr: 60254,
  cc-znp      srcendpoint: 9,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 39,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 930299,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 75,
  cc-znp      data:
  cc-znp       <Buffer 1d 78 10 0d 00 28 2b 32 30 30 30 2b 32 30 30 30 36 30 31 38 35 32 34 30 32 31 30 32 34 30 31 38 30 32 33 30 30 30 30 38 44 30 30 30 30 30 30 30 30 30 ... > },
  cc-znp   fcs: 194,
  cc-znp   csum: 194 } +58ms
  serialport:main _read reading +10ms
  serialport:bindings read +12ms
  serialport:unixRead Starting read +11ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 64516, srcaddr: 60254, srcendpoint: 9, dstendpoint: 1, wasbroadcast: 0, linkquality: 39, securityuse: 0, timestamp: 930299, transseqnumber: 0, len: 75, data: <Buffer 1d 78 10 0d 00 28 2b 32 30 30 30 2b 32 30 30 30 36 30 31 38 35 32 34 30 32 31 30 32 34 30 31 38 30 32 33 30 30 30 30 38 44 30 30 30 30 30 30 30 30 30 ... > } +5ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2ms
This is the cluster 64516 1 0
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5ms
  serialport:unixRead waiting for readable because of code: EAGAIN +8ms
  serialport:poller Polling for "readable" +21ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  serialport:poller received "readable" +916ms
  serialport:bindings read +926ms
  serialport:unixRead Starting read +918ms
  serialport:unixRead Finished read 8 bytes +1ms
  serialport:main binding.read finished +928ms
  cc-znp { sof: 254,
  cc-znp   len: 3,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'srcRtgInd',
[... 81 lines cut ...]
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 39,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 949067,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 8,
  cc-znp      data: <Buffer 08 0f 0a 29 00 19 00 00> },
  cc-znp   fcs: 227,
  cc-znp   csum: 227 } +47ms
  serialport:main _read reading +10ms
  serialport:bindings read +12ms
  serialport:unixRead Starting read +12ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 513, srcaddr: 60254, srcendpoint: 9, dstendpoint: 1, wasbroadcast: 0, linkquality: 39, securityuse: 0, timestamp: 949067, transseqnumber: 0, len: 8, data: <Buffer 08 0f 0a 29 00 19 00 00> } +5ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +18ms
  serialport:unixRead waiting for readable because of code: EAGAIN +21ms
  serialport:poller Polling for "readable" +34ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +2ms
  zigbee2mqtt:debug 1/24/2019, 11:59:31 AM Received zigbee message of type 'attReport' with data '{"cid":"hvacThermostat","data":{"runningState":0}}' of device 'FC600' (0x001e5e090223703d)
  zigbee2mqtt:warn 1/24/2019, 11:59:31 AM No converter available for 'FC600' with cid 'hvacThermostat', type 'attReport' and data '{"cid":"hvacThermostat","data":{"runningState":0}}'
  zigbee2mqtt:warn 1/24/2019, 11:59:31 AM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 1/24/2019, 11:59:31 AM Received zigbee message of type 'devChange' with data '{"cid":"hvacThermostat","data":{"runningState":0}}' of device 'FC600' (0x001e5e090223703d)
  zigbee2mqtt:warn 1/24/2019, 11:59:31 AM No converter available for 'FC600' with cid 'hvacThermostat', type 'devChange' and data '{"cid":"hvacThermostat","data":{"runningState":0}}'
  zigbee2mqtt:warn 1/24/2019, 11:59:31 AM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.

Chrischi-, 2019-01-24:
yeah looks good...

add some fromZigbee converters to your `devices.js` and don't forget to make a PR whenn you successfully integrated the device.

frozenwave11, 2019-01-25:
@Chrischi- Hi , I'm testing yours implementation (fromZigbee.js) for bitron thermostat, so far everything is working. 
About file toZigbee do you think I could also use your code lines, of not how should I know which one to use?

Chrischi-, 2019-01-26:
You can use mine or you can take the converters from the ecozy. 

You can take all converters you need, they are normally declared in the ZCL and should work accordingly. You can test this with a "get" if you get values for the part you made a "get".

if you not now which one to use, take a look here: https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/information/mqtt_topics_and_message_structure.md#ecozy-smart-heating-thermostat-1tst-eu-bitron-wireless-wall-thermostat-with-relay-av201032

closed 2019-04-03 by stale[bot]

eemenda, 2020-05-25:
Hi, Finally you got it?
It's salus FC600 integrated in zigbee2mqtt ?
How can I add it?
Thanks

gavlexx, 2020-06-02:
> Hi, Finally you got it?
> It's salus FC600 integrated in zigbee2mqtt ?
> How can I add it?
> Thanks

Also interested in...

Peppito, 2021-02-28:
> > Hi, Finally you got it?
> > It's salus FC600 integrated in zigbee2mqtt ?
> > How can I add it?
> > Thanks
> 
> Also interested in...

Hi @Chrischi- , did you reach the end with the implementation? I would love to reuse it. Bought a new flat with 3x salus fc600. I also have the gateway but would like to integrate it into hass.

gavlexx, 2021-02-28:
I failed to integrate Salus FC-600 into my smart house with Home Assistant and replaced it with Moes BAC-002. It works like a charm especially with alternative firmware from [klausahrenberg](https://github.com/klausahrenberg/WThermostatBeca) or [fashberg](https://github.com/fashberg/WThermostatBeca)