Koenkk/zigbee2mqtt #919: Mqtt lights state not refrected on Home Assistant

P2mb2, 2019-01-22:
Hi,

I have HA Hassio instalation and zigbee2mqtt edge. Every Mqtt that i have works fine, except lights. I have one xiaomi aqara zigbee light and a philips hue lightstrip V2, both were discovered by zigbee2mqtt and if i call services on/off on home assistant, they goes on and off correctly but the switch on the HA interface is always showed as Disable/off. When i click the frontend light switch, it turns on (blue), the bulb turns on, but then the HA switch goes off again (grey), and the bulb stays on.
Does anyone have the same behaviour?

Any sugestion on how to investigate this?
Thanks
JC

khzd, 2019-01-23:
Hi
A way would be to check the debug output. Just set debug on in configuration and restart zigbee2mqtt. Post the log concerning the on off request for both devices 
Cheers

BudBundi, 2019-01-23:
I had the same problem with some Gledopto lights, when using this lights in a group (new feature) and create the required mqtt light in HA this light will show the right state and color. Just my two cents.

P2mb2, 2019-01-23:
When I click on the frontend of HA to turn this Aqara Zigbee Bulb on, i get the switch blue and the light goes on. After 2 seconds the switch comes back to off and the light stays on.

The Zigbee2mqtt log:
```
2019-1-23 21:45:39 - debug: Received MQTT message on 'zigbee2mqtt/0x00158d0002b7d9ec/set' with data 'ON'
2019-1-23 21:45:39 - info: Zigbee publish to device '0x00158d0002b7d9ec', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-1-23 21:45:39 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002b7d9ec', payload '{"state":"ON","linkquality":55,"last_seen":"2019-01-23T21:44:29.758Z","brightness":255,"color_temp":153}'
2019-1-23 21:45:39 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.light.aqcn02' (0x00158d0002b7d9ec)
```

I have a lot of this:

```
2019-1-23 21:46:55 - debug: Check online 0x00158d0002b7d9ec 0x00158d0002b7d9ec
2019-1-23 21:47:55 - debug: Check online 0x00158d0002b7d9ec 0x00158d0002b7d9ec
```

The same behaviour with my Philips LightStrip:

```
2019-1-23 21:50:44 - debug: Received MQTT message on 'zigbee2mqtt/kc_philips_striplight/set' with data 'ON'
2019-1-23 21:50:44 - info: Zigbee publish to device '0x0017880103f7896b', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-1-23 21:50:44 - info: MQTT publish: topic 'zigbee2mqtt/kc_philips_striplight', payload '{"state":"ON","last_seen":"2019-01-23T21:39:24.344Z"}'
2019-1-23 21:50:45 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'LST002' (0x0017880103f7896b)
2019-1-23 21:50:45 - info: MQTT publish: topic 'zigbee2mqtt/kc_philips_striplight', payload '{"state":"ON","last_seen":"2019-01-23T21:50:45.119Z"}'
```

Any ideas ?

Koenkk, 2019-01-24:
What HA version are you using?

P2mb2, 2019-01-24:
I have the 0.82.1

Koenkk, 2019-01-24:
@P2mb2 Is it fixed when updating to the latest HA version? (0.86)

BudBundi, 2019-01-24:
@Koenkk for me the problem persists on HA 0.86.1 for the devices in the discovery. But the mqtt-group is okay.