Koenkk/zigbee2mqtt #924: GLEDOPTO GL-C-008 is killing my mesh (not responding)

davekr8, 2019-01-23:
I can reproduce the strange behavior:

When switching the GL-C-008 on my sensor-actor communication seems to be blocked (buffered) somehow: for example, if I press one of my HUE dimmer switches, the flow (node-red) to control a HUE light is not working anymore.

After taking out the GL-C-008 (power off), all the button presses, seems to be processes (like queued) and the light is controlled. After a few seconds the mesh is fully functional again.

Can someone give me a hint, what is going wrong? Or how I can proceed to find the fault?

My logs says I am on 

`zigbee2mqtt version 0.2.0 (commit #753e8c3)`

Here is the json block (database.db) for the specific device:

```
{
  "id": 48,
  "type": "Router",
  "ieeeAddr": "0x00124b001a00321e",
  "nwkAddr": 54540,
  "manufId": 0,
  "epList": [
    11,
    13
  ],
  "status": "offline",
  "joinTime": null,
  "endpoints": {
    "11": {
      "profId": 49246,
      "epId": 11,
      "devId": 528,
      "inClusterList": [
        0,
        3,
        4,
        5,
        6,
        8,
        768
      ],
      "outClusterList": [],
      "clusters": {
        "genBasic": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "genIdentify": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "genGroups": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "genScenes": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "genOnOff": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "genLevelCtrl": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "lightingColorCtrl": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        }
      }
    },
    "13": {
      "profId": 49246,
      "epId": 13,
      "devId": 528,
      "inClusterList": [
        4096
      ],
      "outClusterList": [
        4096
      ],
      "clusters": {
        "lightLink": {
          "dir": {
            "value": 3
          },
          "attrs": {}
        }
      }
    }
  },
  "_id": "MKAYr4RGYsVPuTzr",
  "manufName": "GLEDOPTO",
  "powerSource": "Mains (single phase)",
  "modelId": "GL-C-008",
}
```

Koenkk, 2019-01-24:
These clogging issues should be fixed in the latest dev branch, can you check?

davekr8, 2019-01-24:
I upgraded to the latest and greatest dev commit. Still the same.
Even after removing the GLEDOPTO from the mesh (still powered on) it disturbs my remaining mesh, see:
```
2019-1-24 22:10:10 - info: Zigbee publish to device '0x00178801036b9739', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-1-24 22:10:11 - info: MQTT publish: topic 'home/zigbee/HUE_bulb_sofa_down_2', payload '{"state":"ON","brightness":254}'
2019-1-24 22:10:11 - info: Succesfully configured 0x7cb03eaa00ad2d23 (0x7cb03eaa00ad2d23)
2019-1-24 22:10:12 - info: Succesfully configured 0x84182600000ed63f (0x84182600000ed63f)
2019-1-24 22:10:13 - info: MQTT publish: topic 'home/zigbee/0x7cb03eaa00ad2d23', payload '{"state":"OFF","linkquality":0}'
2019-1-24 22:10:18 - info: Succesfully configured 0x84182600001030dd (0x84182600001030dd)
2019-1-24 22:10:18 - info: MQTT publish: topic 'home/zigbee/0x84182600001030dd', payload '{"state":"OFF","linkquality":55}'
2019-1-24 22:10:19 - info: MQTT publish: topic 'home/zigbee/0x84182600000ed63f', payload '{"state":"OFF","linkquality":23}'
2019-1-24 22:10:31 - info: Succesfully configured 0x84182600000fa9af (0x84182600000fa9af)
2019-1-24 22:10:32 - info: MQTT publish: topic 'home/zigbee/0x84182600000fa9af', payload '{"state":"OFF","linkquality":23}'
2019-1-24 22:10:41 - error: Failed to configure 0x7cb03eaa00ad2a55 (0x7cb03eaa00ad2a55) ('Error: Timed out after 10000 ms')
2019-1-24 22:10:50 - error: Failed to configure HUE_switch_2 (0x0017880103cb4dc4) ('undefined')
2019-1-24 22:11:00 - error: Failed to configure 0x84182600000f0c49 (0x84182600000f0c49) ('Error: Timed out after 10000 ms')
2019-1-24 22:11:00 - error: Failed to configure HUE_switch_1 (0x0017880103e5ee4a) ('undefined')
```
It somehow takes over some control. Strange. Try to reset the device.

davekr8, 2019-01-24:
After resetting and joining the network again, I see the communication still being disturbed.
Btw: I have a relative stable mesh with 5 routers (in every corner of out flat). Have never seen any errors and very rarly I see an sensor-actor relation not working or with a notable delay.

Koenkk, 2019-01-26:
This is indeed really strange, could you do some zigbee sniffing, perhaps this could give us a pointer. https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/how_tos/how_to_sniff_zigbee_traffic.md

closed 2019-04-03 by stale[bot]