Koenkk/zigbee2mqtt #927: Hangs on router status

oskarn97, 2019-01-24:
After upgrading to the latest firmware and version of zigbee2mqtt, after a few hours, no more packages from devices are coming in because it hangs up when requesting the router status. After a reboot, everything works again for a few hours.

`  zigbee2mqtt:debug 2019-1-24 10:49:35 Check online routerOsmc 0x00124b0018ed1934
  zigbee2mqtt:debug 2019-1-24 10:49:35 Check online routerWz 0x00124b0018ed22b6
  zigbee2mqtt:debug 2019-1-24 10:49:35 Check online routerBett 0x00124b001b7b1dbe
  zigbee2mqtt:debug 2019-1-24 10:49:35 Check online routerSz 0x00124b0014aa777a
  zigbee2mqtt:debug 2019-1-24 10:49:35 Check online routerCC2591 0x00124b001bc743cc
  zigbee2mqtt:debug 2019-1-24 10:49:35 Check online routerFlo 0x00124b001bd12020
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +55s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +3s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +2s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee2mqtt:debug 2019-1-24 10:50:35 Check online routerKu 0x00124b001936b370
  zigbee2mqtt:debug 2019-1-24 10:50:35 Check online routerOsmc 0x00124b0018ed1934
  zigbee2mqtt:debug 2019-1-24 10:50:35 Check online routerWz 0x00124b0018ed22b6
  zigbee2mqtt:debug 2019-1-24 10:50:35 Check online routerBett 0x00124b001b7b1dbe
  zigbee2mqtt:debug 2019-1-24 10:50:35 Check online routerSz 0x00124b0014aa777a
  zigbee2mqtt:debug 2019-1-24 10:50:35 Check online routerCC2591 0x00124b001bc743cc
  zigbee2mqtt:debug 2019-1-24 10:50:35 Check online routerFlo 0x00124b001bd12020
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +55s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +2ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +3s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +2s
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +2ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +0ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +1ms`

Koenkk, 2019-01-24:
When commenting https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L38 does it keep working? (so we can confirm this is indeed the root cause).

oskarn97, 2019-01-24:
Will try!

oskarn97, 2019-01-25:
Yes, that works!

closed 2019-04-03 by stale[bot]