Koenkk/zigbee2mqtt #928: build faild

DerDoubleD, 2019-01-24:
Hello

Sorry for my bad english

the "npm install" ends with error.
See https://hastebin.com/obuzixixuy.coffeescript

Package: bl   

node --version
v10.15.0
npm --version
6.7.0

Can you please help me!?

Koenkk, 2019-01-24:
this is a warning which indicates dependies with security issues, this will automatically be solved when the dependencies of zigbee2mqtt are updated.

DerDoubleD, 2019-01-25:
Good morning.

Thanks for your answer.
What can i do to get it work.
Because all my devices dosn't work since the update.
Or can i rollback?
Thanks

Best regards
Daniel

Koenkk, 2019-01-25:
Can you provide the log when executing `npm start`

DerDoubleD, 2019-01-25:
Thank you for supporting me.
In the log I see no errors but no device gets data
The file state.json is empty.

Here is the log.

------------------------------ log----------------------------------
 npm start

> zigbee2mqtt@1.0.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2019-1-25 17:11:13 Logging to directory:
'/opt/zigbee2mqtt/data/log/2019-01-25.17-11-13'
  zigbee2mqtt:info 2019-1-25 17:11:14 Starting zigbee2mqtt version 1.0.1
(commit #43fc66c)
  zigbee2mqtt:info 2019-1-25 17:11:14 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-1-25 17:11:18 zigbee-shepherd started
  zigbee2mqtt:info 2019-1-25 17:11:18 Coordinator firmware version:
'undefined'
  zigbee2mqtt:info 2019-1-25 17:11:18 Currently 10 devices are joined:
  zigbee2mqtt:info 2019-1-25 17:11:18 Xiaomi_Temperatur_Aussen_Nord
(0x00158d00020ee251): WSDCGQ01LM - Xiaomi MiJia temperature & humidity
sensor  (EndDevice)
  zigbee2mqtt:info 2019-1-25 17:11:18 Xiaomi_Temperatur_2
(0x00158d00020f2705): WSDCGQ01LM - Xiaomi MiJia temperature & humidity
sensor  (EndDevice)
  zigbee2mqtt:info 2019-1-25 17:11:18 Xiaomi_Button_1 (0x00158d00020f1af2):
WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
  zigbee2mqtt:info 2019-1-25 17:11:18 Xiaomi_Fensterkontakt_2
(0x00158d00023dbb2f): MCCGQ01LM - Xiaomi MiJia door & window contact sensor
(EndDevice)
  zigbee2mqtt:info 2019-1-25 17:11:18 Xiaomi_Fensterkontakt_Garagentor
(0x00158d000239c7f8): MCCGQ01LM - Xiaomi MiJia door & window contact sensor
(EndDevice)
  zigbee2mqtt:info 2019-1-25 17:11:18 Xiaomi_Fensterkontakt_4
(0x00158d00022f13cc): MCCGQ11LM - Xiaomi Aqara door & window contact sensor
(EndDevice)
  zigbee2mqtt:info 2019-1-25 17:11:18 Xiaomi_Fensterkontakt_5
(0x00158d0002286b0d): MCCGQ11LM - Xiaomi Aqara door & window contact sensor
(EndDevice)
  zigbee2mqtt:info 2019-1-25 17:11:18 Xiaomi_Fensterkontakt_3
(0x00158d00022f144e): MCCGQ11LM - Xiaomi Aqara door & window contact sensor
(EndDevice)
  zigbee2mqtt:info 2019-1-25 17:11:18 0x00158d000213179c
(0x00158d000213179c): WXKG02LM - Xiaomi Aqara double key wireless wall
switch (EndDevice)
  zigbee2mqtt:info 2019-1-25 17:11:18 Xiaomi_Rauchmelder_Hobby
(0x00158d00028e23fd): JTYJ-GD-01LM/BW - Xiaomi MiJia Honeywell smoke
detector (EndDevice)
  zigbee2mqtt:warn 2019-1-25 17:11:18 `permit_join` set to  `true` in
configuration.yaml.
  zigbee2mqtt:warn 2019-1-25 17:11:18 Allowing new devices to join.
  zigbee2mqtt:warn 2019-1-25 17:11:18 Set `permit_join` to `false` once you
joined all devices.
  zigbee2mqtt:info 2019-1-25 17:11:18 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-1-25 17:11:18 Connecting to MQTT server at
mqtt://localhost
  zigbee2mqtt:info 2019-1-25 17:11:18 zigbee-shepherd ready
  zigbee2mqtt:info 2019-1-25 17:11:18 Connected to MQTT server
  zigbee2mqtt:info 2019-1-25 17:11:18 MQTT publish: topic
'zigbee2mqtt/bridge/state', payload 'online'

------------------------------ log----------------------------------

Gruß
Daniel Deml

Koenkk, 2019-01-26:
Can you post a bit more of your logging? also when trigging the sensors.

DerDoubleD, 2019-01-26:
There is no more. After 2 hour the same.
I will change the loglevel to debug and try again.

Koenkk, 2019-01-26:
`zigbee2mqtt:info 2019-1-25 17:11:18 MQTT publish: topic
'zigbee2mqtt/bridge/state', payload 'online'` is the last line of the log?

DerDoubleD, 2019-01-26:
Yes.

Koenkk, 2019-01-26:
What coordinator firmware version are you using? Could you update to the latest?

DerDoubleD, 2019-01-27:
The firmware is from march 2018.
I ordered a ccdebugger and then i want to update it.

Thank You!

DerDoubleD, 2019-03-07:
With the new firmware the problem is solved.
Thanks for the support

closed 2019-03-07 by DerDoubleD