Koenkk/zigbee2mqtt #929: Unsupported device: Salus SP600 Smart Plug with Power Monitoring

MarkAdamson, 2019-01-24:
Hi, I'm trying to add support for this device, but running into difficulty. 

So far, I've paired the plug with zigbee2mqtt, and added it to devices.js as follows:
```    // Salus
    {
        zigbeeModel: ['SP600'],
        model: 'SP600',
        vendor: 'Salus',
        description: 'Smart plug',
        supports: 'on/off, power measurement',
        fromZigbee: [],
        toZigbee: [],
    },
```

I hoped to then see some 'No converter available' messages, but no such luck - no messages from the plug. I guess that I need to add a configure section, but I don't really know what (or how to figure out what) to put in there. I've tried copy/pasting sections from potentially similar devices, but I just get 'failed to configure' in the log.

Here is the device entry from database.db:
https://pastebin.com/M4DvyDUC

Here is my fairly uninformative log:
https://pastebin.com/Jw0vy1zV

Please let me know if there's anything else you need.

Koenkk, 2019-01-26:
Can you try with:

```js
    {
        zigbeeModel: ['SP600'],
        model: 'SP600',
        vendor: 'Salus',
        description: 'Smart plug',
        supports: 'on/off, power measurement',
        fromZigbee: [],
        toZigbee: [],
       configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 9);
            const onOff = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 5, repChange: 0};
            const actions = [
                  (cb) => device.foundation('genOnOff', 'configReport', [onOff], foundationCfg, cb),
                  (cb) => device.bind('genOnOff', coordinator, cb),
                  (cb) => device.report('seMetering', 'instantaneousDemand', 1, 5, 1, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

MarkAdamson, 2019-01-28:
Works a treat, thanks! I'll sort out a PR soon.

Can I ask, how do you know what actions to set up in the configure section? Can you figure it out just from the database and log, or does it require some knowledge of the zigbee protocol? And if so do you have any suggestions of things to read to get a better understanding?

I can see myself possibly wanting to add more devices in the future, and would like to be able to make better headway on my own if I could.