Koenkk/zigbee2mqtt #931: CC2531 and RFX2401C - compatible with zigbee2mqtt firmware ?

sharkillator, 2019-01-25:
Hi,

I orderd this device [CC2531-RFX2401C-PA-LNA-RF-Front-End-](https://www.ebay.fr/itm/CC2531-RFX2401C-PA-LNA-RF-Front-End-2-4GHz-IEEE-802-15-4-ZigBee-USB-dongle/141941450431?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649)
On this board CC2531 and RFX2401C are connected in this way:
CC2531 P1_1 (9 pin) -> RFX2401C TXEN (5 pin)
CC2531 P1_4 (6 pin) -> RFX2401C RXEN (6 pin)

Will CC2531 firmware be able to use the RFX2401C ? I would really like to be able to use such a device as classic CC2531 antenna is not very powerful.
Thanks for your help.

Sharki

Koenkk, 2019-01-25:
It's probably not compatible, I would personally recommend a CC2530+CC2591 or CC2530+RFX2401 (which are compatible and widely used).

sharkillator, 2019-01-25:
I really would have preferred the cc2531 format.

If I buy a 2530, this one would be ok ? [Cc2530](https://s.click.aliexpress.com/e/cbfFxJbj)

sharkillator, 2019-01-25:
Do you think it would be difficult to make the firmware compatible with the cc2531and rfx2501c ?

Koenkk, 2019-01-25:
I don't know how hard this would be. Instructions for the current firmware can be found here: https://github.com/koenkk/z-stack-firmware. I guess it should work with some minor changes.

closed 2019-04-02 by stale[bot]