Koenkk/zigbee2mqtt #932: Possible solutions to inconveniences when adding a new device?

mihalski, 2019-01-25:
Every time I add a new device (Xiaomi in my case) it has a name like 0x00158d0001135dc0. Before I get a chance to rename it gets auto-discovered by Home Assistant and then I need to:
- rename it via zigbee2mqtt,
- stop Home Assistant, 
- purge the entry from the .storage/core.device_registry,
- restart zigbee2mqtt, 
- start Home Assistant

Is there a better way? Or would it be worth adding a setting much like allow/deny joining that allows new devices to be held back from being published to mqtt until they have been named?

Landrash, 2019-01-25:
Don't start Home Assistant until you renamed your device?

mihalski, 2019-01-26:
That has the same disadvantages (ie only a few times in a week I can risk stopping Home Assistant).

Koenkk, 2019-01-26:
- Stop zigbee2mqtt
- Set `homeassistant: false`
- Start zigbee2mqtt
- Pair device
- Stop zigbee2mqtt
- Set `friendly_name` and `homeassistant: true`
- Start zigbee2mqtt
- Enjoy!

Pockrandt, 2019-01-26:
use the new Device Registry for painless renaming 

1. update your home assistant 
2. let them join and be discovery by their id 
3. change Name AND entity id from the UI with the cogwheel
(use configuration > Integrations > MQTT )

mihalski, 2019-01-26:
That's the problem though. The device name is NOT editable.

Pockrandt, 2019-01-26:
what kind of device do you want to rename?
no issues with
* Aqara human body movement and illuminance sensor (RTCGQ11LM)
* Aqara door & window contact sensor (MCCGQ11LM)
* Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)
* Aqara wireless switch (WXKG11LM)
* and others like osram and tradfri

mihalski, 2019-01-26:
Home Assistant provides no way to rename a device. Thus this situation:

[image]

Pockrandt, 2019-01-27:
sorry for the misunderstanding, through with device renaming you mean the child's elements.

btw why do yo need to rename the root device? looking at the logs?

mihalski, 2019-01-27:
I have about 60 devices auto-discovered via MQTT and I expect that to rise to at least 80 or so. It's MUCH easier to find them when they are alphabetically sorted together with other related devices.

closed 2019-04-04 by stale[bot]