Koenkk/zigbee2mqtt #934: MCCGQ01LM: Device with modelID 'undefined' is not supported.

sylarevan, 2019-01-25:
Hi there,

I am discovering zigbee2mqtt's world and I am now trying to pair all my Xiaomi devices with it.
So far, all seems OK, except one Xiaomi door sensor MCCGQ01LM which should be supported according to the list of supported devices.

I get the following message:
`Jan 25 16:51:56 raspberrypi npm[2242]:   zigbee2mqtt:warn 1/25/2019, 4:51:56 PM Device with modelID 'undefined' is not supported.
Jan 25 16:51:56 raspberrypi npm[2242]:   zigbee2mqtt:warn 1/25/2019, 4:51:56 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html`

So if I understand correctly, the device is paired, but not recognized. 

Could you help me find why? Is there some different hardware version of the same sensor?

sylarevan, 2019-01-25:
FYI, here is what I have in database.db:
`{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d00026aed62","nwkAddr":51867,"manufId":4151,"epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":26\
0,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"v\
alue":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":\
{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"s2tv0HgrnG6Xp6MI"}`

tringler, 2019-01-25:
I have the same problem with the Xiaomi Button, but successfully paired the door sensor and temperature sensor.

Dreamray, 2019-01-26:
Same device, same problem, I tried many times, the last time, I reinstalled zigbee2mqtt and ran zigbee2mqtt through 'npm start' instead of 'systemctl start zigbee2mqtt', although the error is displayed, but the pairing has been successful. I don't know if 'npm start' made it work, but you can try it. 
Sorry for my english, I translated it by google

sylarevan, 2019-01-26:
OK, so what you mean is that a device can be in some way "partially" paired to the coordinator? Anyway, I will try your instructions for pairing Xiaomi devices, which are quite different from the official how-to.

sylarevan, 2019-01-26:
I have removed the device from the network by publishing to zigbee2mqtt/bridge/config/remove, stopped zigbee2mqtt, removed the device from database.db and restarted zigbee2mqtt.
After carefully applying the pairing method by @tringler, I confirm the door sensor is now correctly recognized.

So in the end, it was only some kind of a "bad pairing", which results in the device to be actually included in the network without being properly recognized.

closed 2019-01-26 by sylarevan