Koenkk/zigbee2mqtt #936: Error: Timed out after 10000 ms

Smarti15, 2019-01-25:
Hallo, ich hab einen CC2531 am ioBroker verbunden und dort bereits erfoglreich mit dem ZigBee Adapter einen Mi SmartPlug eingerichtet. Danach hab ich einen Temperatur Sensor von Xiaomi über das SmartPlug angemeldete. Heute wollte ich einen Zweiten Sensor anmelden, indem ich auf den Paarungs Button am SmartPLug im Adapter geklickt habe. Leider erscheint dort kein Counter und nach ein paar Sekunden die Meldung: Error: 
Join failure.. Error: Timed out after 10000 ms
Pairing started 0x00158d0002835711

JLFN, 2019-01-25:
Not paired try it until you pair it

Smarti15, 2019-01-25:
Hi JLFN,

I do not understand. What am I doing wrong. can you explain that to me briefly?

thx

tringler, 2019-01-25:
I have the same problem - I paired all my Xiaomi temperature sensors, but with my buttons and door sensors I get these errors:
```
  zigbee2mqtt:error 2019-1-25 22:37:49 Cannot get the Node Descriptor of the Device: 0x00158d00015aa7c9 (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 2019-1-25 22:38:09 Cannot get the Node Descriptor of the Device: 0x00158d00015aa7c9 (Error: Timed out after 10000 ms)
```

I had the same issues with the temperature sensors until I shut down my old Xiaomi gateway and thought that blocked my pairings. But now I have the same error with the buttons and door sensors.

closed 2019-04-02 by stale[bot]

rsalunga29, 2019-11-12:
Thanks for this @tringler , I have mine worked at well. @Koenkk can we update the documentation for this device? Documentation can be found here: https://www.zigbee2mqtt.io/devices/MCCGQ01LM.html
> It is all about timing (at least for the door shutter):
> Step 1:
> Hold the reset button up to 5 seconds until it blinks 3x times, release it and it should blink 1x time
> Step 2:
> Press reset button and WAIT until it blinks 2x times
> --> Repeat Step 2 20x times
> 
> It will not work if you not wait until the 2x times blinks in Step 2.
> 
> Did I miss it in the documentation? Can we add that?

Koenkk, 2019-11-12:
Shouldnt be needed in zigbee2mqtt 1.7.0 anymore.