Koenkk/zigbee2mqtt #939: AduroSmart Eria

gi-el, 2019-01-25:
Hi!

I'm in the process of adding support for AduroSmart Eria devices. I have a wireless switch and some lamps. At the moment I'm running into two issues, that might be related.

Zigbee2mqtt is running in a Docker container with the image `koenkk/zigbee2mqtt:1.0.1-arm32v6`.

First, the switch. I managed to pair it, and add some lines in `devices.js`. Now, when zigbee2mqtt starts, at least I see:
`0x00158d000198d02e (0x00158d000198d02e): 81825 - AduroSmart Eria wireless dimming switch (EndDevice)`

However, any button press results in:
```
/app/node_modules/zcl-packet/lib/functional.js:27
        throw new Error('Unrecognized cluster');
        ^

Error: Unrecognized cluster
    at new FuncPayload (/app/node_modules/zcl-packet/lib/functional.js:27:15)
    at /app/node_modules/zcl-packet/lib/zcl.js:33:22
    at Dissolve.<anonymous> (/app/node_modules/zcl-packet/lib/zcl.js:121:9)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at Dissolve.emit (events.js:211:7)
    at Dissolve.<anonymous> (/app/node_modules/dissolve-chunks/index.js:73:29)
    at emitNone (events.js:106:13)
    at Dissolve.emit (events.js:208:7)
    at emitReadable_ (/app/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:456:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.0.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-01-25T22_37_27_505Z-debug.log
```

As per https://github.com/Koenkk/zigbee2mqtt/issues/852#issuecomment-454089241 I've added the `console.log` line in `af.js`, but no more logging appears.

The second issue is that I cannot pair the lamps at all. When I put one in pairing mode (and the lamp confirms so by flashing as per the user manual), nothing happens on the zigbee2mqtt side. Not sure what to do next here.

Any help is appreciated. These devices seem a good cheap alternative to Philips Hue lamps - and more importantly (to me) they claim a CRI of 90, so the white light should be of better quality.

gi-el, 2019-01-25:
With some logging in `zcl-packet/lib/functional.js` it looks like `clusterID` is `64716`.

gi-el, 2019-01-26:
I'm able to pair with the lamps if they are really close to the controller - did not realize that was an issue as I never encountered that with ZWave.

The lamp shows up as `ZLL-ExtendedColo`, which makes me believe it's Zigbee Light Link. However, when I use the `generic.light_onoff_brightness_colortemp_colorxy` convertors, messages sent are timed out, and no messages seem to be received. When the lamp is turned on, the log shows only this:
```
zigbee-shepherd spinlock: false []
zigbee-shepherd Device: 0x00158d0001a5bad8 already in network
zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd
zigbee2mqtt:debug 2019-1-26 08:26:38 Check online 0x00158d0001a5bad8 0x00158d0001a5bad8
zigbee-shepherd:request REQ --> ZDO:nodeDescReq
zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp
```

I put in a Home Assistant mapping, and the light shows up as 'off', as no messages about its status are received or sent. Trying to turn the light 'on' (while it in reality already is on):
```
zigbee2mqtt:debug 2019-1-26 08:46:06 Received MQTT message on 'zigbee2mqtt/0x00158d0001a5bad8/set' with data '{"state": "ON"}'
zigbee2mqtt:info 2019-1-26 08:46:06 Zigbee publish to '0x00158d0001a5bad8', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee-shepherd:request REQ --> AF:dataRequest, transId: 19
zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 19
zigbee2mqtt:error 2019-1-26 08:46:36 Zigbee publish to '0x00158d0001a5bad8', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms
```

I will keep messing with this, but any help is appreciated.

Koenkk, 2019-01-26:
Could you post the `database.db` entry of this device?

gi-el, 2019-01-26:
The light, that seems to do very little:
```
{"id":5,"type":"Router","ieeeAddr":"0x00158d0001a5bad8","nwkAddr":34555,"manufId":4151,"manufName":"Trust International B.V.\u0000","powerSource":"Mains (single phase)","modelId":"ZLL-ExtendedColo","epList":[1,2],"status":"online","joinTime":1548519855,"endpoints":{"1":{"profId":49246,"epId":1,"devId":4096,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":49246,"epId":2,"devId":528,"inClusterList":[0,3,4,5,6,8,25,768,65535],"outClusterList":[0,3,4,5,6,8,25,768],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":3},"attrs":{}},"genScenes":{"dir":{"value":3},"attrs":{}},"genOnOff":{"dir":{"value":3},"attrs":{}},"genLevelCtrl":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"lightingColorCtrl":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"jYs4aKBZWQXrGlzc"}
```

The switch, that crashes `zigbee2mqtt` because of "Unknown cluster":
```
{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d000198d02e","nwkAddr":37999,"manufId":4151,"manufName":"ADUROLIGHT","powerSource":"Battery","modelId":"Adurolight_NCC","epList":[1],"status":"online","joinTime":1548543783,"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,3,8,4096,64716],"outClusterList":[3,4,6,8,4096,64716],"clusters":{"64716":{"dir":{"value":3},"attrs":{}},"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":3},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"f25iNaccbDXCgCLF"}
```

Koenkk, 2019-01-27:
Lets start with the light.

- Can you post a link to this device?
- What features does this light support (e.g. color temperature, color, etc.)?
- Can you try with the following in `devices.js`:
```js
    {
        zigbeeModel: ['ZLL-ExtendedColo'],
        model: 'TODO',
        vendor: 'AduroSmart Eria',
        description: 'TODO',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
        ep: (device) => {
            return {
                '': 2,
            };
        },
    },
```

gi-el, 2019-01-27:
When instead of the `extend` I add the `fromZigbee` and `toZigbee` lines for the generic light, it works! Brightness, on/off, color and color temperature all as expected.

Koenkk, 2019-01-28:
Great, could you make a PR? Then we can continue with the next device.

gi-el, 2019-01-28:
I'm happy to. There are some (minor) issues - we can discuss those on the PR.

gi-el, 2019-01-28:
One issue that I think is not related to the code, so discussing here:

I have three of these bulbs and they are very close together. Looking at the network map, one is connected to the controller, the two others are connected to the first one acting as a router. One of the lamps (consistently the same one) that is connected through the router sometimes gets in a state where it doesn't seem to accept commands. The log shows
```
zigbee2mqtt:error 2019-1-28 09:37:51 Zigbee publish to '0x00158d0001990169', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 2 failed with error Error: AF data request fails, status code: 233. MAC no ack.
```

This happened a few times, and I seemed to be able to get it working again by turning the physical power off and on to the lamp, and restarting zigbee2mqtt. Or so I thought... it seems to be the _exact_ physical position of the lamp. Switching out one lamp with another, the lamp in the same position now becomes non-responsive. If I move the lamp literally an inch up or down, it works. In that exact same spot, it doesn't. It shows online in the network map, has good link quality, connects through a router (other lamp) that's very close, but doesn't accept commands. Hold it an inch higher - all good. Have you ever seen something like this before? Happy to move this to a new issue if you prefer, because while troubleshooting and typing I realize it has little to do with the original issue.

Koenkk, 2019-01-28:
Can you update to the latest dev branch? This has a more complete network map.

gi-el, 2019-01-28:
Looks like they're all connected to each other as routers with good link quality, but also to the controller with close to zero link quality. Assuming depending on the exact location some messages don't get through, and aren't routed through the routers as there also is a direct connection. Is there anything that can be done about that (e.g., retry through a router)? Not that big of an issue - as said, I'm already in the process of getting an RF+antenna setup and this is only relevant if you're at the edge of the network range I suppose...

Working on PR against latest dev branch - only issue I see is that the devices don't seem to update status after setting it - i.e., the `fromZigbee` functions don't get called under normal operation and hence the state doesn't include brightness, color, etc, but only ON or OFF. `fromZigbee` functions do get called when requesting `zigbee2mqtt/<dev>/get` and the state is updated correctly. What's the advised way of automatically calling the get function after set (this all for proper Home Assistant integration)?

gi-el, 2019-01-29:
I think these lamps just don't send updates when their state is changed. Even with the lamps positioned close to the controller and the commands obviously succeeding, state only shows on or off and nothing else. When issueing an explicit get for brightness/color, the lamp responds and the state gets updated.

Koenkk, 2019-01-30:
Can you try with

```js
    // AduroSmart
    {
        zigbeeModel: ['ZLL-ExtendedColo'],
        model: '81809',
        vendor: 'AduroSmart',
        description: 'ERIA colors and white shades smart light bulb A19',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 2);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
        ep: (device) => {
            return {
                '': 2,
            };
        },
    },
```

gi-el, 2019-02-01:
Doesn't seem to fix anything. The only time the state actually seems to get updated (besides on/off, which always worked) is when explicitly publishing to the `get` topic. The color and brightness are not being updated on the `state` topic at any other moment.

Koenkk, 2019-02-02:
The brightness and color is never reported back, as Home Assistant doesn't need a confirmation for this (the slider will stay at the position where you moved it to).

gi-el, 2019-02-02:
1) Independent of Home Assistant, if I subscribe to a device's state topic I think the expected behavior would be to get updates when the device's state changes. If somebody else changes the state (by publishing to the `state/set` topci), I would probably expect a state update. With these lights, this isn't happening. This is an issue with the lights, but it's something we can likely work around in zigbee2mqtt
2) According to the documentation and my experience, Home Assistant does actually want confirmation. It tries to display the actual state of the light, not necessarily what it thinks it should be. This way, you're sure that your command has been received by the lamp. The docs for MQTT lights say: "When a state topic is not available, the light will work in optimistic mode. In this mode, the light will immediately change state after every command. Otherwise, the light will wait for state confirmation from the device (message from state_topic)."

So for this specific light, I see three solutions:
a) Set the Home Assistant MQTT light entry to optimistic. I've tried this (by changing the JSON entry for the light in `homeassistant.js`) and it works, it does give the expected behavior in HA, but it doesn't actually solve for 1) above.
b) Force the reading of the state over ZigBee after writing it. I've tried this as well by making new light entries for this light and adding `readAfterWriteTime: 100` in the new entries in `toZigbee.js`. This also works nicely and actually makes sure the current state is correctly reflecting what the light actually is doing
c) Implement some form of optimistic mode in zigbee2mqtt. If a device has "optimistic" set, we would publish the expected state once it has been successfully sent to the device (and cache it in `state.json`). I have not tried this.

My preference would be option b) above, as it actually gives the most accurate information and mimics what the light normally should be doing anyway. Let me know what you think and I'll make a PR for your preferred solution.

Then we can move on to the switch :).

Koenkk, 2019-02-02:
I think I misunderstood you.

The `state_topic` is used to determine the ON/OFF state of the device. Zigbee2mqtt published to this topic when the command succeed (!= confirmed by the device). This is because many devices don't confirm this. 

Can you confirm when the zigbee command succeed, the state is correctly updated?

gi-el, 2019-02-03:
Thanks for your help getting the lights working perfectly! I'm having some slightly different models on the way, so more will be supported soon!

Next, on to the remote. Could you give some pointers as to how to solve the unknown cluster error, or maybe just ignore it?

Koenkk, 2019-02-03:
Add a `console.log(clusterId, direction, cmd)` to https://github.com/Koenkk/zcl-packet/blob/master/lib/functional.js#L25 to find out the cluster ID. 

After that you have to define the cluster and it's commands like I did in: https://github.com/Koenkk/zcl-id/commit/4db172c253c3381bb009366cdc4208a25182607f and https://github.com/Koenkk/zcl-packet/commit/e233d2298488b4799c35663792e243ffbad6c6e8

gi-el, 2019-02-05:
I think we can close this issue - both the lights and the switch are supported now in the dev branch. Thanks for all your help!

closed 2019-02-05 by gi-el