Koenkk/zigbee2mqtt #942: Domoticz not starting after installing plugin

SBTephan, 2019-01-26:
After installing the plugin in domoticz, domoticz won'start.

this is wat the log said:

2019-01-26 14:49:19.893  Error: Domoticz received fatal signal 11 !...
2019-01-26 14:49:19.911  Error:   /home/pi/domoticz/domoticz() [0x262154]
2019-01-26 14:49:19.911  Error:   /home/pi/domoticz/domoticz(_Z14signal_handleri+0x4c) [0x26220c]
2019-01-26 14:49:19.912  Error:   /lib/arm-linux-gnueabihf/libc.so.6(__default_sa_restorer_v2+0) [0xb6c1b180]

Any ideas?

Best regards,
Stephan

deennoo, 2019-01-26:
Maybe asking to Domoticz plugin creator

SBTephan, 2019-01-26:
Didn't know that was a different developer. will post it there

Koenkk, 2019-01-26:
Please open an issue here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues

closed 2019-01-26 by Koenkk