Koenkk/zigbee2mqtt #944: Can a device work with zigbee2mqtt and Alexa at the same time?

giaffa86, 2019-01-26:
Hi guys,
I have configured a OSRAM Smart+ Plug on Home Assistant through zigbee2mqtt addons. But after this I wasn't able to keep it works with Echo Plus. Echo doesn't find the device anymore after the pairing with Home assistant. I just want to ask if this is a normal behavior and how deal with it.
Thanks in advance

sehraf, 2019-01-26:
That's normal behavior since the plug can be pared with either zigbee2mqtt or Echo Plus but not both at the same time.

Koenkk, 2019-01-26:
A zigbee device can only be in one zigbee network at a time. So unfortunately the answer is no.

Koenkk, 2019-01-26:
I've never used haaska, but https://www.home-assistant.io/components/alexa/ will work.

Koenkk, 2019-01-26:
Not for the manual setup.

dreimer1986, 2019-02-18:
Cloud and Manual setup work fine. Manual has the downside that you always have to say "Alexa, tell Home Assistant to switch Living room on" Cloud is as easy as you had it before though. I used Manual before Cloud existed and now put all the coded stuff on hold because Cloud simply works. ^^

closed 2019-04-26 by stale[bot]