Koenkk/zigbee2mqtt #946: Xiaomi MCCGQ01LM batterry

pavka28, 2019-01-26:


alcar21, 2019-01-27:
Hello, it takes a long time to show the battery level.

stoinov, 2019-03-04:
@Koenkk I think I figured the problem - the battery level is reported as `"100.00"` instead of `100.00` like all the other devices.

Koenkk, 2019-03-04:
@stoinov can you provide your zigbee2mqtt log? My log shows a number:

`
zigbee2mqtt:info 3/4/2019, 6:30:22 PM MQTT publish: topic 'zigbee2mqtt/veranda_climate_sensor', payload '{"temperature":7.14,"linkquality":70,"humidity":67.1,"pressure":994,"battery":74,"voltage":2955}'
`

The coresponding converter also forces it to a float (https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L294) so I don't understand how you got a string.

stoinov, 2019-03-04:
The sensor is contact one - MCCGQ01LM:
```
zigbee2mqtt:info 2019-3-4 18:40:59 livingroom/door (0x00158d00027b27c4): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
zigbee2mqtt:info 2019-3-4 18:40:59 MQTT publish: topic 'base/livingroom/door', payload '{"battery":"100.00","voltage":3035,"linkquality":136}'
```
It's missing from the file you linked to.

Koenkk, 2019-03-04:
It's strange indeed, I've just checked for the contact sensor:

```
zigbee2mqtt:info 3/4/2019, 6:55:55 PM MQTT publish: topic 'zigbee2mqtt/stairs_cupboard_door_sensor', payload '{"contact":true,"linkquality":70,"battery":100,"voltage":3015}'
```

Could you perhaps try to catch a message when the sensor reports this?

stoinov, 2019-03-04:
I haven't been using them for a while and it seem this was some stale record. After couple of activation I got a number:
```
zigbee2mqtt:info 2019-3-4 19:03:46 MQTT publish: topic 'base/livingroom/door', payload '{"battery":"100.00","voltage":3035,"linkquality":131,"contact":true}'
zigbee2mqtt:info 2019-3-4 19:03:48 MQTT publish: topic 'base/livingroom/door', payload '{"battery":100,"voltage":3025,"linkquality":128,"contact":true}'
```
I can confirm that the `state.json` was also updated properly

Koenkk, 2019-03-05:
Closing this as there is no response from OP.

closed 2019-03-05 by Koenkk