Koenkk/zigbee2mqtt #947: aqara battery

pavka28, 2019-01-26:
Unsupported device
First check: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html

Bug report
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

danpowell88, 2019-01-27:
I think you are missing some details here

pavka28, 2019-01-27:
sorry....

how often battery status, values  is updated

closed 2019-04-04 by stale[bot]