Koenkk/zigbee2mqtt #949: Aqara MCCGQ11LM sensor not being routed

clockbrain, 2019-01-27:
Not a bug report yet - more of a straw poll of how widespread this issue may be.

I'm finding that all of my Aqara door and window sensors will only communicate directly with the cc2531 coordinator and refuse to use various routers available and much closer to them. Other end devices (Hue dimmer switches) will talk via those routers just fine.

It isn't the coordinator device limit as I have only 12 devices at the moment. The most recent 4 devices are the Aqara MCCGQ11LM sensors and they all behave the same, i.e. only communicating directly with the cc2531 coordinator. When I paired them they were close to the coordinator but there was also a router less than 1 metre away. It was also the same arrangement as when I paired the Hue dimmer switches so I don't think the pairing process was an issue.

I note there have been various issues raised over time about other Xiaomi mains devices not functioning as routers but my issue is that the end devices themselves won't route. I've also seen comments to the effect that Xiaomi devices aren't fully ZCL compliant. Could this be the issue?

I'm running dev zigbee2mqtt version 1.0.1 (commit #dd6e34a) on firmware version: '20180815'

This isn't an issue with the network map code. I see the same picture when I plot my network with Digi XCTU. The top 4 end devices are the Aqara sensors. The other 2 end devices are Hue dimmer switches.
[image]

 Does anyone else see the same non-routing issue with the Aqara MCCGQ11LM sensors?

Koenkk, 2019-01-27:
The MCCGQ11LM is an end-device which cannot route.

If you mean: why doesn't my MCCGQ11LM connect to the nearest router?
- After pairing, these device (sometimes?) don't try to find a better route.
- The choice of router can be influenced by re-pairing it close to the router.

clockbrain, 2019-01-28:
Yes, I meant why doesn't the MCCGQ11LM connect to the nearest router?

I've since done some more reading and realised my assumption about self-healing of the mesh for end devices was incorrect. End devices only pair with a single router or coordinator and stay paired with that node for life (until re-paired).

Perhaps the Zigbee network documentation page for end devices should mention that end devices are 'sticky' to their paired router/coordinator device.

Thanks for your help @Koenkk

Vout, 2019-01-28:
@clockbrain not sure if you are correct because is have seen Xiaomi switches "moving around" inside the network to another router (i have paired all of them through the coordinator since pairing with an router didn't work), however i have seen that the Zigbee 'mesh network" isn't always the most logic one.

apeinspace, 2019-01-28:
> I've since done some more reading and realised my assumption about self-healing of the mesh for end devices was incorrect. End devices only pair with a single router or coordinator and stay paired [...]

Can you provide a link where this is specified ? I would like to read up on that as well.

In fairly certain I have seen end-devices "roam" to a new router without re-pairing them. Unfortunately I have observed as well that my Xiaomi equipment does not "roam" reliably/at all as Koenkk said.

ryanbeaton, 2019-01-28:
Well what @clockbrain writes is not in line with what has been written before it does make sense.  We know the coordinator only has limited memory and can only handle a certain number (depending on firmware version) of devices.  That's why we have to add routers because they only take up one slot on the coordinator and they give us slots for more devices.  If a device is using a router slot (so it doesn't take up one on the coordinator) then it is probably stuck there or has to move to another router so not to overload the coordinator.

I'm still trying to get a complete understanding of how the network works so anything anyone can add would be appreciated.

clockbrain, 2019-01-29:
@Koenkk fair enough. Does this mean automatic re-parenting doesn't work for any end device (e.g. limitation of CC2531/Z-Stack) or it only works for certain models of end device & router? In either case it would be good to have this limitation noted in the documentation as it has a bearing on network design and/or device selection.

Koenkk, 2019-01-29:
@clockbrain it's a limitation of the device, some re-parent some don't (e.g. some/all Xiaomi devices). Can you make a PR to add this here: https://github.com/Koenkk/zigbee2mqtt/blob/master/docs/information/zigbee_network.md#end-device ?

clockbrain, 2019-01-31:
Documentation updated via PR [https://github.com/Koenkk/zigbee2mqtt/pull/983](https://github.com/Koenkk/zigbee2mqtt/pull/983)

closed 2019-01-31 by clockbrain