Koenkk/zigbee2mqtt #951: beacon enabled communication network

msadeghz, 2019-01-27:
Hi friends,
I've bought a smart switch and socket of a chinese company and want to add them to the network. I used SmartRF packet sniffer and found that the end devices send beacon request and when I use their specific hub, the hub responses to the request. 
is anyway to zigbee2mqtt hub supports beacon communication mode?

Koenkk, 2019-01-27:
AFAIK this shouldn't be a problem. 

Did you try to pair it with zigbee2mqtt?

msadeghz, 2019-01-28:
from the end-device side I see the join message and get the network PANID and the network address. I receive "tcDeviceInd" message in hub.
[image]

Koenkk, 2019-01-30:
In https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L107 you see that that is already done (https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L115)

closed 2019-04-07 by stale[bot]