Koenkk/zigbee2mqtt #955: Running cc253 868 MHz zigbee Channel

JLFN, 2019-01-28:
Running 868 MHz zigbee Channel as option.
Following table-2 mention zigbee/wpan 868MHz channel. This zigbee band has only one channel designated as channel-0, which delivers about 20 Kbps data rate. It is used in Europe region
it it possible to run CC2530 with 868mhz with an 868mhz antenna of course.
 I have  Greenwave Reality Energy Monitor that i cant get in with 2.4Ghz i tried different channels.

JLFN, 2019-01-28:
Looks like its running on Channel 15.
[image]

khzd, 2019-01-28:
You have to switch to channel 15 to pair :( se configuration.yaml

JLFN, 2019-01-28:
> You have to switch to channel 15 to paire :( se configuration.yaml

Yes iam running Chanel 15 now what about the panid ?

```.yaml 
  # Optional: ZigBee pan ID
  pan_id: 0x1a62
```

JLFN, 2019-01-28:
@Koenkk  You have any tips?

JLFN, 2019-01-28:
@Koenkk Do you have any tips?
[image]

Koenkk, 2019-01-29:
@JLFN looks similar, are you unable to pair it with zigbee2mqtt?

JLFN, 2019-01-29:
@Koenkk I cant pair, i cant even get incomming device. No changed when i start debug mode neither. I wonder if its something that need extPanId? Do you have any tips where to begin?

ptvoinfo, 2019-01-29:
@JLFN It is necessary to capture traffic between a working coordinator and the device like we did it for Livolo.

JLFN, 2019-01-29:
@ptvoinfo cc2531 or cc2530 with zigbee2mqtt or you do mean the official gateway of the vendor?

ptvoinfo, 2019-01-29:
@JLFN I mean the official gateway.

JLFN, 2019-01-30:
@ptvoinfo  I have recorded the gateway with the device . Please help me out
[Greenwave reader.zip](https://github.com/Koenkk/zigbee2mqtt/files/2813912/Greenwave.reader.zip)

ptvoinfo, 2019-01-31:
@JLFN Is it possible to capture traffic for a pairing process of the official gateway and the reader?

JLFN, 2019-01-31:
@ptvoinfo tried that but doenst give any security header with z.bee-seckey .

ptvoinfo, 2019-01-31:
Strange. At the pairing stage, the gateway should send unencrypted data.

JLFN, 2019-01-31:
@ptvoinfo  This you see this part ?
[image]

ptvoinfo, 2019-01-31:
Look here:
https://www.youtube.com/watch?v=4vG4CVNAm_A

JLFN, 2019-01-31:
@ptvoinfo  Yes i know it works i did it on my other zigbee networks and got the z.bee key easy but here its not showing the same behavor.

endor-force, 2019-02-01:
I just got myself one of these. 
There is a serial interface on the board, 4 pins on the top left, 1 = VCC, 2 = RX, 3 =TX 4 = GND. 
Terminal 8n1 115200bps
Note, 3,3V - do not use 5v serial interfaces.

When it is powered on it outputs this data, replaced some values with X to obfuscate.:
N: GR0B-1108-00221
IEEE: D4AXX802000000XX
Key: A5ADE3AF48AD3720FAF219DBCBXXXXXX
FW: 0.63
HW: 3
CH: 0x6718000
Z-Stack: 2.5.1
P1 type: 1, scale: u, res: u, base: u, count: u
P1 on: 500 us, poll: 50 ms
P1 IR state: 0, cal: 0, thres: 0
P2 type: 4, scale: u, res: u, base: u, count: u
P2 on: 500 us, poll: 50 ms
P2 IR state: 0, cal: 0, thres: 0
Meter state: 0

JLFN, 2019-02-01:
@endor-force  Can help out in anyway? How did you get the key ?

JLFN, 2019-02-01:
@endor-force did you use this key with whireshark to try to decode? I have two gateways but nothing happens when i try to pair. like it doenst want to pair.

endor-force, 2019-02-01:
Nope have not connected wireshark yet, i'm actually having a deconz / conbee setup but i can not see the device in their ui.. will test with wireshark as well.

JLFN, 2019-02-01:
@endor-force I have a sniffer i speak Swedish also can we talk on discord or some place please?

endor-force, 2019-02-01:
@JLFN I'm in home assistant discord server as endor.
https://discord.gg/Y56RSW

JLFN, 2019-02-05:
802.15.4 JenNet is used not Zigbee.

closed 2019-02-06 by JLFN