Koenkk/zigbee2mqtt #960: Accessing zigbee2mqtt through the browser?

FRFrancisqueti, 2019-01-29:
Hello Koen.
Is it possible to send commands to zigbee2mqtt via browser using http: // localhost: 1883?
Is there any way to control (and configure) zigbee devices acting directly on zigbee2mqtt, without the aid of an HA?

Thank you.

Koenkk, 2019-01-29:
Yes, just send to MQTT commands (this is what HA also does) https://koenkk.github.io/zigbee2mqtt/information/mqtt_topics_and_message_structure.html

FRFrancisqueti, 2019-01-29:
Could you show an example? I had already consulted this tutorial but could not mount the correct syntax to post to the browser.

Thank you.

Koenkk, 2019-01-29:
You can't control it from the browser, you need to use an MQTT client (e.g. https://mqttfx.jensd.de/)

JLFN, 2019-01-29:
You can use nodered

FRFrancisqueti, 2019-01-29:
Yes, but I'm doing my own hub, developing in dotnet and wanted to talk directly to zigbee2mqtt ...

FRFrancisqueti, 2019-01-29:
I'm just working on this tutotial ... very confusing writing on zigbee2mqtt. I can read, but I do not know what to send to zigbee2mqtt to act the sensor (Smartthings outletv4) ... if someone can help ...

FRFrancisqueti, 2019-01-31:
Hello friends, I managed to upload the server with MQTTnet and send commands to zigbee2mqtt, thanks for the help.
Koen, one last question, I edited the file opt / zigbee2mqtt / li / state on line 5 (const saveInterval = 1000;) to have an update time of opt / zigbee2mqtt / data / state.json every 1 second because through that json I collect data from attached devices for my API. But this is generating "Error: EBUSY:" by closing the zigbee2mqtt server when sending a command to the device at the same time. Would you like to create a topic for this state.json update to be made only when you want?

Thank you.

FRFrancisqueti, 2019-01-31:
Follow the zigbee2mqtt log:

 zigbee2mqtt:debug 2019-1-31 08:28:59 Saving state to file C:\opt\zigbee2mqtt\data\state.json
  zigbee2mqtt:debug 2019-1-31 08:29:00 Saving state to file C:\opt\zigbee2mqtt\data\state.json
fs.js:115
    throw err;
    ^

Error: EBUSY: resource busy or locked, open 'C:\opt\zigbee2mqtt\data\state.json'
    at Object.openSync (fs.js:439:3)
    at Object.writeFileSync (fs.js:1190:35)
    at State.save (C:\opt\zigbee2mqtt\lib\state.js:50:12)
    at Timeout.timer.setInterval [as _onTimeout] (C:\opt\zigbee2mqtt\lib\state.js:19:45)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.0.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Rizzo\AppData\Roaming\npm-cache\_logs\2019-01-31T10_29_00_654Z-debug.log

Koenkk, 2019-01-31:
You shouldn't have to modify zigbee2mqtt for this, rather just listen to what zigbee2mqtt sends to `zigbee2mqtt/+`

closed 2019-04-08 by stale[bot]