Koenkk/zigbee2mqtt #961: No zigbee2mqtt issue! - Heiman smoke detector HS3SA vs. HS1SA

h4nc, 2019-01-29:
Does someone here know the difference between the two Heiman smoke detectors?

There are two models HS3SA and HS1SA?

They look a little different. 
Which one is newer? Is the one an updated version of the other one?

agehring80, 2019-03-12:
Hello.

Nothing new ? I cannot find a detailed description.
What are the features, what is the difference ?

Please.

h4nc, 2019-03-12:
I ordered a HS1SA and got a HS3SA.

So I have the HS3SA now. 
I don’t think there is difference.

Also there are different endings like -N and -M.
Reseller told me they are both Zigbee.

closed 2019-05-18 by stale[bot]

dfsx1, 2020-03-22:
I hope someone can answer this, as HS3SA device is not in their official product list?
http://www.heimantech.com/product/smart_zigbee.html