Koenkk/zigbee2mqtt #962: OSRAM issues with Broadcast Messages and all lights

jamestutton, 2019-01-29:
Hi so I recently added "support" for the Osram/lightify Switch mini. But it seems the challenges with these device may not quite be over yet.

So the issue I seem to be having is although both the Bulbs (OSRAM AC03641) and the remote are both paired with zigbee2mqtt and can control the bulbs via MQTT mesages no issue.  The issue is the switches seem to control the bulbs via some sort of broadcast messages.  This would be fine but the messages seem to uncontrollable and effect all lights.  So anOn/Off command on any switch turns on/off "ALL" my osram bulbs in the house.  "Not good"

And to top it off this happens even when zibee2mqtt is stopped.  So it appears the bulbs/remotes are doing so sort of direct pairing although no direct pairing has taken place.  

I have a couple of CC2351 so can do some sniffing but ultimately hoping someone here will have some next steps to get to the bottom of this.

My current thinking is that when the remote joins it should probably be sent some sort of group or other command to disable its broadcast mode, but my zigbee knowledge is fairly limited at this point.

Have tried OSRAM support but they not been overly helpful and unfortunately don't have an original bridge/gateway to test how it works in the vendors own app.

jamestutton, 2019-01-29:
It appears other have seen similar things on the Smart-things side 

[Smart things](https://community.smartthings.com/t/release-osram-lightify-switch-mini/115412/18)

And with deconz its been raised [issue 294](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/294)

We just to figure how the native APP does it or if anyone here has the gateway and can do a packet capture perhaps.  Id love to get these working as currently one of the cheapest zigbee switches on the market and quite powerful as effectively a 6 button remote as all 3 buttons have short and long press options.

Koenkk, 2019-01-30:
It's interesting to know what the device actually sends, I expect a broadcast to `0x0000` (all devices). Can you confirm this by doing zigbee sniffing?

In order to fix this, we indeed need to have sniffing of the original gateway.

jamestutton, 2019-01-30:
Unfortunately don't have the gateway and am seeing if osram willing/able to
provide some further detail around how the gateway operates.

As for sniffing currently away so will get packet captures sent in next
couple of days but from recall it does indeed use the 0x0000 address.

 In some ways the master on/off switch is off great use(by main door on
leaving for example). Just not when I want to control light in a single
room.

jamestutton, 2019-02-01:
Full Output of a single message below, hopefully it with give some clues:

```
Frame 12643: 48 bytes on wire (384 bits), 48 bytes captured (384 bits) on interface 0
    Interface id: 0 (-)
        Interface name: -
    Encapsulation type: IEEE 802.15.4 Wireless PAN (104)
    Arrival Time: Jan  1, 1970 02:13:49.053343000 BST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 4429.053343000 seconds
    [Time delta from previous captured frame: 0.053262000 seconds]
    [Time delta from previous displayed frame: 0.053262000 seconds]
    [Time since reference or first frame: 4425.648148000 seconds]
    Frame Number: 12643
    Frame Length: 48 bytes (384 bits)
    Capture Length: 48 bytes (384 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: wpan:zbee_nwk:zbee_aps:zbee_zcl:zbee_zcl_general.onoff]
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x4c09
    Frame Control Field: 0x8841, Frame Type: Data, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
        .... .... .... .001 = Frame Type: Data (0x1)
        .... .... .... 0... = Security Enabled: False
        .... .... ...0 .... = Frame Pending: False
        .... .... ..0. .... = Acknowledge Request: False
        .... .... .1.. .... = PAN ID Compression: True
        .... ...0 .... .... = Sequence Number Suppression: False
        .... ..0. .... .... = Information Elements Present: False
        .... 10.. .... .... = Destination Addressing Mode: Short/16-bit (0x2)
        ..00 .... .... .... = Frame Version: IEEE Std 802.15.4-2003 (0)
        10.. .... .... .... = Source Addressing Mode: Short/16-bit (0x2)
    Sequence Number: 88
    Destination PAN: 0x1a63
    Destination: 0xffff
    Source: 0x4c09
    [Extended Source: Osram_aa:00:ad:9a:2c (7c:b0:3e:aa:00:ad:9a:2c)]
    [Origin: 12434]
    FCS: 0x3510 (Correct)
ZigBee Network Layer Data, Dst: Broadcast, Src: 0x0ec8
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
    Destination: 0xfffd
    Source: 0x0ec8
    Radius: 10
    Sequence Number: 59
    [Extended Source: Ember_00:0f:24:0e:b5 (00:0d:6f:00:0f:24:0e:b5)]
    [Origin: 1097]
    ZigBee Security Header
ZigBee Application Support Layer Data, Dst Endpt: 255, Src Endpt: 2
    Frame Control Field: Data (0x08)
    Destination Endpoint: 255
    Cluster: On/Off (0x0006)
    Profile: Home Automation (0x0104)
    Source Endpoint: 2
    Counter: 244
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 116
    Command: Off (0x00)
```

Koenkk, 2019-02-02:
So as expected, this switch indeed send a broadcast request. We need to find out if this device requires additional configuration/how the OSRAM gateway deals with this.

pixeldoc2000, 2019-03-12:
I have the Osram "Switch 4x EU-LIGHTIFY", Osram Mini Switch, Osram Lightify Gateway and Sniffer.

What part of the traffic to we need to sniff?

I guest the pairing and normal operation?

tb-killa, 2019-03-12:
@Koenkk doesn't work the adding ``lightingColorCtrl 0 76`` anymore with the switch?

jamestutton, 2019-03-15:
Just an update on this issue.  Have made various attempts to contact OSRAM for information but no luck from them.

jamestutton, 2019-03-15:
Hi Pixel, just seen your message. So do you have some bulbs as well?  First thing is to prove it actually work as expected with the OSram gateway.  So can you pair an "Osram Mini Switch" with the gateway and it only then control single bulb.  If so can you list the steps to do this.  Ie do you pair the switch to gateway then hold switch near bulb to pair bulb (like tradfri). Or do you link both bulb and switch to app then somehow add them to a group.  Does the switch still control the bulb if you unplug the gateway.

pixeldoc2000, 2019-03-16:
Yes, i have Osram bulbs as well.

Switch Buttons:
[image]

To pair hold Button 1 + 2 for 3 seconds. Led will flash blue multiple times.
To reset hold Button 2 + 3 for 10 seconds. Led will flash green multiple times.
This info is given in the Osram Lightify App to control the Osram Lightify Bridge.

Here is the Capture of pairing the Osram Mini Switch to Osram Lightify Bridge:
https://pastebin.com/UxGJAWqL

Bridge: 84:18:26:00:00:8b:8f:16
Mini Switch: 000d6f00101cf178

Will do further sniffing ASAP.

pixeldoc2000, 2019-03-16:
Capture of pairing the Osram Lightify Switch 4x EU to Osram Lightify Bridge:
https://pastebin.com/iNb1NE5S
Bridge IEEE: 84:18:26:00:00:8b:8f:16
Switch 4x EU IEEE: 000d6f000c632502

jamestutton, 2019-03-18:
So that all looks to make sense.  So are you able to set one to control a single device and capture what ever is sent to the switch to update it and also what the switch sends afterwards.  Guessing it must send something to disable broadcast mode and bind it to a cluster or something somehow.

pixeldoc2000, 2019-03-31:
Here is the capture of Osram Mini Switch Button 1 configured via Osram Lightify Bridge to control an Osram RGBW Bulb:
https://pastebin.com/sUR1ceKd

Osram RGBW Bulb IEEE: 8418260000ca39b4

[image]

It works like this in Lightify. You select the switch in the Lightify app. Select the button of the switch you want to change (in case of the Osram Mini Switch all three buttons can be changed from all devices to a specific device). You can choose different action for short and long press of the button.
[image]

Than the bridge will write the config to the switch after holding button 1 & 2 of the switch for 3 seconds (same as pairing). The Switch will flash blue and the config upload will start.
[image]

Looks like the switch than will setup clusters with the device assigned to switch button and the bridge.

First two clusters with the bridge, than two with the bulb:

     92 01:00:40,821895    ZigBee ZDP 0x0001                0x5595                127    Bind Request Src: Ember_00:10:1c:f1:78, Dst: Osram_00:00:8b:8f:16, Power Configuration (Cluster ID: 0x0001)
    119 01:00:43,830869    ZigBee ZDP 0x0001                0x5595                127    Bind Request Src: Ember_00:10:1c:f1:78, Dst: Osram_00:00:8b:8f:16, Poll Control (Cluster ID: 0x0020)
    423 01:01:27,145586    ZigBee ZDP 0x0001                0x5595                127    Bind Request Src: Ember_00:10:1c:f1:78, Dst: Osram_00:00:ca:39:b4, On/Off (Cluster ID: 0x0006)
    429 01:01:28,104005    ZigBee ZDP 0x0001                0x5595                127    Bind Request Src: Ember_00:10:1c:f1:78, Dst: Osram_00:00:ca:39:b4, Level Control (Cluster ID: 0x0008)

pixeldoc2000, 2019-03-31:
> So can you pair an "Osram Mini Switch" with the gateway and it only then control single bulb. 

Yes. By default the switch control all lights. You have to configure every switch buttons in the Lightify App via the bridge to control a specific device.

> If so can you list the steps to do this.

See above.

> do you pair the switch to gateway then hold switch near bulb to pair bulb (like tradfri). Or do you link both bulb and switch to app then somehow add them to a group.

You pair the switch and the bulb with the bridge.

> Does the switch still control the bulb if you unplug the gateway.

Yes.

aveseli, 2019-05-07:
did someone make any progress here?
would be cool if the all on/all off broadcast message could be stopped without the osram gateway

Koenkk, 2019-05-09:
I will try to get my hand on this device

Koenkk, 2019-05-14:
I've received the device.

- I managed to fix the broadcast issue (it's only sending to the coordinator now)
- I've improved the integration, the following `action`'s are send now: `down`, `down_hold`, `down_release`, `up`, `up_hold`, `up_release`, `circle_click`, `circle_hold`, `circle_release`.
- Can be tested in the latest dev branch (http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2019-05-14 by Koenkk

aveseli, 2019-05-15:
@Koenkk thanks a lot! I will give it a try!

RyuzakiKK, 2019-05-18:
@Koenkk I just tried it with the dev branch and I can confirm that it works, with a caveat.
If the switch is routed through a router, and not directly to the coordinator, the switch is still able to send a broadcast signal to the devices in my network.

As a router I'm using another cc2531.

Could this be fixable with a router firmware upgrade?

jamestutton, 2019-05-18:
Are you sure that is the case? In my testing so far I found the switch
sends broadcast on initial join and shortly afterwards if pressed but
within 60s or so I'm guessing the new binds are sent and it no longer
broadcasts.

RyuzakiKK, 2019-05-18:
@jamestutton After I saw that the broadcast messages continued, I disconnected my router and re paired the switch very close to the coordinator. Then I re plugged the router and since then the switch is working as expected without any more broadcast messages.

So I assumed that the problem was the router but I'm not 100% sure.
Now it is working but I don't know if the switch is still routing through the router or not (is there a way to see the message route?).

But if it is working also for you then probably we can assume that there was something odd in my configuration that a reset and a re pair solved.

martin3000, 2019-08-29:
I'm using ZHA and after pairing the switch with the zigbee stick, the switch did not control my lights. Only after binding the switch with lights, they get switched. So maybe there is something in the ZHA code or ZHAQUIRKS that does this. 
https://github.com/dmulcahey/zha-device-handlers/tree/dev/zhaquirks/osram
https://github.com/home-assistant/home-assistant/tree/dev/homeassistant/components/zha

Ascia42, 2020-02-02:
Hello everyone.
I read this thread with great interest because i have the same problem. I bougth a OSRAM switch mini "AC0251600NJ/AC0251100NJ" to switch my lights. But after pairing the switch with my RaspBerryPi-powered zigbee2mqtt, it broadcasts to all my Osram Smart+ Plugs (zigbee switchable socket).

When looking through the debug-output I can see that zigbee2mqtt detects all messages from the switch, but all sockets turn on/off also, even if the RaspberryPi is powered down. So assume I have this broadcast-probleme described in this issue.

Following this thread I switched to dev-branch. 
I am using the project from https://github.com/ct-Open-Source/ct-Smart-Home, which uses the zigbee2mqqt docker. My docker-compose.yml looks like this. 
> [...]
zigbee:
    image: koenkk/zigbee2mqtt:latest-dev
    volumes:
      - ./data/zigbee:/app/data
      - /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters
    devices:
      - "/dev/ttyACM0:/dev/ttyACM0"
    restart: always
[...]

To get the newest devices.js I followed the steps

> cd /opt
git clone https://github.com/Koenkk/zigbee-herdsman-converters.git

and added the following setting to my compose.yml as you can see above.

> -v /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters

Additional question: Is it "right github-style" to add to an existing issue or would it be more correct to open a new one.

Thanks in advance, any hint will be appreciated.

Cheers
Ascia

p.s: Thanks for all the great work you put in zigbee2mqtt, such a great project!!

sebastianovide, 2020-10-17:
I've used the OSRAM Smart+ mini switch without problems until today. Today I've upgraded my coordinator to a electrolama zzh! and now I'm experiencing this problem...  has not problem not been fixed in this firmware ?

Koenkk, 2020-10-17:
@sebastianovide try to reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure (make sure to wakeup the remote by clickng a button right before doing so), check log if OK.

sebastianovide, 2020-10-17:
> @sebastianovide try to reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure (make sure to wakeup the remote by clickng a button right before doing so), check log if OK.

no luck... I've did it, then the switch stop working completely. So I've reset it and repaired again. The problem persisted. I've tried again to reconfigure with "zigbee2mqtt/bridge/configure" but I get this error

```
info  2020-10-17 15:38:12: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'down'
info  2020-10-17 15:38:12: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'down'
debug 2020-10-17 15:38:18: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'osram switch'
info  2020-10-17 15:38:18: Configuring 'osram switch'
error 2020-10-17 15:38:28: Failed to configure 'osram switch', attempt 1 (Error: Bind 0x000d6f000e1b7b68/1 genOnOff from '0x00124b0018e1f550/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
```

Koenkk, 2020-10-17:
@sebastianovide can you provide the herdsman debug log when pressing a button?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

sebastianovide, 2020-10-17:
after reconfiguring, nothing at all.... clicking the buttons will not give any log... 

 so i had to reset and repair. 

```
debug 2020-10-17 15:53:23: Received Zigbee message from 'osram switch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"up","battery":100,"elapsed":676970,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"","battery":100,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
debug 2020-10-17 15:53:23: Received Zigbee message from 'osram switch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"up","battery":100,"elapsed":26,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
debug 2020-10-17 15:53:23: Received Zigbee message from 'osram switch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"up","battery":100,"elapsed":7,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
debug 2020-10-17 15:53:23: Received Zigbee message from 'osram switch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"up","battery":100,"elapsed":8,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
debug 2020-10-17 15:53:23: Received Zigbee message from 'osram switch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"up","battery":100,"elapsed":9,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
debug 2020-10-17 15:53:23: Received Zigbee message from 'osram switch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"up","battery":100,"elapsed":8,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"","battery":100,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"","battery":100,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"","battery":100,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"","battery":100,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"","battery":100,"last_seen":"2020-10-17T15:53:23+01:00","linkquality":81,"update_available":false,"voltage":3000}'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
info  2020-10-17 15:53:23: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
```

sebastianovide, 2020-10-18:
and from `docker-compose logs`

```
zigbee2mqtt      | Zigbee2MQTT:info  2020-10-18 11:51:12: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'down'
zigbee2mqtt      | winston:file: written true false
zigbee2mqtt      | winston:file: logged 548676 debug 2020-10-18 11:51:12: Received Zigbee message from 'osram switch', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 2 with groupID 0
zigbee2mqtt      |
zigbee2mqtt      | winston:file: logged 548903 info  2020-10-18 11:51:12: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"down","battery":100,"elapsed":63,"last_seen":"2020-10-18T11:51:12+01:00","linkquality":45,"update_available":false,"voltage":3000}'
zigbee2mqtt      |
zigbee2mqtt      | winston:file: logged 549055 debug 2020-10-18 11:51:12: Received Zigbee message from 'osram switch', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 2 with groupID 0
zigbee2mqtt      |
zigbee2mqtt      | winston:file: logged 549281 info  2020-10-18 11:51:12: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"down","battery":100,"elapsed":8,"last_seen":"2020-10-18T11:51:12+01:00","linkquality":45,"update_available":false,"voltage":3000}'
zigbee2mqtt      |
zigbee2mqtt      | winston:file: logged 549442 debug 2020-10-18 11:51:12: Received Zigbee message from 'Piano 1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt      |
zigbee2mqtt      | }
zigbee2mqtt      | 2020-10-18T10:51:12.234Z mqttjs:client sendPacket :: emitting `packetsend`
zigbee2mqtt      | 2020-10-18T10:51:12.234Z mqttjs:client sendPacket :: writing to stream
zigbee2mqtt      | 2020-10-18T10:51:12.234Z mqtt-packet:writeToStream generate called
zigbee2mqtt      | 2020-10-18T10:51:12.234Z mqtt-packet:writeToStream generate: packet.cmd: publish
zigbee2mqtt      | 2020-10-18T10:51:12.234Z mqtt-packet:writeToStream publish: packet: { cmd: 'publish', topic: 'zigbee2mqtt/osram switch/action', payload: 'down', qos: 0, retain: false, messageId: 26230, dup: false }
zigbee2mqtt      | 2020-10-18T10:51:12.235Z mqtt-packet:writeToStream writeVarByteInt: writing to stream: <Buffer 25>
zigbee2mqtt      | 2020-10-18T10:51:12.235Z mqtt-packet:writeToStream writeNumberCached: number: 31
zigbee2mqtt      | 2020-10-18T10:51:12.235Z mqtt-packet:writeToStream writeNumberCached: <Buffer 00 1f>
zigbee2mqtt      | 2020-10-18T10:51:12.235Z mqtt-packet:writeToStream publish: payload: 'down'
zigbee2mqtt      | 2020-10-18T10:51:12.235Z mqttjs:client sendPacket :: writeToStream result true
zigbee2mqtt      | 2020-10-18T10:51:12.236Z mqttjs:client sendPacket :: invoking cb
zigbee2mqtt      | 2020-10-18T10:51:12.236Z mqttjs:client _sendPacket :: (mqttjs_c290ece5) ::  end
zigbee2mqtt      | winston:file: logged 549669 info  2020-10-18 11:51:12: MQTT publish: topic 'zigbee2mqtt/Piano 1', payload '{"brightness":249,"color_temp":153,"elapsed":1387,"last_seen":"2020-10-18T11:51:12+01:00","linkquality":93,"state":"OFF","update_available":false}'
zigbee2mqtt      |
zigbee2mqtt      | winston:file: logged 549879 info  2020-10-18 11:51:12: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"","battery":100,"last_seen":"2020-10-18T11:51:12+01:00","linkquality":45,"update_available":false,"voltage":3000}'
zigbee2mqtt      |
zigbee2mqtt      | winston:file: logged 550089 info  2020-10-18 11:51:12: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"","battery":100,"last_seen":"2020-10-18T11:51:12+01:00","linkquality":45,"update_available":false,"voltage":3000}'
zigbee2mqtt      |
zigbee2mqtt      | winston:file: logged 550186 info  2020-10-18 11:51:12: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'down'
zigbee2mqtt      |
zigbee2mqtt      | winston:file: logged 550283 info  2020-10-18 11:51:12: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'down'
zigbee2mqtt      |
zigbee2mqtt      | 2020-10-18T10:51:12.238Z serialport/stream _read reading { start: 43197, toRead: 22339 }
zigbee2mqtt      | 2020-10-18T10:51:12.239Z serialport/binding-abstract read
zigbee2mqtt      | 2020-10-18T10:51:12.240Z serialport/bindings/unixRead Starting read
zigbee2mqtt      | 2020-10-18T10:51:12.241Z serialport/bindings/unixWrite write returned: wrote 20 bytes
zigbee2mqtt      | 2020-10-18T10:51:12.241Z serialport/bindings/unixWrite Finished writing 20 bytes
zigbee2mqtt      | 2020-10-18T10:51:12.242Z serialport/stream binding.write write finished
zigbee2mqtt      | 2020-10-18T10:51:12.242Z serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
zigbee2mqtt      |   errno: -11,
zigbee2mqtt      |   code: 'EAGAIN',
zigbee2mqtt      |   syscall: 'read'
zigbee2mqtt      | }
zigbee2mqtt      | 2020-10-18T10:51:12.242Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
zigbee2mqtt      | 2020-10-18T10:51:12.242Z serialport/bindings/poller Polling for "readable"
zigbee2mqtt      | 2020-10-18T10:51:12.256Z serialport/bindings/poller received "readable"
zigbee2mqtt      | 2020-10-18T10:51:12.257Z serialport/bindings/unixRead Starting read
zigbee2mqtt      | 2020-10-18T10:51:12.258Z serialport/bindings/unixRead Finished read 6 bytes
zigbee2mqtt      | 2020-10-18T10:51:12.259Z serialport/stream binding.read finished { bytesRead: 6 }
zigbee2mqtt      | 2020-10-18T10:51:12.260Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
[... cut ...]
zigbee2mqtt      | 2020-10-18T10:51:12.265Z serialport/bindings/unixRead Starting read
zigbee2mqtt      | 2020-10-18T10:51:12.266Z serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
zigbee2mqtt      |   errno: -11,
zigbee2mqtt      |   code: 'EAGAIN',
zigbee2mqtt      |   syscall: 'read'
zigbee2mqtt      | }
zigbee2mqtt      | 2020-10-18T10:51:12.267Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
zigbee2mqtt      | 2020-10-18T10:51:12.267Z serialport/bindings/poller Polling for "readable"
```

Koenkk, 2020-10-18:
@sebastianovide does it work now? Looking at https://github.com/Koenkk/zigbee2mqtt/issues/962#issuecomment-711024551 it seems events are provided now.

sebastianovide, 2020-10-18:
no. It still turning on and off **all** the lights...

Koenkk, 2020-10-19:
@sebastianovide and what happens if you configure now? (while making sure the remote is awake and seeing Successfully configured in the logs)

sebastianovide, 2020-10-19:
for some reason now it is refusing to configure. I've press a button, I see the switch is sending that command ("up") and immediately after I send the configure but it breaks...

```
info  2020-10-19 15:14:34: MQTT publish: topic 'zigbee2mqtt/osram switch', payload '{"action":"","battery":100,"last_seen":"2020-10-19T15:14:33+01:00","linkquality":54,"update_available":false,"voltage":3000}'
info  2020-10-19 15:14:34: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
info  2020-10-19 15:14:34: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
info  2020-10-19 15:14:34: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
info  2020-10-19 15:14:34: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
info  2020-10-19 15:14:34: MQTT publish: topic 'zigbee2mqtt/osram switch/action', payload 'up'
debug 2020-10-19 15:14:35: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'osram switch'
info  2020-10-19 15:14:35: Configuring 'osram switch'
error 2020-10-19 15:14:45: Failed to configure 'osram switch', attempt 2 (Error: Bind 0x000d6f000e1b7b68/1 genOnOff from '0x00124b0021600247/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
```

Koenkk, 2020-10-19:
@sebastianovide maybe bind fails when the device is connected trough a router. Try repairing it very close to the coordinator.

frank-f, 2021-01-17:
> @sebastianovide maybe bind fails when the device is connected trough a router. Try repairing it very close to the coordinator.

This helped me too. IMO it should be added as a note on the device page, because the behaviour was very unexpected for me.

bkbilly, 2023-09-22:
I just bought this device and I tried adding it on my zigbee network, but it controls all the lights and switches I have connected, even if I stop the zigbee2mqtt service.

I've tried resetting the device (middle + down buttons for 10 sec) and pairing it near the coordinator (middle + up buttons for 3 sec), but everytime it connects, it still controls all the devices.

Could you suggest what else to try?

dziedzic99, 2024-07-16:
For future reference a hackish solution which worked for me:
Create an empty group
Bind each button to this group with On off cluster
Now the button just produces events. No more random turning stuff on and off.

oliver073, 2024-10-17:
> For future reference a hackish solution which worked for me: Create an empty group Bind each button to this group with On off cluster Now the button just produces events. No more random turning stuff on and off.

Hello, I have exactly the same problem. Can you please describe your solution in a bit more detail.
Thanks in advance,
Oliver

dziedzic99, 2024-10-17:
> Hello, I have exactly the same problem. Can you please describe your solution in a bit more detail.
> Thanks in advance,
> Oliver

Hi,
My setup is based of Z2M running as an addon to Home Assistant.
I have created an **empty** group called _group.null_ and bound the three endpoints to this group. This doesn't prevent Z2M from seeing actions (keypresses). 
In no way this is a good solution for this problem but it's a successful workaround.

Here are some screenshots from my interface to possibly help you set this up. 

[image]

[image]

oliver073, 2024-10-17:
Thanks!
Now I've done it. I didn't remember to press a button to wake up the switch while binding.

Did you delete the original 3 bindings?

dziedzic99, 2024-10-17:
> Did you delete the original 3 bindings?

Yes, don't know whether it was necessary, but the setup as shown in the screenshot works ok and doesn't explosively trigger every light and fan in the house.

Does it work for you now?

dziedzic99, 2024-11-11:
For the record here's a similar issue: #12397 
In reference to that, [this page](https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html) mentions the workaround performed in the same manner as one I have mentioned above.
Should we perhaps include such information [here](https://www.zigbee2mqtt.io/devices/AC0251100NJ_AC0251600NJ_AC0251700NJ.html)?

backlink-market, 2025-02-18:
Here's a little trick to pair the remote with the coordinator without removing all other OSRAM devices from the network.

Simply wrap the remote in aluminum foil, leaving some space for your fingers. Then, move the remote as close as possible to the coordinator antenna and start the binding process. Since most signals are blocked by the foil, the remote will connect only to your coordinator. This method worked for all my devices on the first try.