Koenkk/zigbee2mqtt #964: Xiaomi temperature sensor is lost connection after restart PRI

Ikif, 2019-01-29:
CC2531 ,with latest firmware
After shutting down the PRI and turn on , the Xiaomi temperature sensor is lost connection. In home assistant I see last sensor values before shutting down.
To return the sensor to connect back enough press ~5 second pair button on the sensor. It is easy for 1 sensor , but it is terrible who has 30 sensors/switches.
How to prevent connection loss after פower outage?After shutting down the PRI and turn on , the Xiaomi temperature is lost connection. In home assistant I see last sensor values before shutting down.
To return the sensor to connect back enough press ~5 second pair button on the sensor. It is easy for 1 sensor , but it is terrible who has 30 sensors/switches.
How to prevent connection loss after Power outage?
https://hastebin.com/jorimahido.apache

Koenkk, 2019-01-30:
Can you add one router to the network and check if this still happens? I cannot reproduce this.

KristijanL, 2019-02-02:
same here! latest FW, latest hassio, brand new sensors, after rpi reboot only door sensor is back but all of the three temp sensors are unknown.

Ikif, 2019-02-04:
> Can you add one router to the network and check if this still happens? I cannot reproduce this.
Hi. I don't have additional router to add.

Koenkk, 2019-02-04:
What devices do you have in your network?

KristijanL, 2019-02-04:
Currently i have only 2 type of devices:
2x - Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)
by Xiaomi
Firmware: Zigbee2mqtt 1.1.0 

1x - Aqara door & window contact sensor (MCCGQ11LM)
by Xiaomi
Firmware: Zigbee2mqtt 1.1.0 

window contact sensor is working fine, only temperature sensors are having issues.

Koenkk, 2019-02-04:
My first guess is that the temperature sensor removes itself from the network when there is no router available (as it works fine for many users). Can you try to add a router to the network?

KristijanL, 2019-02-04:
As soon as i get one, i will try to add it to the network. 
How can i debug this issue?

deuntje, 2019-02-11:
I have the same issue with the temperature sensor, but in my case the sensor stops sending metrics without any remarkable reason. The network was not down at the moment and multiple routers were online too. The temperature sensor stopped working in the middle of the night... 

I can't get it working again without removing it completely from databases.db and configurations.yaml and then bind it again. After re-joining the sensor will work again for a while, but after x time the sensor stops working.

In the logs there is nothing which indicates a problem. When the sensor stops working and I create an networkmap it seems that the sensor is still connected to the coordinator.  

Please let me know if I can help solving this issue with this sensor by providing logs or something.

Ton1965, 2019-02-12:
I had a similar problem with one of the two Xiaomi temperature sensors I have. Another worked without problems. The issue disappeared after I had replaced the faulty sensor battery although the battery level shown was close to 80%. No problems for several months since then.

deuntje, 2019-02-12:
@Ton1965 I have already replaced the battery, I've read that solution before. The battery was also reported as 100% when the device was working.

weegum60, 2019-03-04:
Hi,
I have the same issue with my temperature sensors. It disconnect by itself after few days. Did you guys found any solutions ?

Thank you

marc-gist, 2019-03-04:
Anyone? I just noticed that all 5 of my temp sensors dropped out :(
All other sensors, motion, etc seem to be ok.

closed 2019-05-11 by stale[bot]