Koenkk/zigbee2mqtt #965: ZoneID attribute not used correctly

neerdoc, 2019-01-29:
I have been reading the Zigbee standards, and I believe that the ZoneID attribute is not used correctly since it is set in the shepherd-converters files in the source code. According to the standard it should be a unique value _per device_ (or _per zone_?) and thus must be set at run time so no clashes occur.

> A unique reference number allocated by the CIE at zone enrollment time. Used by IAS devices to reference specific  zones  when communicating  with the  CIE. The ZoneID of each zone stays fixed until that zone is unenrolled.

I would guess that this means that all used ZoneIDs should be stored in the database so that the uniqueness is ensured.

Koenkk, 2019-01-30:
Thanks, although I still don't understand why this should be unique, I don't see any unexpected behavior?

Koenkk, 2019-01-30:
@JLFN maybe I'm misreading but I don't see anything about zoneID in that link?

Koenkk, 2019-01-31:
@neerdoc I think it's rather a bad practice than against the standard. If I interpret the documentation correctly, it's is perfectly fine to have multiple devices in one zone. However, that doesn't solve this issue.

What we need is to:
- Find out how this zoneID is practically used (what are the consequences/advantages when having devices in the same zone)?
- Allow the zoneID to be configured by the user.

closed 2019-04-08 by stale[bot]