Koenkk/zigbee2mqtt #966: Ikea Tradfri state reporting

jbmbn, 2019-01-29:
Hi,

First of all, great work. I'm currently testing zigbee2mqtt and it's exactly what I've been looking for.

Like issue #151 and issue #534, I've been looking into getting the Tradfri light bulbs to return their current state. I've been able to solve this by adding the following configure section to the converter for Tradfri model LED1623G12 dimmable light bulb in devices.js.

```
configure: (ieeeAddr, shepherd, coordinator, callback) => {
    const device = shepherd.find(ieeeAddr, 1);
    const cfgOnOff = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 3, maxRepIntval: 300, repChange: 0};
    const cfgLevel = {direction: 0, attrId: 0, dataType: 32, minRepIntval: 3, maxRepIntval: 0, repChange: 0};
     const actions = [
        (cb) => device.bind('genOnOff', coordinator, cb),
        (cb) => device.bind('genLevelCtrl', coordinator, cb),
        (cb) => device.foundation('genOnOff', 'configReport', [cfgOnOff], foundationCfg, cb),
        (cb) => device.foundation('genLevelCtrl', 'configReport', [cfgLevel], foundationCfg, cb),
    ];
    execute(device, actions, callback);
},
```

This configuration sends a state report every 5 minutes or when state changes. It also sends a brightness report when the brightness of the ligth bulb changes. This works nicely. I've tested it using a remote to control the light bulb so the coordinator does not know the settings have changed.

Note that I'm not the first to try this configuration because the Tradfri control outlet model E1603 has a similar configuration. However it uses a maxRepIntval of 1000 seconds which means that in a quick test, you'll not see any reports.

Using this configuration, the light bulb periodically sends state reports, but light_onoff_brightness does not handle the reports so I added the following section in fromZigbee.js

```
     generic_brightness: {
         cid: 'genLevelCtrl',
         type: 'attReport',
         convert: (model, msg, publish, options) => {
             return {brightness: msg.data.data['currentLevel']};
         },
     },
```

and "fz.generic_state,fz.generic_brightness" to the fromZigbee property of generic.light_on_onoff_brightness in devices.js

Using this configuration, reporting is working properly. However when the light bulb looses power, the configuration is lost too. So when the light bulb is turned on again, we need to send the configuration again. To do so, I created an extension: deviceAnnounce, that captures the devIncoming events and reconfigures the device. Here's my code for the extension:

```
const logger = require('../util/logger');
const settings = require('../util/settings');
const utils = require('../util/utils');
const Queue = require('queue');
const zigbeeShepherdConverters = require('zigbee-shepherd-converters');

/**
 * This extension checks for additional actions to be handled
 * when a devices announces itself in the network.
 */
class DeviceAnnouncementHandler {
    constructor(zigbee, mqtt, state, publishDeviceState) {
        this.zigbee = zigbee;
        this.mqtt = mqtt;
    }

    onMQTTConnected() {
    }

    stop() {
    }

    onZigbeeMessage(message, device, mappedDevice) {
	//check if message is of announcement type
	if (message.type == 'devIncoming' || message.type == 'endDeviceAnnce') {
            logger.info('Announcement: Device incoming...');
        } else {
	    return;
	}

        if (!device) {
            //logger.warn('Message without device!');
            return;
        }

        if (device.ieeeAddr === this.coordinator) {
            //logger.debug('Ignoring message from coordinator');
            return;
        }

        if (!mappedDevice) {
            //logger.warn('Message without mappedDevice!');
            return;
        }

        //check device for post announce handler
	if(mappedDevice.configure) {
            logger.info('Announcement: need to re-configure');
            this.configure(device,mappedDevice);
	}
    }

    configure(device, mappedDevice) {
        const ieeeAddr = device.ieeeAddr;
        const friendlyName = settings.getDevice(ieeeAddr) ? settings.getDevice(ieeeAddr).friendly_name : 'unknown';
        logger.debug(`Configuring ${friendlyName} (${ieeeAddr}) ...`);
        // Call configure function of this device.
        mappedDevice.configure(ieeeAddr, this.zigbee.shepherd, this.zigbee.getCoordinator(), (ok, msg) => {
            if (ok) {
                logger.info(`Succesfully configured ${friendlyName} (${ieeeAddr})`);
            } else {
                logger.error(`Failed to configure ${friendlyName} (${ieeeAddr}) ('${msg}')`);
	    }
        });
    }

}
``` 

However, after powering on, sometimes we receive a devIncoming event. Other times (most of the times actually) a endDeviceAnnce event is received. ZIgbee shepherd forwards devIncoming events to zigbee2mqtt. The endDeviceAnnce event is not forwarded by zigbee shepherd. To solve this, I've added the following code to event_handlers.js

```
controller.removeListener('ZDO:endDeviceAnnceInd', hdls.endDeviceAnnceInd);
....
....
controller.on('ZDO:endDeviceAnnceInd', hdls.endDeviceAnnceInd);
....
....
handlers.endDeviceAnnceInd = function (msg) {
    // { srcaddr, extaddr, request, removechildren, rejoin }
    debug.shepherd('Device: %s endAnnounce.', msg.ieeeaddr);
    var dev = this._findDevByAddr(msg.ieeeaddr);

    if (dev) {
        this.emit('ind:enddeviceannce', dev);
    }
};
```

And to shepherd.js I added the following code:

```
    this.on('ind:enddeviceannce', function (dev) {
	 var endpoints = [];

	 _.forEach(dev.epList, function (epId) {
	     endpoints.push(dev.getEndpoint(epId));
         });

         self.emit('ind', { type: 'endDeviceAnnce', endpoints: endpoints, data: dev.getIeeeAddr() });
    });
```
With these changes, my Tradfri light bulb nicely sends it's current state every 5 minutes, pinging the coordinator even after switching the power off and on again. And also sends state and brightness updates on change when using a remote. For Tradfri lights, this might sufficiently implement issue #775.

The only thing I've not solved yet is to retrieve the current state (and brightness) of the light bulb when it's powered on. Maybe @koenkk can point me in the right direction on how to actively acquire the state in the deviceAnnounce extension.

Regards, Jeroen

jbmbn, 2019-01-30:
OK. Created a PR for deviceAvailablitiy extension. Note that zigbee-shepherd does not propagate the endDeviceAnnce event at the moment and I could not find the correct version of zigbee shepherd to also make those changes.

Koenkk, 2019-01-31:
@jbmbn thanks! This is the zigbee-shepherd version that is being used: https://github.com/koenkk/zigbee-shepherd

jbmbn, 2019-01-31:
Created PRs for my changes to zigbee-shepherd and zigbee-shepherd-converters

Koenkk, 2019-02-01:
I've implemented the reporting feature in the dev branch. As having to change the `configure()` method in `devices.js` to setup reporting is not convenient, you can now set `report: true` for a specific device. Zigbee2mqtt will discover the supported features of this device and automatically setup reporting. For the Ikea devices it will also re-setup reporting when they are powered on again. Documentation (see `report`): https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/configuration/device_specific_configuration.md

@jbmbn can you confirm that this works properly?

Koenkk, 2019-02-01:
It's not working yet, need to fix some things.

Koenkk, 2019-02-01:
Should be working in the dev branch now!

Koenkk, 2019-02-01:
@jbmbn great, please test with the latest changes.

jbmbn, 2019-02-03:
I've been testing using the latest dev branch. I've been testing with a tradfri LED1623G12 and LED1545G12. Before, I only tested using the LED1623G12 which is using the ZLL profile. I expected all Tradfri leds to use that profile but the LED1545G12 is using the HA profile.

Both types behave similar but not exactly the same. The LED1623G12 needs to be reconfigured after powerloss while the LED1645G12 seems to keep its configuration after powerloss.

First the 1645 did not seem to handle reporting at all although it accepted the configuration in the reporting extension. After resetting the 1632 (6 time off and on), reporting did not work anymore. After adding my configure section to devices.js, reporting started working again. So I added the same configuration to the 1645 in devices.js and now both lightbulbs do reporting. So that's good news. Reporting seems to work nicely with tradfri lightbulbs.

However, the reporting.js extension does not seem to configure reporting correctly. I've not been able to figure out why yet.

I did find an issue when starting. When a node exists in configuration.yaml the onZigbeeStarted will try to send messages to a node even if it's not available (i.e. power is off). It results in errors and I'm think that reporting.js is trying to configure the reports too early.

Another thing I found is that the zigbee.report() function does not work for the ZLL profile. There's no delegate endpoint for the ZLL profile. I've created a quick and dirty solution mapping the ZLL profile into the delegate endpoint for HA. Then the zigbee.report() funtion works fine. However, I could only get reporting working correctly by adding a configure section in devices.js

So maybe deviceConfigure and reporting extensions should be combined into one. Also, I think reporting will not only work on Ikea lightbulbs but I think it'll work on other brands of lightbulbs too. Maybe the "isIkeaTradfriDevice" condition is not needed, especially because there's an additional check on report=true setting in configuration.yaml.

One more thing about the parameters for reporting. There are several options. First, the repChange parameter defines how much the value needs to change before sending a report. Setting it to 0 means report any change. The minTime specifies the minimum time between reports. So when a value changes multiple times is a short timespan, the minTime reduces the number of reports.

The maxTime parameter is interesting. If you set it to 300, it will send a report at least every 5 minutes. If we use this on getOnOff, it can be used as a ping signal. If you set minTime >0 and maxTime to 0, no periodic reports will be generated but only on change. I would suggest to use this on dimmer and color values.

Well, this is a lot of text and not a lot of result yet. Basically, reporting is working and I like the generic setup of the reporting extension but unfortunately, it not working flawlessly yet. Especially, after restarts and devices (re)joining the network.

For the moment: To be continued.

Koenkk, 2019-02-04:
What you added manually in `configure()` should be exactly the same as what `reporting.js` does. Meaning there shouldn't be a difference, reading your post it seems there is which is strange.

jbmbn, 2019-02-13:
Took me a while but I finally figured it out. To enable reporting, we not only need to send a configReport request, but first we need to bind the cluster. Without the binding, the device does not know where to send the reports to. I've created a pull request that adds the bind cluster request before configuring the attribute report.

Furthermore I made some changes to the reporting.js extension to have some more flexibilty in the way reports are configured. My idea is to have the genOnOff attribute reporting periodically (once per five minutes, but I forgot to fix maxReportInterval) so it can be used as a ping signal. Other reports  are set to a much larger timeout (one hour) and of course when the value changes. This way, the device state is send to the coordinator when needed and every 5 minutes we know that the device is still alive.

I've also moved the report enable switch to devices.js in zigbee-shepherd-converters because I think you'll always want devices that support reporting, to report their state.

The only thing that's not fixed yet is retrieving the initial state of the device when it get online. Sometimes the attributes are reported when the device is powered on, but sometimes they are not or just one of them is reported. I have not yet figured out how to solve that.

jbmbn, 2019-02-13:
I also made a change to deviceConfigure.js. I think there was a race condition that may cause configuring a device multiple times. So I added a bit more state when configuring a device.

Koenkk, 2019-02-14:
@jbmbn when using device availability, the state of a device is retrieved once it gets online.

jbmbn, 2019-02-22:
Created pull request with new implementation of reporting.

peterforeman, 2019-03-10:
I'm experiencing problems with the LED1623G12 when they experienced powerloss. If they receive power again AND I'm allowing joining devices the devices will work fine. If I disable device joining and the devices power on after power loss they are not able to be controlled. I have to cut power again, enable joining and restoring power again to get them controllable.

```
 zigbee2mqtt:error 3/10/2019, 10:35:40 AM Zigbee publish to device '0x000b57fffec0fa53', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
```

Does your described problem/solution also fix this case?

closed 2019-05-16 by stale[bot]

dmak, 2023-11-04:
> you can now set `report: true` for a specific device

I missed the documentation if that option is available for a specific device, e.g. [LED1623G12](https://www.zigbee2mqtt.io/devices/LED1623G12.html). Also not clear if that option could be set globally:
```
advanced:
  report: true
```
Previously the option was described [here](https://www.zigbee2mqtt.io/information/report.html), but now that page is gone.
Any additional information is appreciated.