Koenkk/zigbee2mqtt #967: Problem updating to 1.1.0

sascha777, 2019-01-30:
Hi,
I just tried to update to 1.1.0 (from 1.0.1) but I get an error with git in the process:

```
[08:26:49] openhabian@openHABianPi:/opt/zigbee2mqtt$ git checkout HEAD -- npm-shrinkwrap.json
[08:26:57] openhabian@openHABianPi:/opt/zigbee2mqtt$ git pull
error: Pulling is not possible because you have unmerged files.
hint: Fix them up in the work tree, and then use 'git add/rm <file>'
hint: as appropriate to mark resolution and make a commit.
fatal: Exiting because of an unresolved conflict.
```

Is that due to an unmerged file or is that due to a problem on my system?
Best,
Sascha

Koenkk, 2019-01-30:
What is the output of `git status` ?

JLFN, 2019-01-30:
@Koenkk  I get the same error. 
https://hastebin.com/mowoguvefo.sql

Koenkk, 2019-01-30:
Can you try with the following commands (this will remove any local changes).

```js
echo "Stopping zigbee2mqtt..."
sudo systemctl stop zigbee2mqtt

echo "Creating backup of configuration..."
cp -R data data-backup

echo "Updating..."
git reset --hard HEAD
git pull

echo "Installing dependencies..."
rm -rf node_modules
npm install

echo "Restore configuration..."
cp -R data-backup/* data
rm -rf data-backup

echo "Starting zigbee2mqtt..."
sudo systemctl start zigbee2mqtt

echo "Done!"
```

sascha777, 2019-01-30:
I tried the hard pull:
https://hastebin.com/fahegexuvo.sql

Starting in the console then leads to the following error:
https://hastebin.com/wehamefeku.sql

The log says:
`[20:54:10] openhabian@openHABianPi:/opt/zigbee2mqtt$ cat /home/openhabian/.npm/_logs/2019-01-30T19_54_10_594Z-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v8.15.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.1.0~prestart: zigbee2mqtt@1.1.0
6 info lifecycle zigbee2mqtt@1.1.0~start: zigbee2mqtt@1.1.0
7 verbose lifecycle zigbee2mqtt@1.1.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.1.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.1.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.1.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.1.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.1.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.1.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at EventEmitter.emit (events.js:214:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at emitTwo (events.js:126:13)
13 verbose stack     at ChildProcess.emit (events.js:214:7)
13 verbose stack     at maybeClose (internal/child_process.js:915:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
14 verbose pkgid zigbee2mqtt@1.1.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.52-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v8.15.0
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.1.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.1.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

Also trying a fresh install does not seem to work:
https://hastebin.com/ekomuvuhil.sql

Any further ideas what I can do?
Thank's for your support!

sascha777, 2019-01-30:
Got it working with another fresh install using
`sudo git clone -b 1.1.0 https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt`