Koenkk/zigbee2mqtt #970: [FR}] donation/patreon link

SamJongenelen, 2019-01-30:
Dude, awesome software. Congratulations on the 1.10 release! This package & documentation is now really going to help me remove 3rd party bridges 👍 
Will restart my Hassio tonight, since it hasn't detected a new version yet..

I'd like a feature request; patreon/donate link for the project so I can buy you a beer.

JLFN, 2019-01-30:
@SamJongenelen  there is a link already  https://www.paypal.me/koenkk

closed 2019-01-30 by Koenkk

frankforpresident, 2020-03-05:
@Koenkk, I would like to donate some BAT earned with my brave browser. If you register I could send  you some. https://creators.brave.com/sign-up

Koenkk, 2020-03-05:
@frankforpresident done (koenkanters94@gmail.com). thanks!

frankforpresident, 2020-03-10:
> @frankforpresident done ([koenkanters94@gmail.com](mailto:koenkanters94@gmail.com)). thanks!

Maybe not linked to github yet? Can't send them via email (or not sure how)

> NOTE: This creator has not yet signed up to receive contributions from Brave users. Your browser will keep trying to contribute until they verify, or until 90 days have passed.Learn more.

Koenkk, 2020-03-10:
@frankforpresident it seems that in order to receive assets I need to fully verify my uphold account. This would require me to send my passport and other personal information which I don't want to do for now. So unfortunately I cannot accept it.

frankforpresident, 2020-03-11:
> @frankforpresident it seems that in order to receive assets I need to fully verify my uphold account. This would require me to send my passport and other personal information which I don't want to do for now. So unfortunately I cannot accept it.

no problem. Just donated via paypal 😉