Koenkk/zigbee2mqtt #971: Crash after re-plugging CC2531

Bruceforce, 2019-01-30:
Hi,

I'm using zigbee2mqtt in a docker container with latest-dev tag from today. If I re-plug my CC2531 stick with coordinator firmware 20190109 the app crashes. If I wait the 60 seconds and restart the container again it's working, but I think it should try to self repair in a loop instead of crashing.

zigbee2mqtt logs: https://pastebin.com/Mc29fRuY
/root/.npm/_logs/2019-01-30T14_51_11_830Z-debug.log: https://pastebin.com/kKpyubkZ

Koenkk, 2019-01-30:
After replugging, before starting zigbee2mqtt, please press the reset button (the one close to the usb port). The green light should go off and you can now start zigbee2mqtt.

Bruceforce, 2019-01-30:
Is this behavior expected? If yes, then the message `Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)` is a bit confusing / misleading. 

Is there a place where this can be documented or is it already documented and I just haven't seen it?

ptvoinfo, 2019-01-31:
@Koenkk I've fixed this bug in my last pull request.

Koenkk, 2019-01-31:
@ptvoinfo thanks, should be solved in the latest dev branch.

sylarevan, 2019-01-31:
@Koenkk Could you indicate how to switch from master to dev branch? I think I have the exact same issue (error at npm start). 
Switching to dev does not require any re-pairing process, right? And finally, how to switch back to master eventually?

sultanakaneel, 2019-02-02:
> After replugging, before starting zigbee2mqtt, please press the reset button (the one close to the usb port). The green light should go off and you can now start zigbee2mqtt.

I have the same problem, pressing the button doesn't turn off the green light however

Koenkk, 2019-02-03:
@sultanakaneel did you also replug it?

sultanakaneel, 2019-02-03:
> @sultanakaneel did you also replug it?

Yep. I stopped zigbee2mqtt, pulled the usb, put it back in, pressed button closest to usb (green light didn't change) and started zigbee2mqtt

Koenkk, 2019-02-03:
@sultanakaneel strange, anyway, can you check if this has been fixed in zigbee2mqtt? Hassio 1.1.1 update is on it's way (should be available within an hour).

sultanakaneel, 2019-02-03:
@Koenkk still experiencing the same problem, sort of. It now gets stuck on the following:

[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-02-03T17:43:48: PM2 log: Launching in no daemon mode
2019-02-03T17:43:48: PM2 log: App [npm:0] starting in -fork mode-
2019-02-03T17:43:48: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.1 start /zigbee2mqtt-1.1.1
> node index.js
  zigbee2mqtt:info 2019-2-3 17:43:51 Logging to directory: '/share/zigbee2mqtt/log/2019-02-03.17-43-51'
  zigbee2mqtt:info 2019-2-3 17:43:52 Starting zigbee2mqtt version 1.1.1 (commit #unknown)
  zigbee2mqtt:info 2019-2-3 17:43:52 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-2-3 17:43:56 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)

It never moves past this bit

Koenkk, 2019-02-03:
@sultanakaneel is it fixed by reflashing the stick?

sultanakaneel, 2019-02-03:
@Koenkk I'm afraid I don't have the hardware to flash it, bought it preflashed

Koenkk, 2019-02-03:
@sultanakaneel do you know what firmware was flashed on it? Maybe that's the reason of: https://github.com/Koenkk/zigbee2mqtt/issues/971#issuecomment-460009215 ?

sultanakaneel, 2019-02-03:
@Koenkk I'm pretty sure it was flashed 2-3 days ago, so I assume it is the latest version. Is there any way I can check?

Koenkk, 2019-02-03:
@sultanakaneel not if you cant get zigbee2mqtt to start (it's logged on startup).

roygbiv856, 2019-02-03:
@Koenkk I have the same problem as OP. I just upgraded to 1.1.1 and I have the latest firmware flashed on my 2531. I tried flashing the newest firmware again on a new unused 2531 and I am still getting the same problem.

Koenkk, 2019-02-03:
@roygbiv856 can you post the complete log? Does it also hang in your case?

roygbiv856, 2019-02-03:
@Koenkk Yes, sir. It just hangs. When I first updated to 1.1.1 today, it would tell me to press the reset button on the 2531 then hang. Now, it just says error while starting shepherd and hangs. I am not sure if this is completely related, but ever since I updated to 1.1.1, my entire hassio freezes up and I need to unplug and plug my pi. This has already happened twice today and it has never happened before in the 3 months I've used hassio.

https://pastebin.com/wQduw7aV

Koenkk, 2019-02-03:
Can you post the debug log (`zigbee_shepherd_debug` options https://github.com/danielwelch/hassio-zigbee2mqtt)

roygbiv856, 2019-02-03:
Yes, I set `zigbee_shepherd_debug` from the z2m add on page. I have to run an errand, but I can post that output when I get home (HA is still frozen at the moment). I have tried using every serial path my pi currently recognizes for the sniffer.

Koenkk, 2019-02-03:
Can you set the `log_level` to `debug` and post the output of that?

sultanakaneel, 2019-02-03:
@Koenkk https://pastebin.com/23JLsuDt

@roygbiv856 Yes it freezes for me as well. The longer z2m runs the slower the pi gets, till it freezes completely. Had to turn off start on boot and can only run it for a few minutes.

Koenkk, 2019-02-04:
@danielwelch do you understand why zigbee shepherd debug logging is not shown when `zigbee_shepherd_debug` is set to `true`?

danielwelch, 2019-02-04:
Not sure why it wouldn’t be showing up, nothing has changed in the run
script in quite some time. Can you post your JSON-formatted add-on config?

sultanakaneel, 2019-02-04:
@danielwelch 
`
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": false,
  "permit_join": false,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://homeassistant",
  "serial_port": "/dev/ttyAMA0",
  "devices": [],
  "network_key": [],
  "log_level": "debug"
}

`

danielwelch, 2019-02-04:
I don’t see zigbee_shepherd_debug anywhere in there...

sultanakaneel, 2019-02-04:
@danielwelch oh right I replaced  "zigbee_shepherd_debug": "true" with log_level, I assume I should've left em both in like this?
https://pastebin.com/Grg78Sp7

danielwelch, 2019-02-04:
@sultanakaneel yep

sultanakaneel, 2019-02-04:
@Koenkk looks like I messed up with the log, new output is this https://pastebin.com/Sd84VjB7 let it run for 40 minutes, quite sure the pi just froze up again

Koenkk, 2019-02-04:
@sultanakaneel can you post the output of `ls /dev`

sultanakaneel, 2019-02-04:
@Koenkk 
https://i.imgur.com/NFjLXHE.png

roygbiv856, 2019-02-04:
@Koenkk I tried reflashing my original cc2531, then flashing a new never used 2531 and the problem persists. At this point, z2m is completely unusable and freezing up my rpi3 after about 10 minutes. Unfortunately, I am going to have to uninstall. I really loved using it, but it's breaking my entire home automation set up. I really hope a future update fixes it!

roygbiv856, 2019-02-04:
@Koenkk Ok I know this is going to sound crazy, but I couldn't just give up on my beloved z2m. I tried putting 2 shephard 2531s into my pi...and it worked! I tried taking each one out to see which was the real shephard, but it wouldn't start without both of them. I even paired a xiaomi temp sensor, so it really is working. One 2531 has a green LED on, the other has no LEDs on. It's extremely strange, but it definitely did the trick. If you had to guess which is the actual shepherd, do you think it's the one with the light on or light off?

https://pastebin.com/40MKEjhd

Koenkk, 2019-02-05:
@roygbiv856 the one with the LED ON, however this is indeed really strange. Are you sure the serial port was right?

roygbiv856, 2019-02-05:
@Koenkk I checked the serial ports before plugging in the 2nd shepherd, then checked again after plugging it in. So, I am positive about what serial port I used in my config.  I'll do some more testing when I get a chance, but now that I got the network working again, I am hesitant to do anything too crazy. Didn't realize how much I need z2m until it was gone

closed 2019-04-13 by stale[bot]