Koenkk/zigbee2mqtt #972: CC2531 routers stop reporting after some time

petro49er, 2019-01-30:
Hello,

I currently run homeassistant and started using zigbee2mqtt.  I currently have 5 CC2531 devices, 1 coordinator and 4 routers placed around my house.  I am back in the midst of testing as the routers drop off periodically and stop responding.  The rate at which they stop responding seems random and I can't pinpoint why.  Sometimes they stop for a short period, but more recently they drop off for longer periods or completely.  Obviously when the router stops responding to the coordinator, the devices connected stop working as well.  Is this a range limitation of the CC2531 or might I have another issue?  If so, what setup would give me a larger range with a more reliable setup?

Koenkk, 2019-01-30:
When they are connected, what is the linkquality?

petro49er, 2019-01-30:
Linkquality is almost always 0 for all the routers for whatever reason.  Every now and then the routers will jump up to 5 or so for a bit, but that's about the highest I've ever seen them.  They almost always report 0.  They will go fine for a few days and then drop off periodically.  Sometimes they show back up, and sometimes they don't.  So it looks like a range issue?  Would the RF TO USB (CC2530 CC2591) black dongle work from AliExpress?  If so, is this something I should replace all my routers and coordinator with as well?  Thank you for your help and work on this project!

petro49er, 2019-01-31:
> When they are connected, what is the linkquality?
> 
> You can also try with a CC2530 as coordinator, which has improved range: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_create_a_cc2530_router.html

Is this a problem that linkquality is almost always 0?  Would the RF TO USB (CC2530 CC2591) black dongle work from AliExpress work as well as the solution you presented above?  Thanks!

petro49er, 2019-02-01:
> @petro49er I don't have any experience with that dongle. It should be enough to keep your current CC2531 as coordinator and add a CC2530 router.
> 
> Linkquality 5 is also a very low linkquality, my routers work reliable with a linkquality of 30+

Yeah my linkquality is usually reporting 0-5 (Usually 0).  How far are your routers placed from the coordinator in order to get a linkquality of 30+?  I have walls to contend with in between, but the routers are anywhere from 15-35ish feet apart.  I suppose I need to experiment moving them closer, but I don't want to lose sensors placed further away.

Koenkk, 2019-02-02:
Distance is not the only thing, e.g.
- Wifi interference https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence
- The range of the CC2531 can greatly be improved by using a usb extension cable (so that the CC2531 is not directly plugged in the PC)

closed 2019-04-10 by stale[bot]