Koenkk/zigbee2mqtt #973: Installation trouble and some questions

dyfcom, 2019-01-30:
Hey, I am new to this.
I use a cc2531 with the latest Z-Stack Firmware on a Rasp B3+ with Openhabian installed.

I got a lot of trouble with the installation and i don't know its everything correct or not.
My cc2531 is connected to: `crw-rw---- 1 root dialout 166, 0 Jan 30 16:31 /dev/ttyACM0`

At my Mqttbroker the state of the bridge is offline. 
So, when i go `cd /opt/zigbee2mqtt` start npm with `sudo npm start`, the state turns to online!
Log:
`> zigbee2mqtt@1.1.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2019-1-30 16:25:50 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-30.16-25-50'
  zigbee2mqtt:info 2019-1-30 16:25:52 Starting zigbee2mqtt version 1.1.0 (commit #cf7e0d1)
  zigbee2mqtt:info 2019-1-30 16:25:52 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-1-30 16:25:53 zigbee-shepherd started
  zigbee2mqtt:info 2019-1-30 16:25:53 Coordinator firmware version: '20181024'
  zigbee2mqtt:info 2019-1-30 16:25:53 Currently 0 devices are joined:
  zigbee2mqtt:warn 2019-1-30 16:25:53 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-1-30 16:25:53 Allowing new devices to join.
  zigbee2mqtt:warn 2019-1-30 16:25:53 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-1-30 16:25:53 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-1-30 16:25:53 Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 2019-1-30 16:25:53 zigbee-shepherd ready
  zigbee2mqtt:info 2019-1-30 16:25:54 Connected to MQTT server
  zigbee2mqtt:info 2019-1-30 16:25:54 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'`
but after hitting CRTL+C it goes offline again.

I configured the deamon out of the "getting-started guide", but after a reboot it doesn't change everything.
The log says this:
`● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-01-30 16:31:56 CET; 2s ago
 Main PID: 937 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─937 npm
           ├─951 sh -c node index.js
           └─952 node index.js

Jan 30 16:31:56 openHABianPi systemd[1]: Started zigbee2mqtt.
Jan 30 16:31:57 openHABianPi npm[937]: > zigbee2mqtt@1.1.0 start /opt/zigbee2mqtt
Jan 30 16:31:57 openHABianPi npm[937]: > node index.js
`
And one question to the pairing a new device, in which log will I find the new device?

Thanks and a nice day.

dyfcom, 2019-01-30:
### And one question to the pairing a new device, in which log will I find the new device?

ok founded, it is in log of `sudo npm start`

Koenkk, 2019-01-30:
Please post your log (`sudo journalctl -u zigbee2mqtt.service -f`)

closed 2019-04-07 by stale[bot]