Koenkk/zigbee2mqtt #976: Aqara Motion Sensor not showing model.

HJKLMN, 2019-01-30:
Connected 2 new Aqara motion sensors today. 
 first one has no model in DB. 2nd one is ok.

From the log:
1/30/2019, 6:11:13 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":
.......................
{"ieeeAddr":"0x00158d0002e35208","type":"EndDevice","friendly_name":"0x00158d0002e35208"},{"ieeeAddr":"0x00158d0002e3500f","type":"EndDevice","model":"RTCGQ11LM","friendly_name":"0x00158d0002e3500f"}]}'

I know perhaps deleting from DB and readding might help, but just wanted to report it.
The log from joining failed device:

1/30/2019, 5:11:03 PM - info: New device with address 0x00158d0002e35208 connected!
1/30/2019, 5:11:03 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0002e35208"}'

The log from working device:
1/30/2019, 6:04:17 PM - info: New device with address 0x00158d0002e3500f connected!
1/30/2019, 6:04:18 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0002e3500f"}'

What is the best way to troubleshoot/fix?

Koenkk, 2019-01-31:
- Stop zigbee2mqtt
- Remove all entries of `0x00158d0002e35208` from `database.db
- Start zigbee2mqtt
- Repair using https://koenkk.github.io/zigbee2mqtt/getting_started/pairing_devices.html#xiaomi

HJKLMN, 2019-01-31:
Solved, THX

closed 2019-01-31 by HJKLMN