Koenkk/zigbee2mqtt #977: Gledopto GL-B-007Z brightness and ct is not updated

milakov, 2019-01-30:
Hi, I am on the latest dev branch, happily using newly added Gledopto GL-B-007Z bulb from Home Assistant. I can control it perfectly, but one thing puzzles me: if I update the lamp brightness or color temperature with a service (either from automation or from web GUI) it is not reflected in GUI. Only On/Off status is reflected.
I don't know if it is supposed to work at all, that is if it is a real bug?

khzd, 2019-01-30:
hi, can you post your log with debug option for this action?

Koenkk, 2019-01-31:
Zigbee2mqtt only confirms the state, not the brightness or color. This shouldn't be needed as home assistant assumes this after the change: https://www.dropbox.com/s/6yt248z51rrf334/2019-01-31%2019.37.01.gif?dl=0

milakov, 2019-01-31:
Yes. it works fine if I change the brightness or color (or ct) through GUI. But if I change it with service "light.turn_on" then the change is not reflected in GUI. If this is theoretically possible to fix (that is if the bulb reports its attributes when changed to the coordinator) then we might at least create an issue for that. It might probably be of low priority though. I admit I might not understand how all this works!

Koenkk, 2019-01-31:
Yes, this is currently being worked on here: #966

closed 2019-01-31 by Koenkk

milakov, 2019-01-31:
Thanks a lot!